brendan-w/python-OBD #94: OBD speed not updated after restarting car ecu

tharunjoy, 2017-07-22:
Hi,
I recently used python-OBD to monitor real time car speed using Raspberry pi2. I was able to successfully connect to obd and watch the car speed real-time. But if I switch the car ignition OFF, I am getting None value in speed and after while switching ON the ignition the speed value is not updated. Please help me to resolve and thanks in advance

bcavileer, 2017-11-19:
I'm having the same problem.  I'm using the Async

`connection.watch(obd.commands.RPM, callback=update_rpm, force=True)  # keep track of the RPM`
`connection.start()`

If I start the connection before the vehicle is running, I won't get any data.
If I stop the engine and start again, I won't get any data.

I only get data when I start watching and the vehicle is already running.

andrepbap, 2018-04-03:
The only way that I found to solve this problem is starting a new connection for every null RPM value. I don't know if this is the better solution, but works.

exemple:
```
import sys
import obd
import time

stop = False
    
def new_rpm(r):
    global stop
    if not r.is_null():
        value = r.value.magnitude
        stop = False
    else:
        value = 0
        stop = True
    print (value)
    sys.stdout.flush()
    time.sleep(0.3)
    
def new_speed(r):
    if not r.is_null():
        value = r.value.magnitude
    else:
        value = 0
    print (value)
    sys.stdout.flush()
    time.sleep(0.3)

def connect():
    print("connecting")
    global connection
    connection = obd.Async("/dev/ttyUSB0", fast=False)
    connection.watch(obd.commands.RPM, callback=new_rpm, force=True)
    connection.watch(obd.commands.SPEED, callback=new_speed, force=True)
    connection.start()

connect()

while True:
    if stop == False:
        time.sleep(1)
    else:
        connection.stop()
        connection.unwatch_all()
        stop = False
        connect()
```