brendan-w/python-OBD #95: Failed to read port

Etregoning, 2017-09-13:
I am trying to use python-obd to connect to my car with an ELM327 1.5 bluetooth adapter. When I first tried to connect it with the default settings (`connection = obd.OBD()`) it told me "failed to set baudrate" -- so I did some research on my car (2003 honda civic si) and found out that it uses the ISO9141-2 protocol, which calls for 10400 baudrate. So I manually set the baudrate, and the port -- and even tried manually setting the protocol as well, but when i did this, it tells me failed to read port. I turned on the debug logging and it said this. I have googled and searched other issues on this github, but none of them gave these same errors. I'm running mac OSX sierra by the way.

**This is my code:**

`import obd`

`ports = obd.scan_serial()`

`print(ports)`
`obd.logger.setLevel(obd.logging.DEBUG)`
`connection = obd.OBD(portstr=ports[2], baudrate=10400, fast=False)`

**This is my output:**

`/usr/local/Cellar/python3/3.6.2/Frameworks/Python.framework/Versions/3.6/bin/python3.6 ` `/Users/elliot/PycharmProjects/42-ESD/main.py`
`['/dev/tty.lpss-serial2', '/dev/tty.lpss-serial1', '/dev/tty.OBDII-SPP']`
`[obd.obd] ======================= python-OBD (v0.6.1) =======================`
`[obd.obd] Explicit port defined`
`[obd.elm327] Initializing ELM327: PORT=/dev/tty.OBDII-SPP BAUD=10400 PROTOCOL=auto`
`[obd.elm327] write: b'ATZ\r\n'`
`[obd.elm327] wait: 1 seconds`
`[obd.elm327] Failed to read port`
`[obd.elm327] read: b''`
`[obd.elm327] write: b'ATE0\r\n'`
`[obd.elm327] Failed to read port`
`[obd.elm327] read: b''`
`[obd.elm327] closing port`
`[obd.elm327] write: b'ATZ\r\n'`
`[obd.elm327] ATE0 did not return 'OK'`
`[obd.obd] Closing connection`
`[obd.obd] Cannot load commands: No connection to car`
`[obd.obd] ` `===================================================================`

rafacpitta, 2017-09-14:
Don't know what setup you're using, but when I use my ELM327 Bluetooth Adapter I had to create a serial port for the bluetooth connection, then I use the `obd.OBD(/dev/my_device)`. Whats happens when you try to communicate with your ELM327 using only pyserial?

```
import serial
import time
ser = serial.Serial('/dev/tty.OBDII-SPP', timeout=1, baudrate=10400)
ser.open()
ser.write(b'ATZ\r\n')
time.sleep(1)
print ser.readlines()
```

mweitzel2005, 2017-11-10:
Duplicate with #58 ?

ctraltdel6, 2017-12-05:
@rafacpitta I am getting this error as well, trying to run on raspberry pi zero/w. Ran the elm through serial as suggested and came back with '{\xab'

masudap, 2018-10-28:
I had the same problem and came across this thread and also other. Here is a specific solution to solve this problem and may not work always, but for me it worked and i hope it might help others.

I solve the problem as described below in a windows 8.1 OS by the following two steps:

1. after pairing the the pc with the elm327, find out the blue tooth port number manually. 
    i.e If using Windows 8/10, navigate: Right-click Start > Control Panel > In the search box, enter "Bluetooth" then select Change Bluetooth settings. a new window will open and in there click the tab "COM ports". there u will see the listed COM ports and note down the one whose direction is outgoing. for example in my case it lists "COM21" and "COM22"  with the names OBDII SPP and OBD respectively.  and "COM21" was the outgoing one and this i need to use.

2. modify the scan_serial() in in the file utils.py so that it will always return the desired port without scanning anything.

i modify the function scan_serial() as listed below, and in essence i commented out the scanning portion of the code and set the return values as result= ['COM21'] which i found in the previous step in case of my computer. 
the location of the utils.py file in my case was: "C:\Users\your_name\AppData\Local\Programs\Python\Python36\Lib\site-packages\obd"

###############################################################

def scan_serial():
    """ Lists serial port names

        :raises EnvironmentError:
            On unsupported or unknown platforms
        :returns:
            A list of the serial ports available on the system
    """
    if sys.platform.startswith('win'):
        ports = ['COM%s' % (i + 1) for i in range(256)]
    elif sys.platform.startswith('linux') or sys.platform.startswith('cygwin'):
        # this excludes your current terminal "/dev/tty"
        ports = glob.glob('/dev/tty[A-Za-z]*')
    elif sys.platform.startswith('darwin'):
        ports = glob.glob('/dev/tty.*')
    else:
        raise EnvironmentError('Unsupported platform')

    #result = []
    result= ['COM21']
    """for port in ports:
        try:
            s = serial.Serial(port)
            s.close()
            result.append(port)
        except (OSError, serial.SerialException):
            pass
	"""
    return result

####################################################################

apecone, 2018-11-19:
@Etregoning , the "failed to set baudrate" might be due to commands being terminated with **\r\n** instead of **\r**.  Please take a look at #122 and let me know if you're still having the same issues.  Since #122 has to do with how commands are written to the ELM327 chip, it's possible it might be related.

alistair23, 2019-02-14:
Closing as the issue appears to be fixed

closed 2019-02-14 by alistair23

sakshamahluwalia, 2019-05-12:
I m still getting the, "failed to set baudrate" issue can someone help me out.

apecone, 2019-05-12:
@sakshamahluwalia - Which version of python-obd are you using?  Also, which OBD adapter are you using?

alistair23, 2019-05-13:
@sakshamahluwalia Do you mind opening a new issue then with details on what you are seeing? Can you make sure you are using the latest master as well

sakshamahluwalia, 2019-05-15:
@alistair23 I dont have the logs but the issue was somewhere along these lines on my mac:
```
[obd.obd] ======================= python-OBD (v0.6.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=/dev/tty.OBDII-SPP BAUD=10400 PROTOCOL=auto
[obd.elm327] write: b'ATZ\r\n'
[obd.elm327] wait: 1 seconds
[obd.elm327] Failed to read port
[obd.elm327] read: b''
[obd.elm327] write: b'ATE0\r\n'
[obd.elm327] Failed to read port
[obd.elm327] read: b''
[obd.elm327] closing port
[obd.elm327] write: b'ATZ\r\n'
[obd.elm327] ATE0 did not return 'OK'
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
```
I have since tried it on ubuntu and after connecting to the elm327 device the issue does not persists... do you still want me to open a ticket?

apecone, 2019-05-15:
Your output says you’re running version 0.6.1.  You need to use the latest checkpoint on the master branch.  Please checkout  latest master and let us know if you’re still having this problem.

sakshamahluwalia, 2019-05-16:
Nope the issue has been fixed.

keewi-bdonahue, 2019-07-03:
I am on the newest version and am still seeing this issue.
`[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/tty.V-LINK-SPP']
[obd.obd] Attempting to use port: /dev/tty.V-LINK-SPP
[obd.elm327] Initializing ELM327: PORT=/dev/tty.V-LINK-SPP BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: b''
[obd.elm327] Response from baud 9600: b''
[obd.elm327] Response from baud 230400: b''
[obd.elm327] Response from baud 115200: b''
[obd.elm327] Response from baud 57600: b''
[obd.elm327] Response from baud 19200: b''
[obd.elm327] Failed to choose baud
[obd.elm327] closing port
[obd.elm327] write: b'ATZ\r'
[obd.elm327] Failed to set baudrate
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
[obd.obd] Query failed, no connection available`

tdrobbin, 2019-08-05:
Hi, I'm also getting the same issue with the latest version, using a Mac if it makes a difference.

hamad12125, 2019-08-08:
still same issue 
`>>> connection = obd.Async(ports[1],fast=False)
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=/dev/ttyUSB0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: '\x7f\x7f\r'
[obd.elm327] Response from baud 9600: '\x7f\x7f\r'
[obd.elm327] Response from baud 230400: '\x7f\x7f\r'
[obd.elm327] Response from baud 115200: '\x7f\x7f\r'
[obd.elm327] Response from baud 57600: '\x7f\x7f\r'
[obd.elm327] Response from baud 19200: '\x7f\x7f\r'
[obd.elm327] Failed to choose baud
[obd.elm327] closing port
[obd.elm327] write: 'ATZ\r'
[obd.elm327] Failed to set baudrate
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
`

hamad12125, 2019-08-08:
i use raspberry pi and 
AG-COM KKL 409.1 USB Interface Diagnostic Cable for Audi & Volkswagen - OBD2 / OBDII

zeesh1979, 2020-11-12:
any update? i am also facing same error 'no connection to car' i am using v 0.7.1 on rpi3....i get the baudrate and initial commands run ok but after ATL0 didnt return OK i get no connection to car error

sakshamahluwalia, 2020-11-12:
You will have to add a delay with the requests to the car in the elm327.py

zeesh1979, 2020-11-12:
Thanks. i added and it solved the issue. now i moved to running commands to get the speed and rpm values...below is log of what the debug showed

[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/rfcomm0']
[obd.obd] Attempting to use port: /dev/rfcomm0
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: b'?\r\r>'
[obd.elm327] Choosing baud 38400
[obd.elm327] write: b'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'\r\rELM327 v2.1\r\r>'
[obd.elm327] write: b'ATE0\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'ATE0\rOK\r\r>'
[obd.elm327] write: b'ATH1\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'ATL0\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'ATSP0\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'SEARCHING...\r18 DA F1 11 06 41 00 BC 3F A8 03 \r18 DA F1 1D 06 41 00 98 3A 80 01 \r\r>'
[obd.elm327] write: b'ATDPN\r'
[obd.elm327] read: b'A7\r\r>'
[obd.protocols.protocol] map ECU 17 --> ENGINE
[obd.protocols.protocol] map ECU 29 --> UNKNOWN
[obd.elm327] Connected Successfully: PORT=/dev/rfcomm0 BAUD=38400 PROTOCOL=7
[obd.obd] querying for supported commands
[obd.obd] Sending command: b'0100': Supported PIDs [01-20]
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'18 DA F1 11 06 41 00 BC 3F A8 03 \r18 DA F1 1D 06 41 00 98 3A 80 01 \r\r>'
[obd.obd] Sending command: b'0120': Supported PIDs [21-40]
[obd.elm327] write: b'0120\r'
[obd.elm327] read: b'18 DA F1 1D 06 41 20 80 00 A0 01 \r18 DA F1 11 06 41 20 80 1D B0 01 \r\r>'
[obd.obd] Sending command: b'0140': Supported PIDs [41-60]
[obd.elm327] write: b'0140\r'
[obd.elm327] read: b'18 DA F1 11 06 41 40 7A D0 00 00 \r18 DA F1 1D 06 41 40 CA C0 00 00 \r\r>'
[obd.obd] Sending command: b'0600': Supported MIDs [01-20]
[obd.elm327] write: b'0600\r'
[obd.elm327] read: b'NO DATA\r\r>'
[obd.obd] finished querying with 73 commands supported

when i run below script, it says Speed and RPM not supported. if i put force=True then no valus is returned. if i send 010C and 010D from serial tool, i get the values

import obd
import time
conn=obd.OBD()
cmd = obd.commands.SPEED # select an OBD command (sensor)
response = connection.query(cmd) # send the command, and parse the response
print("speed",response.value) # returns unit-bearing values thanks to Pint
cmd = obd.commands.RPM # select an OBD command (sensor)
response = connection.query(cmd) # send the command, and parse the response
print("speed",response.value) # returns unit-bearing values thanks to Pint

if i add below to script, i get error that connection has no print_discovered
connection.print_discovered()

zeesh1979, 2020-11-12:
forgot to add output of above speed command without force=True

'b'010D': Vehicle Speed' is not supported