brendan-w/python-OBD #96: "[obd.elm327] Failed to read port" on while loop

bernardoaraujor, 2017-11-16:
I'm trying to continuously read the speed of the car:

```
import obd
import time

connection = obd.OBD() # auto-connects to USB or RF port
cmd = obd.commands.SPEED  # select an OBD command (sensor)

while True:
    response = connection.query(cmd) # send the command, and parse the response
    print(response.value) # returns unit-bearing values thanks to Pint
    time.sleep(1)

```

However, this is what I'm getting:

```
datatraffic@datatraffic-Lenovo-B40-70:~/PycharmProjects/obd$ python3 test.py 
0 kph
[obd.elm327] Failed to read port
None
[obd.elm327] Failed to read port
None
[obd.elm327] Failed to read port
None
[obd.elm327] Failed to read port
None
```

OPrange, 2018-07-11:
Did you figure it out how to fix this?

alistair23, 2018-11-22:
I am unable to reproduce this on master. I'm closing this. If you continue to have the problem please create a new issue including vehicle information and full logs.

closed 2018-11-22 by alistair23