brendan-w/python-OBD #101: UnicodeDecodeError when connecting to ELM327 OBD2 USB interface

dieterbauwens, 2018-02-02:
When connecting an ELM327 OBDII interface, a UnicodeDecodeError occurs in obd\elm327.py, line 465, in __read
`string = buffer.decode()`

This is **fixed** by changing line 465 in obd\elm327.py to:

`string = buffer.decode("utf-8", "ignore")`

Afterwards, the OBDII interface continues to work fine.

Could this change be applied to the trunk?

Anyway, thank you for this great library, nice piece of work!
Cheers,
Dieter

Details:

Observed on OS: Mac OS Sierra 10.12.6, Windows 7
OBD interface: ELM327 OBD2 USB interface (CAN-BUS)
(https://obdwarenhuis.nl/obd-kabels/interfaces/obd-2-usb-interface-can-bus)

`import obd`
`from obd import OBDStatus`
`from time import sleep`
`obd.logger.setLevel(obd.logging.DEBUG)`
`connection = obd.OBD('/dev/tty.usbserial', baudrate=38400, protocol=None, fast=True)`

Error trace:
[obd.obd] ======================= python-OBD (v0.6.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=COM5 BAUD=38400 PROTOCOL=auto
[obd.elm327] write: b'ATZ\r\n'
[obd.elm327] wait: 1 seconds
Traceback (most recent call last):
  File "obd-test.py", line 3, in <module>
    connection = obd.OBD("COM5", baudrate=38400, protocol=None, fast=True) 
  File "C:\apps\Python36-32\lib\site-packages\obd\obd.py", line 58, in __init__
      self.__connect(portstr, baudrate, protocol) # initialize by connecting and loading sensors
  File "C:\apps\Python36-32\lib\site-packages\obd\obd.py", line 85, in __connect
    self.interface = ELM327(portstr, baudrate, protocol)
  File "C:\apps\Python36-32\lib\site-packages\obd\elm327.py", line 143, in __init__
    self.__send(b"ATZ", delay=1) # wait 1 second for ELM to initialize
  File "C:\apps\Python36-32\lib\site-packages\obd\elm327.py", line 408, in __send
    return self.__read()
  File "C:\apps\Python36-32\lib\site-packages\obd\elm327.py", line 465, in __read
    string = buffer.decode()
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xfc in position 4: invalid start byte

andythomnz, 2018-07-20:
+1 I had the same problem, this fixed it. Thanks!

dPeS, 2018-08-07:
+1 here