brendan-w/python-OBD #105: obd.OBD() instruction won't execute properly

E-genin, 2018-05-28:
Hi there, 
I'm trying to use obd library to connect an ELM327 Bluetooth to a car, I use the very basic script : 

`import obd

connection = obd.OBD() # auto-connects to USB or RF port

cmd = obd.commands.SPEED # select an OBD command (sensor)

response = connection.query(cmd) # send the command, and parse the response

print(response.value) # returns unit-bearing values thanks to Pint
print(response.value.to("mph")) # user-friendly unit conversions`

The script freezes at this line : 
`connection = obd.OBD() # auto-connects to USB or RF port`
it seems that it enters in an infinite loop and won't get out from there.

Can you help me with this problem.

Thanks in advance.

graycrofford, 2018-05-29:
Mine takes a good 15-20 seconds with some query items before returning a value. Looks like it is frozen but it isn't.

E-genin, 2018-05-29:
Hi, Thank you so much for the reply !!

Does the exact same script I mentioned before work for you ?

graycrofford, 2018-05-29:
I just have a while loop around the response and print lines.

E-genin, 2018-05-30:
Can you tell the Environment you are developing in (OS, Python version ...)

Thanks

alistair23, 2018-11-04:
Did you figure out what is going on here?

alistair23, 2018-11-17:
Is your serial port a standard /dev/ttyS* port?

E-genin, 2018-11-17:
I think it was a /dev/rfcomm port

E-genin, 2018-11-17:
After pairing with the dongle, does one have to do something specific before running the script ? 
Because in my case, I didn't have any results until I "linked" the OBD dongle to the /dev/rfcomm in Linux, or to a COM port in Windows.

alistair23, 2018-11-17:
What do you mean linked?

E-genin, 2018-11-17:
I meant binding the rfcomm to the dongle's MAC address

apecone, 2018-11-19:
Hi @E-genin 

Can you please provide your debug logs?  Here's how...

`obd.logger.setLevel(obd.logging.DEBUG)`

If you provide these logs, I'll take a look :)

E-genin, 2018-11-19:
For that, I'll need a new OBD dongle (I lost the one I had). Once that done, I'll make another test. 
I still remember that it freezes somewhere between the two debugging lines : 
`[obd.obd] Available ports: []`
and the next debugging line.

closed 2018-12-06 by alistair23