brendan-w/python-OBD #106: Read from Canbus interface

graycrofford, 2018-05-29:
Working well with a bluetooth OBD2 adapter, but looking to see if it is able to read from a network can device (http://copperhilltech.com/pican-2-can-interface-for-raspberry-pi-2-3/).

I feel like I'm just missing something since OBD2 bluetooth is just Canbus, should be an easy way to direct obd.OBD() to read from can0.

E-genin, 2018-05-29:
Hi, 
Sorry for asking you here, but I'm trying to use the obd library to connect my laptop the ELM327 bluetooth, but it freezes on obd.OBD() instruction, did you face a similar problem.
I opened an issue here if you want to answer me : https://github.com/brendan-w/python-OBD/issues/105 

I apologize for asking you here and I really hope you find the answer you're looking for.

Thanks in advance

brendan-w, 2018-06-25:
@graycrofford This library currently only has support for the ELM style interfaces, which operate using a somewhat strange modem-like ASCII interface. While it would certainly be possible to write another backend for SocketCAN (which I assume is what you were referring to with `can0`), this is a project I probably won't get to for a long while. The protocol abstraction layer should be able to handle this though, and I'd probably use a package like [this](https://python-can.readthedocs.io/en/2.1.0/interfaces/socketcan.html) to build the interface class.