brendan-w/python-OBD #108: 0 at the end of command sent

kbielak, 2018-06-11:
Hey,
I was trying to send command 010D (Vehicle speed) without success.
For some reason at the end of command 0 appears.

My script:
`cmdc = obd.commands.SPEED
response = connection.query(cmdc, force=True)`

Logs:
`[obd.elm327] write: '010D0\r\n'
[obd.elm327] read: b'\r?WRONG HEXCHAR COUNT\r>'
`

My question is where does this 0 come from and how to get rid of this?
I've tried sending '010' and get correct response for '0100'.
Could you help me?
Thanks in advance, cheers

brendan-w, 2018-06-25:
looks like a problem with detecting the number of ECUs in your car (this normally looks like `010D1`). That number was used to set the ELM's early-return message count, to avoid long timeouts. This logic had other problems, [and I have since ripped it out in favor of something more robust](36784bffc2ca50bad7b0f7941189eda8a9865467). You have two options:

- Clone python-OBD and install the latest master branch
- disable fast mode with `obd.OBD(..., fast=False)` which skips the appending of this message limit altogether. While it will most likely fix your problem, queries will be slower.

alistair23, 2019-02-14:
Closing as the issue appears to be fixed

closed 2019-02-14 by alistair23