brendan-w/python-OBD #111: SyntaxError: invalid syntax at "import obd"

ManfredMayr, 2018-07-15:
Details:
---
Python 3.7.0 (v3.7.0:1bf9cc5093, Jun 27 2018, 04:06:47) [MSC v.1914 32 bit (Intel)] on win32
Type "copyright", "credits" or "license()" for more information.
>>> import obd
Traceback (most recent call last):
  File "<pyshell#0>", line 1, in <module>
    import obd
  File "C:\Program Files (x86)\Python3\lib\site-packages\obd\__init__.py", line 42
    from .async import Async
              ^
SyntaxError: invalid syntax
>>> 
---

"async" is since Python 3.7 a reserved keyword, see https://docs.python.org/3/whatsnew/3.7.html#new-features

OrangeFlare, 2018-08-05:
Reproducible
```
PS F:\Git\HHUD> python start.py
Traceback (most recent call last):
  File "start.py", line 1, in <module>
    import obd
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\obd\__init__.py", line 42
    from .async import Async
              ^
SyntaxError: invalid syntax
```

akorb, 2018-08-18:
Today, Manjaro updated to python 3.7. I'd appreciate if you fix this minor, but annoying issue.

OrangeFlare, 2018-08-20:
https://github.com/brendan-w/python-OBD/pull/112
If this is approved this will fix the issue

alistair23, 2018-11-02:
This has now been fixed in master

closed 2018-11-02 by alistair23