brendan-w/python-OBD #113: '010D: Vehicle Speed' is not supported

H2508, 2018-08-09:
Hello

Could someone please try and shed some light as to why I might be getting the following messages:

[obd.obd] '010D: Vehicle Speed' is not supported
[obd.obd] '010C: Engine RPM' is not supported
[obd.obd] '0111: Throttle Position' is not supported

Thanks

======================= RESTART: /home/pi/obd test.py =======================
[obd.obd] ======================= python-OBD (v0.6.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/ttyUSB0']
[obd.obd] Attempting to use port: /dev/ttyUSB0
[obd.elm327] Initializing ELM327: PORT=/dev/ttyUSB0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: '\x7f\x7f\r?\r\r>'
[obd.elm327] Choosing baud 38400
[obd.elm327] write: 'ATZ\r\n'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'ATZ\r\r\rELM327 v1.5\r\r>'
[obd.elm327] write: 'ATE0\r\n'
[obd.elm327] read: b'ATE0\rOK\r\r>'
[obd.elm327] write: 'ATH1\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: 'ATL0\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: 'ATSP0\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: '0100\r\n'
[obd.elm327] read: b'SEARCHING...\rUNABLE TO CONNECT\r\r>'
[obd.elm327] write: 'ATDPN\r\n'
[obd.elm327] read: b'A0\r\r>'
[obd.elm327] ELM responded with unknown protocol. Trying them one-by-one
[obd.elm327] write: 'ATTP6\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: '0100\r\n'
[obd.elm327] read: b'CAN ERROR\r\r>'
[obd.elm327] Connected Successfully: PORT=/dev/ttyUSB0 BAUD=38400 PROTOCOL=6
[obd.obd] querying for supported commands
[obd.obd] Sending command: 0100: Supported PIDs [01-20]
[obd.elm327] write: '01000\r\n'
[obd.elm327] read: b'CAN ERROR\r\r>'
[obd.OBDCommand] 0100: Supported PIDs [01-20] did not recieve any acceptable messages
[obd.obd] No valid data for PID listing command: 0100: Supported PIDs [01-20]
[obd.obd] Sending command: 0600: Supported MIDs [01-20]
[obd.elm327] write: '0600\r\n'
[obd.elm327] read: b'CAN ERROR\r\r>'
[obd.obd] finished querying with 7 commands supported
[obd.obd] ===================================================================
0.007078886
[obd.obd] '010D: Vehicle Speed' is not supported
[obd.obd] '010C: Engine RPM' is not supported
[obd.obd] '0111: Throttle Position' is not supported
None
None
None

H2508, 2018-08-17:
Thanks, I'll have a go

alistair23, 2018-11-04:
It sounds like the issue is fixed, I'm going to close this. Feel free to reopen if the issue isn't fixed.

closed 2018-11-04 by alistair23

guruvishnuvardan, 2021-08-15:
Hello @H2508,

I am also facing a similar issue, as yours. Please find the details.

I am using USB OB2 Connector, my objective is to get the speed of the Car(Honda City 2015 Model India)

Please find the following Code:
**import obd
connection = obd.OBD("/dev/ttyUSB0")
ports = obd.scan_serial()
print(ports)

cmd = obd.commands.SPEED
r = connection.query(cmd)
print(r.value)
OutPut:
['/dev/ttyUSB0']
[obd.obd] 'b'010D': Vehicle Speed' is not supported
None**
I have made an attempt to print using connection.print_discovered(), Please find the output
**connection.print_discovered()
AttributeError: 'OBD' object has no attribute 'print_discovered'**

**import obd
connection = obd.OBD("/dev/ttyUSB0")
a = connection.get_supported_commands()
Traceback (most recent call last):
File "", line 1, in
AttributeError: 'OBD' object has no attribute 'get_supported_commands'**

**System Information:**
Raspberry Pi 4
Raspbian OS
pyOBD

Can you please let me know, how you resolved this issue.
[image]

Note: I have attached the USB OBD Connector Picture:

H2508, 2021-08-18:
Hello @guruvishnuvardan 
My issue was hardware, my motorcycle has a different odb port and the adaptor to go from that port to the odb adaptor had a dry joint.

I didn't finish this project, decided programming is not for me.

guruvishnuvardan, 2021-08-19:
Thanks @H2508