brendan-w/python-OBD #123: Can't Read Data after connection

shlok97, 2018-11-23:
Here's my code that I'm running to get my car's data

import obd

class Car:
    def __init__(self):
        obd.logger.setLevel(obd.logging.DEBUG)
        self.connection = obd.OBD("/dev/rfcomm0", fast=False)
        self.commands = obd.commands
        print("Protocol Used: ", self.connection.protocol_id(), self.connection.protocol_name())

    def isConnected(self):
        return self.connection.status() == obd.OBDStatus.CAR_CONNECTED

    def get_rpm(self):
        if self.isConnected():
            cmd = self.commands.RPM
            response = self.connection.query(cmd)
            return response.value
        return -1

    def get_speed(self):
        if self.isConnected():
            cmd = self.commands.SPEED
            response = self.connection.query(cmd)
            return response.value
        return -1

    def get_fuel_level(self):
        if self.isConnected():
            cmd = self.commands.FUEL_LEVEL
            response = self.connection.query(cmd)
            return response.value
        return -1

car = Car()

# CarId: 5bf0fec1196ca8772322a57c

i = 0
while i < 100:
    if car.isConnected():
        #do stuff
        if i % 2 == 1:
            print("RPM: ", car.get_rpm(), "SPEED: ", car.get_speed(), "FUEL: ", car.get_fuel_level())
        else:
            print("RPM: ", car.get_rpm(), "SPEED: ", car.get_speed(), "FUEL: ", car.get_fuel_level())
            # print("RUN TIME: ", car.get_run_time(), "FUEL RATE: ", car.get_fuel_rate(), "FUEL TYPE: ", car.get_fuel_type())
        i += 1
    else:
        print("Unable to connect")
    time.sleep(0.5)

And I'm getting the following output

[obd.obd] ======================= python-OBD (v0.7.0) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: b'\x7f\x7f\r?\r\r>'
[obd.elm327] Choosing baud 38400
[obd.elm327] write: b'ATZ\r\n'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'ATZ\r\xfc\r\rELM327 v1.5\r\r>'
[obd.elm327] write: b'ATE0\r\n'
[obd.elm327] read: b'ATE0\rOK\r\r>'
[obd.elm327] write: b'ATH1\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'ATL0\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'ATSP0\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'0100\r\n'
[obd.elm327] read: b'SEARCHING...\rUNABLE TO CONNECT\r\r>'
[obd.elm327] write: b'ATDPN\r\n'
[obd.elm327] read: b'A0\r\r>'
[obd.elm327] ELM responded with unknown protocol. Trying them one-by-one
[obd.elm327] write: b'ATTP6\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'0100\r\n'
[obd.elm327] read: b'NO DATA\r\r>'
[obd.elm327] Connected Successfully: PORT=/dev/rfcomm0 BAUD=38400 PROTOCOL=6
[obd.obd] querying for supported commands
[obd.obd] Sending command: b'0100': Supported PIDs [01-20]
[obd.elm327] write: b'0100\r\n'
[obd.elm327] read: b'NO DATA\r\r>'
[obd.OBDCommand] b'0100': Supported PIDs [01-20] did not recieve any acceptable messages
[obd.obd] No valid data for PID listing command: b'0100': Supported PIDs [01-20]
[obd.obd] Sending command: b'0600': Supported MIDs [01-20]
[obd.elm327] write: b'0600\r\n'
[obd.elm327] read: b'NO DATA\r\r>'
[obd.obd] finished querying with 7 commands supported
[obd.obd] ===================================================================
Protocol Used:  6 ISO 15765-4 (CAN 11/500)
[obd.obd] 'b'010C': Engine RPM' is not supported
[obd.obd] 'b'010D': Vehicle Speed' is not supported
[obd.obd] 'b'012F': Fuel Level Input' is not supported

It looks like it's not getting any data after the "0100" command. I really need your help in this since I can't figure out any solution to this problem even after days of searching.
Thank you.
Shlok Kapoor

shlok97, 2018-11-23:
Hi,
I deleted the obd library and performed the following command to import from master:
pip install git+https://github.com/brendan-w/python-OBD.git@master
But I'm getting the same response even after that. @alistair23

alistair23, 2018-11-23:
I'm not sure then. The device is sending no data back, so it's hard to debug from there. Are you sure it's all connected correctly to your computer?

shlok97, 2018-11-23:
When I connected the device to an Android mobile, I was getting pretty accurate results on the "Turbo" app so I don't think there's any problem with the ECU/OBD.
I've been stuck with this problem for days now. It would be a great help if I can get this fixed.

alistair23, 2018-11-23:
I would start with looking at why you get "ELM responded with unknown protocol. Trying them one-by-one". Maybe you can figure out more from there

shlok97, 2018-11-24:
So I changed my connection from rfcomm0 to rfcomm1 and I started getting the data without any issues. But when I restarted it, I went back to square one. Can you suggest a way through which I can set up an interruption free connection?

alistair23, 2018-11-24:
It sounds like a connection issue to your computer then

closed 2018-11-24 by alistair23

alistair23, 2018-11-24:
I'm not sure. You will have to look around about connecting Bluetooth devices