brendan-w/python-OBD #135: [obd.obd] '010C: Engine RPM' is not supported

Fernien, 2019-02-02:
Hey, i cant figure out why i get this issue and how to fix it. Any Ideas?

[obd.obd] ======================= python-OBD (v0.7.0) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/rfcomm0']
[obd.obd] Attempting to use port: /dev/rfcomm0
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: '?\r\r>'
[obd.elm327] Choosing baud 38400
[obd.elm327] write: 'ATZ\r\n'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'\r\rELM327 v1.5\r\r>'
[obd.elm327] write: 'ATE0\r\n'
[obd.elm327] read: b'ATE0\rOK\r\r>'
[obd.elm327] write: 'ATH1\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: 'ATL0\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: 'ATSP0\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: '0100\r\n'
[obd.elm327] read: b'SEARCHING...\rUNABLE TO CONNECT\r\r>'
[obd.elm327] write: 'ATDPN\r\n'
[obd.elm327] read: b'A0\r\r>'
[obd.elm327] ELM responded with unknown protocol. Trying them one-by-one
[obd.elm327] write: 'ATTP6\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: '0100\r\n'
[obd.elm327] read: b'CAN ERROR\r\r>'
[obd.elm327] Connected Successfully: PORT=/dev/rfcomm0 BAUD=38400 PROTOCOL=6
[obd.obd] querying for supported commands
[obd.obd] Sending command: 0100: Supported PIDs [01-20]
[obd.elm327] write: '0100\r\n'
[obd.elm327] read: b'NO DATA\r\r>'
[obd.OBDCommand] 0100: Supported PIDs [01-20] did not recieve any acceptable messages
[obd.obd] No valid data for PID listing command: 0100: Supported PIDs [01-20]
[obd.obd] Sending command: 0600: Supported MIDs [01-20]
[obd.elm327] write: '0600\r\n'
[obd.elm327] read: b'NO DATA\r\r>'
[obd.obd] finished querying with 7 commands supported
[obd.obd] ===================================================================
[obd.obd] '010C: Engine RPM' is not supported
None

alistair23, 2019-02-02:
Are you using an official ELM327 device? It looks like there is a connection issue

Fernien, 2019-02-02:
I think so. I bought this one https://www.amazon.de/gp/aw/d/B00NSNNNOQ 
Could you check for me ? And if this is the wrong one could you tell me what to get instead ?

alistair23, 2019-02-02:
I'm not sure, I can't read German. The fact that it is responding with an unknown protocol isn't a good sign. You are also getting a CAN error

alistair23, 2019-02-02:
Sometimes the apps work when this doesn't with the knock off chips. The unable to connect message it's probably a good place to start debugging. See if you can figure out why you get that

alistair23, 2019-02-02:
Otherwise try specifying the baud rate or protocol

Fernien, 2019-02-03:
Could you recommend an adapter for me? I got even more issues with this one. Cant even connect with bluetooth anymore. Gonna sent it back.

Fernien, 2019-02-04:
Thanks for the information. Would you tell me which obd adapter worked for you ?

r00ta, 2019-02-04:
> Thanks for the information. Would you tell me which obd adapter worked for you ?

Actually with the adapters from OBDLink i'm fine with a lot protocols ( my preference is always the one with the usb cable http://www.obdlink.com/sxusb/). If this adapter doesn't work as well, i guess that the issue is your car. 

But before buying a new device, have you tried to install Torque app (https://play.google.com/store/apps/details?id=org.prowl.torquefree&hl=it) and pair it with the device you already have?

Ircama, 2019-02-10:
"ELM327 v1.5" means that the adapter does not include a [genuine ELM chip](https://www.elmelectronics.com/products/ics/obd/). Anyway, another simple check would be to update python-OBD to the lastest GitHub version and switch engine on before testing. If the error occurs with ignition on, either the adapter is not properly connected or it is not able to support the actual protocol.

Fernien, 2019-02-10:
> "ELM327 v1.5" means that the adapter does not include a [genuine ELM chip](https://www.elmelectronics.com/products/ics/obd/). Anyway, another simple check would be to update python-OBD to the [...]

I got the latest version because it was the first time ive tried it.
I bought an ELM327 v2.1 adapter. It will arrive next week. If this will fix my issue I will update this thread. Thanks for the information

Fernien, 2019-02-13:
I tried the new adapter and it works fine for me. The important thing (which I almost forgot while trying today) is to definitely turn on the battery/engine of the car. Otherwise it won't output real data.

closed 2019-02-13 by Fernien