brendan-w/python-OBD #136: import obd very slow

kayjayme, 2019-02-20:
On my raspberry pi 'import obd' takes more than 10 seconds to complete. Is there and way to speed this up ?

alistair23, 2019-02-27:
Have you looked at the logs of what it is doing?

It's probably slow figuring out the baud rate, you can hard code that to save time.

closed 2019-03-03 by alistair23

tokfrans03, 2019-06-03:
Same here, why does it take so long to import?

brendan-w, 2019-08-11:
yeah, I noticed this as well. Haven't profiled it, but my hunch would be the massive [DTC](https://github.com/brendan-w/python-OBD/blob/master/obd/codes.py#L33) and [command tables](https://github.com/brendan-w/python-OBD/blob/master/obd/commands.py#L51) that get instantiated upon import. Would probably be best to move these to a file that is searched/accessed at runtime.