brendan-w/python-OBD #137: Could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'

makarg1001, 2019-02-26:
I am using a Usb OBD2 Dongle (ELM 327) to communicate with the vehicle. Trying to run obd_recorder.py on a raspberry pi with linux as the os on it. When the OBD 2 dongle is connected to pi and not to the Vehicle. obd_recorder.py successfully opens up ttyUSB0 with message "Interface sucfessfully /dev/ttyUSB0 opened". But when I connect the other end of ELM327 to the OBD port of the vehicle and then try to run "obd_recorder.py" exception is caught. "Could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0' "

alistair23, 2019-02-26:
Is there anything in your dmesg?

makarg1001, 2019-02-27:
Yes. 

[    0.000000] Booting Linux on physical CPU 0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Linux version 3.4.112-sun7i (root@bpi-iot-ros) (gcc version 4.8.4 (Ubuntu/Linaro 4.8.4-2ubuntu1~14.04.1) ) #2 SMP PREEMPT Sat May 28 20:25:03 CST 2016
[    0.000000] CPU: ARMv7 Processor [410fc074] revision 4 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine: sun7i
[    0.000000] Memory Reserved:
[    0.000000]  SYS  : 0x43000000 - 0x4300ffff  (  64 kB)
[    0.000000]  VE   : 0x44000000 - 0x48ffffff  (  80 MB)
[    0.000000]  G2D  : 0x49000000 - 0x49ffffff  (  16 MB)
[    0.000000]  LCD  : 0x4a000000 - 0x4bffffff  (  32 MB)
[    0.000000] Memory policy: ECC disabled, Data cache writealloc
[    0.000000] sunxi: Allwinner A20 (AW1651/sun7i) detected.
[    0.000000] On node 0 totalpages: 262144
[    0.000000] free_area_init_node: node 0, pgdat c0a7f800, node_mem_map d0000000
[    0.000000]   DMA zone: 512 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 65024 pages, LIFO batch:15
[    0.000000]   Normal zone: 1008 pages used for memmap
[    0.000000]   Normal zone: 128016 pages, LIFO batch:31
[    0.000000]   HighMem zone: 528 pages used for memmap
[    0.000000]   HighMem zone: 67056 pages, LIFO batch:15
[    0.000000] PERCPU: Embedded 8 pages/cpu @d0808000 s8256 r8192 d16320 u32768
[    0.000000] pcpu-alloc: s8256 r8192 d16320 u32768 alloc=8*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 260096
[    0.000000] Kernel command line: console=ttyS0,115200 console=tty1 board=bpi-m1 crypedevice=/dev/mmcblk0p2:cryptroot root=/dev/mapper/cryptroot rootwait rootfstype=ext4 cgroup_enable=memory swapaccount=1 hdmi.audio=EDID:0 disp.screen0_output_mode=1280x720p60 panic=10 consoleblank=0 enforcing=0 loglevel=8
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] allocated 2097152 bytes of page_cgroup
[    0.000000] please try 'cgroup_disable=memory' option if you don't want memory cgroups
[    0.000000] Memory: 1024MB = 1024MB total
[    0.000000] Memory: 885372k/885372k available, 163204k reserved, 270336K highmem
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
[    0.000000]     vmalloc : 0xf0000000 - 0xff000000   ( 240 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xef800000   ( 760 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0008000 - 0xc09d5bc0   (10039 kB)
[    0.000000]       .init : 0xc09d6000 - 0xc0a0e040   ( 225 kB)
[    0.000000]       .data : 0xc0a10000 - 0xc0a880e0   ( 481 kB)
[    0.000000]        .bss : 0xc0a88104 - 0xc0c5d4f8   (1877 kB)
[    0.000000] SLUB: Genslabs=11, HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000]  RCU dyntick-idle grace-period acceleration is enabled.
[    0.000000]  Additional per-CPU info printed with stalls.
[    0.000000] NR_IRQS:192
[    0.000000] Architected local timer running at 24.00MHz.
[    0.000000] sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every 178956ms
[    0.000000] start_kernel(): bug: interrupts were enabled early
[    0.000000] Console: colour dummy device 80x30
[    0.000000] console [tty1] enabled
[    0.013863] Calibrating delay loop... 1816.98 BogoMIPS (lpj=9084928)
[    0.076013] pid_max: default: 32768 minimum: 301
[    0.079708] Mount-cache hash table entries: 512
[    0.083961] Initializing cgroup subsys cpuacct
[    0.087302] Initializing cgroup subsys memory
[    0.090736] Initializing cgroup subsys devices
[    0.094186] Initializing cgroup subsys freezer
[    0.097421] Initializing cgroup subsys blkio
[    0.101107] Initializing cgroup subsys perf_event
[    0.104701] CPU: Testing write buffer coherency: ok
[    0.109910] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.117027] hw perfevents: enabled with ARMv7 Cortex-A7 PMU driver, 5 counters available
[... 986 lines cut ...]
[   86.107282] HTB: quantum of class 10247 is small. Consider r2q change.
[   86.137749] HTB: quantum of class 20247 is small. Consider r2q change.
[   86.175810] HTB: quantum of class 10248 is small. Consider r2q change.
[   86.202631] HTB: quantum of class 20248 is small. Consider r2q change.
[   86.238398] HTB: quantum of class 10249 is small. Consider r2q change.
[   86.268581] HTB: quantum of class 20249 is small. Consider r2q change.
[   86.304571] HTB: quantum of class 10250 is small. Consider r2q change.
[   86.337400] HTB: quantum of class 20250 is small. Consider r2q change.
[   86.373022] HTB: quantum of class 10251 is small. Consider r2q change.
[   86.401959] HTB: quantum of class 20251 is small. Consider r2q change.
[   86.437161] HTB: quantum of class 10252 is small. Consider r2q change.
[   86.464649] HTB: quantum of class 20252 is small. Consider r2q change.
[   86.505467] HTB: quantum of class 10253 is small. Consider r2q change.
[... cut ...]
[  530.409788] hub 5-0:1.0: port 1 disabled by hub (EMI?), re-enabling...
[  530.413808] usb 5-1: USB disconnect, device number 2
[  530.420623] usb 5-1: pl2303_read_int_callback - usb_submit_urb failed with result -19
[  530.666674] pl2303 ttyUSB0: pl2303 converter now disconnected from ttyUSB0
[  530.670750] pl2303 5-1:1.0: device disconnected
[  531.014372] usb 5-1: new full-speed USB device number 3 using sw-ohci
[  531.235443] pl2303 5-1:1.0: pl2303 converter detected
[  531.266316] usb 5-1: pl2303 converter now attached to ttyUSB0

alistair23, 2019-02-27:
You didn't need to include your entire dmesg. It looks like the device should be appearing under ttyUSB0. Can you not see /dev/ttyUSB0 at all?

makarg1001, 2019-02-27:
Sorry for the entire message. I can see the /dev/ttyUSB0 both before and after running the script obd_recorder.py. 

Debug_Output for obd_recorder.py

Logging item: Engine RPM
Logging item: Vehicle Speed
Logging item: Throttle Position
Logging item: Calc Load Value
Logging item: Fuel System Stat
Logging item: Supported PIDs
Logging item: DTC C-F-F
Logging item: S-S DTC Cleared
Logging item: Engine Start MIN
['/dev/ttyUSB0']
Opening interface (serial port)
could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
Not connected

after this I checked and I am able to see /dev/ttyUSB0

alistair23, 2019-02-27:
That is very strange. Can you connect to it with a serial terminal?

makarg1001, 2019-02-27:
I am sorry, I am really new to this. Still a work in progress. I do not know how to connect to it with a serial terminal, if you can explain it breifly it'd be really helpful. 
What is strange to me is that when I don't connect the OBD2 USB dongle to the vehicle(just to the banana pi), the port opens fine and the logging starts when I run obd_recorder.py. Obviously it is unable to record any data in the file as it is not connected to the vehicle. Now, when I connect it to the vehicle it starts to give an exception that is mentioned above. Is it possible that a port is opened keeping in check some other parameters as well which change when I connect to the vehicle?
Also, the exception is caught on the following call
port = serial.Serial('/dev/ttyUSB0', 38400, \
parity = serial.PARITY_NONE, stopbits = 1, bytesize = 8, timeout = SERTIMEOUT )

alistair23, 2019-02-28:
The port maybe in use. Can you ensure that nothing else is running that is using that port.

You can use screen to connect to the serial device. Search around for how to use it if you are unsure how to connect.

makarg1001, 2019-02-28:
Okay. I'll try that. I tried connecting to the serial device using screen. I used 
$ sudo screen /dev/ttyUSB0
The screen always terminated with the exception :
"/var/run/utmp: No such file or directory
[screen is terminating]"
I'll check if the port is in use and then try to connect it using screen. 
Thanks

alistair23, 2019-02-28:
A quick web search found this: https://www.shan.info/item/451-screen-var-run-utmp-no-such-file-or-directory-cannot-find-termcap-entry-for.html

If that doesn't work try searching the error yourself and see if you can fix it.

alistair23, 2019-03-03:
Let me know if it still isn't fixed

closed 2019-03-03 by alistair23

gayathri191855, 2022-03-31:
FileNotFoundError: [Errno 2] No such file or directory: '/dev/ttyUSB1'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/user/.local/bin/lxterm", line 11, in <module>
    load_entry_point('litex', 'console_scripts', 'lxterm')()
  File "/home/user/litex/litex/tools/litex_term.py", line 648, in main
    term.open(port, int(float(args.speed)))
  File "/home/user/litex/litex/tools/litex_term.py", line 321, in open
    self.port = serial.serial_for_url(port, baudrate)
  File "/home/user/.local/lib/python3.8/site-packages/pyserial-3.5-py3.8.egg/serial/__init__.py", line 90, in serial_for_url
    instance.open()
  File "/home/user/.local/lib/python3.8/site-packages/pyserial-3.5-py3.8.egg/serial/serialposix.py", line 325, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/ttyUSB1: [Errno 2] No such file or directory: '/dev/ttyUSB1'

gayathri191855, 2022-03-31:
let me know how to fix the below error

FileNotFoundError: [Errno 2] No such file or directory: '/dev/ttyUSB1'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/user/.local/bin/lxterm", line 11, in <module>
    load_entry_point('litex', 'console_scripts', 'lxterm')()
  File "/home/user/litex/litex/tools/litex_term.py", line 648, in main
    term.open(port, int(float(args.speed)))
  File "/home/user/litex/litex/tools/litex_term.py", line 321, in open
    self.port = serial.serial_for_url(port, baudrate)
  File "/home/user/.local/lib/python3.8/site-packages/pyserial-3.5-py3.8.egg/serial/__init__.py", line 90, in serial_for_url
    instance.open()
  File "/home/user/.local/lib/python3.8/site-packages/pyserial-3.5-py3.8.egg/serial/serialposix.py", line 325, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/ttyUSB1: [Errno 2] No such file or directory: '/dev/ttyUSB1'

bytedicer, 2023-12-09:
2nd that;
user@pc:~/Downloads$ esptool --port /dev/ttyUSB0 --chip esp32 --before default_reset --after hard_reset write_flash --flash_mode dout --flash_freq 40m --flash_size detect   0x0 opendtu-generic.factory.bin
esptool.py v2.8
Serial port /dev/ttyUSB0
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 322, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/ttyUSB0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/esptool", line 3101, in <module>
    _main()
  File "/usr/bin/esptool", line 3094, in _main
    main()
  File "/usr/bin/esptool", line 2891, in main
    esp = chip_class(each_port, initial_baud, args.trace)
  File "/usr/bin/esptool", line 237, in __init__
    self._port = serial.serial_for_url(port)
  File "/usr/lib/python3/dist-packages/serial/__init__.py", line 90, in serial_for_url
    instance.open()
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 325, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'

devkrish23, 2024-07-02:
I am facing the same issue. Is there any solution? Why is this issue closed without any solution?