brendan-w/python-OBD #143: How to simulate obd II ?

BedoM, 2019-03-16:
Thank you very much for your work.
We want to use your solution but must it necessarily be connected to the car to execute and test our code?

Which solution do you use?

thank you and good evening

alistair23, 2019-03-17:
You will need an OBD2 Simulator or Emulator. You can look around online for different versions at different price points. A contributor to this project has written this fork: https://github.com/Ircama/ELM327-emulator

alistair23, 2019-03-17:
Obviously you could also just connect to a car, but you might need to test against multiple cars with different protocols.

BedoM, 2019-03-17:
thank you for your answer, there is no software emulator sooner than hardware?
because we are mounting a student project and it is true that test the code on our car slowed the  project. 

Do you have an idea?

Ircama, 2019-03-17:
*ELM327-emulator* allows simulating different cars and can handle different use cases including hardware failure. It also includes an auxiliary feature that builds the PID dictionary of a specific vehicle by automatically querying all standard PIDs one by one (as well as additional custom PIDs specified by the user via CSV file). The dictionary can then be used to emulate the specific car.

alistair23, 2019-03-21:
Does that work for you?

Ircama, 2019-03-21:
> Does that work for you?

Hopefully. Please open an issue if you found a bug.

BedoM, 2019-03-21:
It's amazing it works well thank you for your help I will not hesitate to post

Ircama, 2019-03-21:
> It's amazing it works well thank you for your help I will not hesitate to post

Good. It would be useful to share dictionaries produced by distinct cars (maybe including custom PIDs), especially when dealing with different protocol than ISO 15765-4. This will surely help testing python-OBD (like usage of custom headers).

closed 2019-03-23 by alistair23