brendan-w/python-OBD #149: Asynchronous Request results in null values after x Time

skast96, 2019-05-08:
Hey,
I got an error with the asynchronous OBD-II Connection which results in null values or an error after a random amount of time. I can´t exactly tell if it´s my OBD-II Bluetooth Device or the Lib itself, so I wanted to ask if some of you got the same behavior. 

The Code which I use is the attached file.
[image]
[image]

alistair23, 2019-05-13:
Can you attach the logs? Also please don't attach pictures, just insert the code directly, it's much easier to read that way.

patfelst, 2019-07-17:
Yes I am getting the same thing on a RPi 3B+, after a random amount of time, query commands begin to return null. I have reportedly best bluetooth adaptor (OBDLink MX Bluetooth), not a cheap one. I will try and get some logs to attach.

patfelst, 2019-07-18:
Hi again. It seems there is a problem with the bluetooth serial port profile (SPP) on my Raspberry Pi. The reason I say this is if I run my python program on my mac laptop (with the bluetooth dongle paired with mac) the program - and this fantastic OBD library - works perfectly. So I need to start with a fresh RPi image (now there is Rasbian 'buster' I was on 'Stretch') as I had installed the blueman utility and some other bluetooth tweaks which I believe may be the problem.

For what it's worth, on the RPi, it works ok for around 30 seconds, sometimes I get a minute, then it just hangs. Sometimes I can see on the RPi desktop in the taskbar, that the bluetooth connection is dropped, other times it says it is still connected, but the program still hangs when waiting for a response from the OBD port. Here is the debug log, scroll to the bottom to see the hang.
[obd_stall_log.txt](https://github.com/brendan-w/python-OBD/files/3406024/obd_stall_log.txt)

alistair23, 2019-07-18:
Hmm... I used an embedded ARM board (not the RaspberryPi) and never have any issues, although I also don't use Bluetooth. I think a clean install is a good idea, keep this issue updated with what you find out.

patfelst, 2019-07-21:
The fresh install made no difference, still drops bluetooth SPP after about 30 seconds. I also have a wired (USB) OBD2 adaptor, and this works perfectly, all I had to do was change the connection object with "fast=False". All other code is identical. I've had it working for over 30 minutes with no problem. Looks like Raspberry Pi has serious bluetooth issues.

alistair23, 2019-07-21:
I'm not sure how to help you unfortunately. It sounds like it is a Bluetooth issue with the Pi. Maybe you can check on the Raspberry Pi Forums to see if there are any solutions?

alistair23, 2019-07-23:
Can you try this: https://github.com/brendan-w/python-OBD/issues/155#issuecomment-514302614

patfelst, 2019-07-24:
Yes, give me a couple of days Alistair. I already tried setting Fast=False, but I can try a larger timeout.

Jitnaught, 2019-07-29:
I just tried it for ~5 minutes and it seemed to fix the problem! I only added fast=False and timeout=30; I did not enable debug.

alistair23, 2019-07-30:
Great! Does someone want to send a PR adding this to the README?

alistair23, 2019-07-31:
I have created a PR to document the issue/fix. Let me know if you have any comments on it: https://github.com/brendan-w/python-OBD/pull/156

patfelst, 2019-08-02:
I finally tried with Fast=False, Timeout=30 and with debug on. Still hangs after about 30 seconds on bluetooth adaptor. What type of ARM SBC and OS do you use Alistair?

alistair23, 2019-08-05:
I use a sopine board from Pine64. I am running a Yocto distro with the 5.2 kernel.

patfelst, 2019-08-11:
The Pine SBCs look good, and well supported with English speaking support / documentation. 

Just as a bit of closure on this [issue](https://github.com/brendan-w/python-OBD/issues/149#issuecomment-512084808), after many hours troubleshooting, I have finally found a solution to the random hanging serial port using the OBD2 MX+ bluetooth adaptor. I had been running it on a Raspberry Pi 3B. I tried on a 3B+ and it works flawlessly with identical code. I suspect there is a different bluetooth module, and/or different bluetooth driver for the 3B. Anyway, I'm happy and can now continue my project with Bluetooth!

alistair23, 2019-08-11:
Yeah, they are pretty good boards. You can use a mainline kernel which is really nice as well.

Ok, so it sounds like the Raspberry Pi 3B Bluetooth has an issue. When you got it working you mean with `fast=False` and `timeout=30`?

patfelst, 2019-08-11:
Initially yes I had `fast=False` and `timeout=30` but when I saw it working, I set `fast=True`, and removed the timeout so it was using the default timeout. I saw the frame rate increase from 10fps up to 40fps!

alistair23, 2019-08-11:
So the Pi 3B+ just works? I thought earlier you said it also didn't work on a 3B+

patfelst, 2019-08-11:
oops, that was a mistake. I double checked it during my troubleshooting and it was definitely a 3B that didn't work. I'll see if I can edit the first post to fix that.