brendan-w/python-OBD #150: [obd.elm327] ATH1 did not return 'OK', or echoing is still ON

luoluo4413, 2019-05-20:
Hi,
    I have use the obd version of 0.7.1,But it has the error info ,Please Help!
Step1:
pi@raspberrypi:~ $ sudo rfcomm connect hci0 00:0D:18:3A:67:89
Connected /dev/rfcomm0 to 00:0D:18:3A:67:89 on channel 1
Press CTRL-C for hangup
 00:0D:18:3A:67:89 , is my ELM327 BlueTooth device's address

Step2:
pi@raspberrypi:~ $ python
Python 2.7.9 (default, Sep 26 2018, 05:58:52) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import obd
>>> obd.logger.setLevel(obd.logging.DEBUG)
>>> obd.OBD()
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/rfcomm0']
[obd.obd] Attempting to use port: /dev/rfcomm0
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: '\x7f\x7f\r?\r\r>'
[obd.elm327] Choosing baud 38400
[obd.elm327] write: 'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'ATZ\r\r\rELM327 v1.5\r\r>'
[obd.elm327] write: 'ATE0\r'
[obd.elm327] read: b'ATE0\rOK'
[obd.elm327] write: 'ATH1\r'
[obd.elm327] read: b'\r\r>'
[obd.elm327] closing port
[obd.elm327] write: 'ATZ\r'
[obd.elm327] ATH1 did not return 'OK', or echoing is still ON
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
<obd.obd.OBD object at 0x76a86910>

Is there something with my operation?

alistair23, 2019-05-20:
There is no ELM327 v1.5 (see page 92: https://www.elmelectronics.com/wp-content/uploads/2017/01/ELM327DS.pdf). You have a knock off device that appears to not be responding correctly. I'm not sure how to help you here, your best bet is probably to return the device and get one with a legitimate ELM327 chip. Otherwise you will have to work on reverse engineering the fake chip and figure out how it's responses differ to the real one and make local changes.

luoluo4413, 2019-05-21:
Thanks for you reply.I have change to another device.Here is the debug info.Looks like the same"[obd.elm327] ATH1 did not return 'OK', or echoing is still ON"。
HELP!

>>> obd.OBD()
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/rfcomm0']
[obd.obd] Attempting to use port: /dev/rfcomm0
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: '?\r\r>'
[obd.elm327] Choosing baud 38400
[obd.elm327] write: 'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'\r\rELM327 v2.1\r\r>'
[obd.elm327] write: 'ATE0\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: 'ATH1\r'
[obd.elm327] read: b'\r\r>'
[obd.elm327] closing port
[obd.elm327] write: 'ATZ\r'
[obd.elm327] ATH1 did not return 'OK', or echoing is still ON
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
<obd.obd.OBD object at 0x75767c70>

luoluo4413, 2019-05-21:
> There is no ELM327 v1.5 (see page 92: https://www.elmelectronics.com/wp-content/uploads/2017/01/ELM327DS.pdf). You have a knock off device that appears to not be responding correctly. I'm not sure [...]

I have changed to another device.Looks like the same.

alistair23, 2019-05-21:
Hmm... That is strange. Is the connection still active? Can you see if you get any reply?

luoluo4413, 2019-05-21:
Nope,It still tell me "Query failed, no connection available".See the detail below.
pi@raspberrypi:~ $ cat myobdtest.py 
import obd
obd.logger.setLevel(obd.logging.DEBUG)
connection = obd.OBD()
c = obd.commands.SPEED
response = connection.query(c)
print (response.value)
connection.close()
pi@raspberrypi:~ $ python myobdtest.py 
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: []
[obd.obd] No OBD-II adapters found
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
[obd.obd] Query failed, no connection available
None
pi@raspberrypi:~ $

luoluo4413, 2019-05-21:
Nope,It still the same.See the detail info below.

pi@raspberrypi:~ $ cat myobdtest.py 
import obd
obd.logger.setLevel(obd.logging.DEBUG)
connection = obd.OBD()
c = obd.commands.SPEED
response = connection.query(c)
print (response.value)
connection.close()
pi@raspberrypi:~ $ python myobdtest.py 
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/rfcomm0']
[obd.obd] Attempting to use port: /dev/rfcomm0
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: '?\r\r>'
[obd.elm327] Choosing baud 38400
[obd.elm327] write: 'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'\r\rELM327 v2.1\r\r>'
[obd.elm327] write: 'ATE0\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: 'ATH1\r'
[obd.elm327] read: b'>'
[obd.elm327] closing port
[obd.elm327] write: 'ATZ\r'
[obd.elm327] ATH1 did not return 'OK', or echoing is still ON
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
[obd.obd] Query failed, no connection available
None
pi@raspberrypi:~ $ python
Python 2.7.9 (default, Sep 26 2018, 05:58:52) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import obd
>>> obd.logger.setLevel(obd.logging.DEBUG)
>>> obd.OBD()
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/rfcomm0']
[obd.obd] Attempting to use port: /dev/rfcomm0
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: ''
[obd.elm327] Response from baud 9600: '?\r\r>?\r\r>'
[obd.elm327] Choosing baud 9600
[obd.elm327] write: 'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'\r\rELM327 v2.1\r\r>'
[obd.elm327] write: 'ATE0\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: 'ATH1\r'
[obd.elm327] read: b'>'
[obd.elm327] closing port
[obd.elm327] write: 'ATZ\r'
[obd.elm327] ATH1 did not return 'OK', or echoing is still ON
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
<obd.obd.OBD object at 0x76a36910>
>>>

luoluo4413, 2019-05-21:
[image]
The bluetooth status
pi@raspberrypi:~ $ sudo service bluetooth status
¡ñ bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled)
   Active: active (running) since Tue 2019-05-21 04:49:33 UTC; 14min ago
     Docs: man:bluetoothd(8)
 Main PID: 729 (bluetoothd)
   Status: "Running"
   CGroup: /system.slice/bluetooth.service
           ©¸©¤729 /usr/lib/bluetooth/bluetoothd --noplugin=sap

May 21 04:49:33 raspberrypi bluetoothd[729]: Excluding (cli) sap
May 21 04:49:33 raspberrypi bluetoothd[729]: Bluetooth management interface 1.14 initialized
May 21 04:57:49 raspberrypi bluetoothd[729]: Endpoint registered: sender=:1.17 path=/MediaEndpoint/A2DPSource
May 21 04:57:49 raspberrypi bluetoothd[729]: Endpoint registered: sender=:1.17 path=/MediaEndpoint/A2DPSink
May 21 04:57:51 raspberrypi bluetoothd[729]: Endpoint unregistered: sender=:1.17 path=/MediaEndpoint/A2DPSource
May 21 04:57:51 raspberrypi bluetoothd[729]: Endpoint unregistered: sender=:1.17 path=/MediaEndpoint/A2DPSink
May 21 04:58:03 raspberrypi bluetoothd[729]: Endpoint registered: sender=:1.22 path=/MediaEndpoint/A2DPSource
May 21 04:58:03 raspberrypi bluetoothd[729]: Endpoint registered: sender=:1.22 path=/MediaEndpoint/A2DPSink
May 21 04:58:04 raspberrypi bluetoothd[729]: Endpoint unregistered: sender=:1.22 path=/MediaEndpoint/A2DPSource
May 21 04:58:04 raspberrypi bluetoothd[729]: Endpoint unregistered: sender=:1.22 path=/MediaEndpoint/A2DPSink
pi@raspberrypi:~ $

luoluo4413, 2019-05-21:
> Hmm... That is strange. Is the connection still active? Can you see if you get any reply?

Nope,It still the same.See the detail info below.

pi@raspberrypi:~ $ cat myobdtest.py
import obd
obd.logger.setLevel(obd.logging.DEBUG)
connection = obd.OBD()
c = obd.commands.SPEED
response = connection.query(c)
print (response.value)
connection.close()
pi@raspberrypi:~ $ python myobdtest.py
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/rfcomm0']
[obd.obd] Attempting to use port: /dev/rfcomm0
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: '?\r\r>'
[obd.elm327] Choosing baud 38400
[obd.elm327] write: 'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'\r\rELM327 v2.1\r\r>'
[obd.elm327] write: 'ATE0\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: 'ATH1\r'
[obd.elm327] read: b'>'
[obd.elm327] closing port
[obd.elm327] write: 'ATZ\r'
[obd.elm327] ATH1 did not return 'OK', or echoing is still ON
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
[obd.obd] Query failed, no connection available
None
pi@raspberrypi:~ $ python
Python 2.7.9 (default, Sep 26 2018, 05:58:52)
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.

import obd
obd.logger.setLevel(obd.logging.DEBUG)
obd.OBD()
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/rfcomm0']
[obd.obd] Attempting to use port: /dev/rfcomm0
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: ''
[obd.elm327] Response from baud 9600: '?\r\r>?\r\r>'
[obd.elm327] Choosing baud 9600
[obd.elm327] write: 'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'\r\rELM327 v2.1\r\r>'
[obd.elm327] write: 'ATE0\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: 'ATH1\r'
[obd.elm327] read: b'>'
[obd.elm327] closing port
[obd.elm327] write: 'ATZ\r'
[obd.elm327] ATH1 did not return 'OK', or echoing is still ON
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
<obd.obd.OBD object at 0x76a36910>

luoluo4413, 2019-05-23:
I got problem solved,I have changed all connection AT commands in "obd\elm327.py" file,With add "delay=1".Such as:
orginal: r = self.__send(b"ATH1")
changed to: r = self.__send(b"ATH1", delay=1)
When it's done,The debug infomation like this below:

pi@raspberrypi:~ $ cat hud.py 
import obd
obd.logger.setLevel(obd.logging.DEBUG)
connection = obd.OBD()
c = obd.commands.SPEED
response = connection.query(c)
print (response.value)
connection.close()
pi@raspberrypi:~ $ python hud.py 
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/rfcomm0']
[obd.obd] Attempting to use port: /dev/rfcomm0
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: '?\r\r>'
[obd.elm327] Choosing baud 38400
[obd.elm327] write: 'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'\r\rELM327 v2.1\r\r>'
[obd.elm327] write: 'ATE0\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: 'ATH1\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: 'ATL0\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: 'AT RV\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'12.1V\r\r>'
[obd.elm327] write: 'ATSP0\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: '0100\r'
[obd.elm327] read: b'SEARCHING...\r7E8 07 41 00 FF FF FF FF FF \r\r>'
[obd.elm327] write: 'ATDPN\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'6\r\r>'
[obd.protocols.protocol] map ECU 0 --> ENGINE
[obd.elm327] Connected Successfully: PORT=/dev/rfcomm0 BAUD=38400 PROTOCOL=6
[obd.obd] querying for supported commands
[obd.obd] Sending command: 0100: Supported PIDs [01-20]
[obd.elm327] write: '0100\r'
[obd.elm327] read: b'7E8 07 41 00 FF FF FF FF FF \r\r>'
[obd.OBDCommand] Message was longer than expected (7>6). Trimmed message: bytearray(b'A\x00\xff\xff\xff\xff')
[obd.obd] Sending command: 0120: Supported PIDs [21-40]
[obd.elm327] write: '0120\r'
[obd.elm327] read: b'7E8 07 41 20 FF FF FF FF FF \r\r>'
[obd.OBDCommand] Message was longer than expected (7>6). Trimmed message: bytearray(b'A \xff\xff\xff\xff')
[obd.obd] Sending command: 0140: Supported PIDs [41-60]
[obd.elm327] write: '0140\r'
[obd.elm327] read: b'7E8 07 41 40 FF FF FF FF FF \r\r>'
[obd.OBDCommand] Message was longer than expected (7>6). Trimmed message: bytearray(b'A@\xff\xff\xff\xff')
[obd.obd] Sending command: 0600: Supported MIDs [01-20]
[obd.elm327] write: '0600\r'
[obd.elm327] read: b'7E8 07 46 FF FF FF FF FF FF \r\r>'
[obd.obd] Sending command: 0620: Supported MIDs [21-40]
[obd.elm327] write: '0620\r'
[obd.elm327] read: b'NO DATA\r\r>'
[obd.obd] finished querying with 218 commands supported
[obd.obd] ===================================================================
[obd.obd] Sending command: 010D: Vehicle Speed
[obd.elm327] write: '010D\r'
[obd.elm327] read: b'7E8 07 41 0D 19 00 00 00 00 \r\r>'
[obd.OBDCommand] Message was longer than expected (7>3). Trimmed message: bytearray(b'A\r\x19')
25.0 kph
[obd.obd] Closing connection
[obd.elm327] closing port
[obd.elm327] write: 'ATZ\r'

Also,It may cause more seconds,I will optimise it.
lol :) :) :)

closed 2019-05-23 by luoluo4413

alistair23, 2019-05-23:
That is still strange. So it sounds like every operation need a delay when using Bluetooth. Maybe that is something worth adding in the future.

clrj14, 2020-01-17:
delay=1 works for me as well. However, it doesn't seem consistent and is hit or miss.

xxEoD2242, 2020-03-08:
Just did this with ELM 1.5 and adding the delay worked.

terrorent, 2020-03-11:
"delay=1" works for my devices.

kshardha, 2020-10-08:
> delay=1 works for me as well. However, it doesn't seem consistent and is hit or miss.

Ensure that you are updating the correct elm327.py file and in correct location. For example, my elm327.py was at '/usr/local/lib/python2.7/dist-packages/obd/elm327.py ' and '/home/pi/.local/lib/python2.7/site-packages/obd/elm327.py'. My python env was using the latter path.

atarbabgei, 2020-12-21:
Hi,

I have a fake ELM327 v1.5 USB version, it seems that when I enable the debug logger feature by adding this code
`obd.logger.setLevel(obd.logging.DEBUG)`
before creating a connection seems to fix "ATH1 did not return 'OK', or echoing is still ON" problem for me.

After getting sucessful connection then I disable the debug logger by adding this code
`obd.logger.removeHandler(obd.console_handler)`

ryanmjacobs, 2021-01-03:
I just ran into this issue. I edited my `elm327.py` file to add a delay of 100ms and it worked like a charm.

I think a nice solution would be to automatically detect the `v1.5` knockoff version, and automatically add a 100ms delay.

```
    def __send(self, cmd, delay=None):
        self.__write(cmd)

        # ADD THIS LINE
        if delay is None:
            delay = 0.1

        if delay is not None:
            logger.debug("wait: %d seconds" % delay)
            time.sleep(delay)

        return self.__read()
```

alistair23, 2021-01-03:
Please feel free to send a PR to discuss it there.

I'm a little hesitant to support knock offs. The problem is what if all the knock offs are the same? We might have a combination of workarounds that all only apply to some.