brendan-w/python-OBD #155: Supported PIDs [01-20] did not receive any acceptable messages

sebastianofranceschin, 2019-07-22:
Hi everyone,
I'm trying to use the library to connect a bluetooth obd adapter to a Raspberry Pi, but i had no luck so far since most of the commands return no value. I've checked if the OBD scanner works (and it does) by connecting it to my smartphone and downloading a random android app (i can see rpm, speed and so on in real time) but it just won't work with the python library. I've tried manually setting every protocol / baud rate but it didn't make a difference. I've tried disabling fast method and also setting force=True on the query.
Here's the log for my last attempt
```
 [obd.obd] ======================= python-OBD (v0.7.1) =======================
    [obd.obd] Using scan_serial to select port
    [obd.obd] Available ports: ['/dev/rfcomm0']
    [obd.obd] Attempting to use port: /dev/rfcomm0
    [obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=auto PROTOCOL=8
    [obd.elm327] Response from baud 38400: '\x7f\x7f\r?\r\r>'
    [obd.elm327] Choosing baud 38400
    [obd.elm327] write: 'ATZ\r'
    [obd.elm327] wait: 1 seconds
    [obd.elm327] read: b'ATZ\r\xfc\r\rELM327 v1.5\r\r>'
    [obd.elm327] write: 'ATE0\r'
    [obd.elm327] read: b'ATE0\rOK'
    [obd.elm327] write: 'ATH1\r'
    [obd.elm327] read: b'OK'
    [obd.elm327] write: 'ATL0\r'
    [obd.elm327] read: b'OK'
    [obd.elm327] write: 'AT RV\r'
    [obd.elm327] read: b'14.0V\r\r>'
    [obd.elm327] write: 'ATTP8\r'
    [obd.elm327] read: b'OK\r\r'
    [obd.elm327] write: '0100\r'
    [obd.elm327] read: b'>'
    [obd.elm327] Connected Successfully: PORT=/dev/rfcomm0 BAUD=38400 PROTOCOL=8
    [obd.obd] querying for supported commands
    [obd.obd] Sending command: 0100: Supported PIDs [01-20]
    [obd.elm327] write: '0100\r'
    [obd.elm327] read: b'NO DATA\r\r>'
    [obd.OBDCommand] 0100: Supported PIDs [01-20] did not receive any acceptable messages
    [obd.obd] No valid data for PID listing command: 0100: Supported PIDs [01-20]
    [obd.obd] Sending command: 0600: Supported MIDs [01-20]
    [obd.elm327] write: '0600\r'
    [obd.elm327] read: b'NO DATA\r\r>'
    [obd.obd] finished querying with 7 commands supported
    [obd.obd] ===================================================================
    Car Connected
    [obd.obd] Sending command: 010C: Engine RPM
    [obd.elm327] write: '010C\r'
    [obd.elm327] read: b'NO DATA\r\r>'
    [obd.OBDCommand] 010C: Engine RPM did not receive any acceptable messages
```
If it's of any help to the issue, i can also attache the log for the smartphone app...

Thanks for your time!

alistair23, 2019-07-22:
This looks like a duplicate of https://github.com/brendan-w/python-OBD/issues/149

It seems the RaspberryPi Bluetooth is just broken

sebastianofranceschin, 2019-07-22:
Thanks for the feedback!
if it were to be a Bluetooth problem, shouldn't all commands return the same problem? i do get a valid response from elm_voltage...

alistair23, 2019-07-22:
It's possible that just some data is dropped, I'm not too sure here.

sebastianofranceschin, 2019-07-23:
HI,
thanks you very much for your time.
This morning i made another attempt and got a slightly different response, here's the log:
```
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/rfcomm0']
[obd.obd] Attempting to use port: /dev/rfcomm0
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: '?\r\r>'
[obd.elm327] Choosing baud 38400
[obd.elm327] write: 'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'\xfc\r\rELM327 v1.5\r\r>'
[obd.elm327] write: 'ATE0\r'
[obd.elm327] read: b'ATE0\rOK'
[obd.elm327] write: 'ATH1\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: 'ATL0\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: 'AT RV\r'
[obd.elm327] read: b'14.2V\r\r>'
[obd.elm327] write: 'ATSP0\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: '0100\r'
[obd.elm327] read: b'\r\r>'
[obd.elm327] write: 'ATDPN\r'
[obd.elm327] read: b'A0\r\r>'
[obd.elm327] ELM responded with unknown protocol. Trying them one-by-one
[obd.elm327] write: 'ATTP6\r'
[obd.elm327] read: b'OK\r\r'
[obd.elm327] write: '0100\r'
[obd.elm327] read: b'>'
[obd.elm327] Connected Successfully: PORT=/dev/rfcomm0 BAUD=38400 PROTOCOL=6
[obd.obd] querying for supported commands
[obd.obd] Sending command: 0100: Supported PIDs [01-20]
[obd.elm327] write: '0100\r'
[obd.elm327] read: b'7E8 06 41 00 BE 3E B8 11 \r\r>'
[obd.OBDCommand] 0100: Supported PIDs [01-20] did not receive any acceptable messages
[obd.obd] No valid data for PID listing command: 0100: Supported PIDs [01-20]
[obd.obd] Sending command: 0600: Supported MIDs [01-20]
[obd.elm327] write: '0600\r'
[obd.elm327] read: b'7E8 06 46 00 C0 00 00 01 \r\r>'
[obd.obd] Sending command: 0620: Supported MIDs [21-40]
[obd.elm327] write: '0620\r'
[obd.elm327] read: b'7E8 06 46 20 80 00 00 00 \r\r>'
[obd.obd] finished querying with 11 commands supported
[obd.obd] ===================================================================
Car Connected
[obd.obd] Sending command: 010C: Engine RPM
[obd.elm327] write: '010C\r'
[obd.elm327] read: b'7E8 04 41 0C 10 10 \r\r>'
[obd.OBDCommand] 010C: Engine RPM did not receive any acceptable messages
```
Do you think it's still a bluetooth related problem? if so, is there anything particular i should check for when buying a usb scanner?

alistair23, 2019-07-23:
> Do you think it's still a bluetooth related problem? if so, is there anything particular i should check when buying a usb scanner?

Just make sure it's a legit ELM chip in there and not a knockoff

> confronting my results with the [troubleshooting examples](https://python-obd.readthedocs.io/en/latest/Troubleshooting/), i see that my responses end with the '>' symbol, while in the examples the '>' symbol is not present... could that be the problem?

Maybe, but I'm not sure why you are getting the extra symbol. You could maybe add some delays or something and see if there are just slowdowns over the connection?

sebastianofranceschin, 2019-07-23:
Well, I've got It working by setting fast=False, timout=30 and debug log enabled, if I comment out the line that enables the console debug It stops working tho...

alistair23, 2019-07-31:
I have created a PR to document the issue/fix. Let me know if you have any comments on it: https://github.com/brendan-w/python-OBD/pull/156

closed 2019-07-31 by alistair23