brendan-w/python-OBD #158: AT L0 wont work

ghostcat17, 2019-10-05:
We are trying to connect to the car. It seems that it can't even connect to the car. Everytime I run this code: 
import obd
obd.logger.setLevel(obd.logging.DEBUG)
ports = obd.scan_serial()       # return list of valid USB or RF ports
print(ports)
connection = obd.OBD() # auto-connects to USB or RF port

cmd = obd.commands.RPM # select an OBD command (sensor)

response = connection.query(cmd) # send the command, and parse the response
ports = obd.scan_serial()
print(ports)
print(response.value) # returns unit-bearing values thanks to Pint
print(response.value.to("mph")) # user-friendly unit conversions

The code stops everytime at the ATL0 it throws an error saying:
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['\\.\\COM3', '\\.\\COM4']
[obd.obd] Attempting to use port: \.\COM3
[obd.elm327] Initializing ELM327: PORT=\.\COM3 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: b'\x7f\x7f\r?\r\r>'
[obd.elm327] Choosing baud 38400
[obd.elm327] write: b'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'ATZ\r\r\rELM327 v1.5\r\r>'
[obd.elm327] write: b'ATE0\r'
[obd.elm327] read: b'ATE0\rOK\r\r>'
[obd.elm327] write: b'ATH1\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: b'ATL0\r'
[obd.elm327] read: b'\r\r>'
[obd.elm327] closing port
[obd.elm327] write: b'ATZ\r'
[obd.elm327] ATL0 did not return 'OK'
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
[obd.obd] Query failed, no connection available
['\\.\\COM3', '\\.\\COM4']
Traceback (most recent call last):
None
  File "C:/Users/justi/PycharmProjects/vobyus/OBDConnection.py", line 13, in <module>
    print(response.value.to("mph")) # user-friendly unit conversions
AttributeError: 'NoneType' object has no attribute 'to'

Could anyone give me some suggestions

Thanks

alistair23, 2019-10-08:
There is no ELM v1.5, so you have a knock off chip. That is probably the issue. you can try with Fast disabled and increase the timeout, maybe that will help it communicate with the car.

ghostcat17, 2019-10-08:
It seems to be working on Jupyter Notebook/Anaconda but it won't work on Pycharm so would this still be a problem with the chip?

alistair23, 2019-10-08:
I am not sure what that is. If it works on one platform but not another I would try changing the time settings.

closed 2020-07-21 by alistair23