brendan-w/python-OBD #160: [obd.elm327] ATH1 did not return 'OK', or echoing is still on

zaneclaes, 2019-10-17:
I am using a new Raspberry Pi 4B with a bluetooth ELM327 dongle. I have been able to successfully use other python obd packages, such as `pyobd-pi` (from which this repo purports to be forked). However, when I try to use `python-obd` I encounter the following error (DEBUG turned on, fwiw). I have also tried using `fast=False, timeout=30` per the README, with the same results.
```
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/rfcomm0']
[obd.obd] Attempting to use port: /dev/rfcomm0
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: '?\r\r>'
[obd.elm327] Choosing baud 38400
[obd.elm327] write: 'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'\r\rELM327 v1.5\r\r>'
[obd.elm327] write: 'ATE0\r'
[obd.elm327] read: b'ATE0\rOK'
[obd.elm327] write: 'ATH1\r'
[obd.elm327] read: b'\r\r>'
[obd.elm327] closing port
[obd.elm327] write: 'ATZ\r'
[obd.elm327] ATH1 did not return 'OK', or echoing is still ON
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
```

alistair23, 2019-10-17:
Can you diff the working output and the non working one?

zaneclaes, 2019-10-18:
@alistair23 I'm not quite sure what you mean. To reiterate: there is only one ELM327 device. The `pyobd-pi` repository works with that device, but the `pythod-obd` (this) repository does not. Are you asking for some kind of logs from the `pyobd-pi` repository?

alistair23, 2019-10-18:
You have one setup that does work `pyobd-pi` and one that doesn't `pythond-obd`. You should run both (not at the same time) and save the debug logs. Then you can compare the logs to see what is different. There must be some difference, if you can find out the difference then you can figure out what is wrong.

zaneclaes, 2019-10-18:
What I don't understand is what "logs" you mean, per my question: "are you asking for some kind of logs from the `pyodb-pi` repository?" Perhaps you're asking for some sort of system logs..?

I achieved the above logs by enabling the `pythod-obd` DEBUG logs, which appear to be specific to this implementation (per the README: `obd.logger.setLevel(obd.logging.DEBUG)`, which appears namespaced to this package). I see no such corresponding documentation for outputting debug logs in the `pyodb-pi` repository, nor am I clear if those are the specific logs you are suggesting to diff.

alistair23, 2019-10-18:
Ah, so `pyobd-pi` doesn't have debug log like `python-obd` does?

In which case that makes things harder. The next step would be to look at the failing code and compare that to the same part in `pyobd-pi`.

ajalberd, 2019-10-22:
Also doesn't work for me, on Arch linux mounted the Bluetooth ELM327 to /dev/rfcomm0 . Even with the fixes above.

zaneclaes, 2019-10-29:
I have a fix open here: https://github.com/brendan-w/python-OBD/pull/162
// cc @alistair23 @ajalberd

alistair23, 2019-11-03:
#162 has been merged, can you try the latest master to see if that fixes the issue?

zaneclaes, 2019-11-14:
Confirmed w/ latest.

closed 2019-11-14 by zaneclaes