brendan-w/python-OBD #161: First value gets read out the following don't, also really slow.

Fernien, 2019-10-20:
Hey, i've got a problem again. Short version of the story: Had a raspi and a obd2 adapter (v1.5 or smth which was aparrently not a right one?) which worked great. It could read out several stats multiple times per second then my raspberrypi died and i moved the sd card to a newer model. Suddenly i couldnt connect anymore execpt addind these delay=1 things in the config which made everything unusable (too slow). Listening to the advice given i bought a new obd2 adapter (v2.1 i think it is) i can connect, it works via screen /dev/rfcomm0 it works the first time in my script (my script: reading multiple values 4 times per second) as well  but after the first value it doesnt output any values anymore. And besides from that its not reading fast like at all. I removed the time.seelp(0.25) from my script to see how fast it could read and i got various results from 1-4 values per second. I don't know what to do to be honest since everything worked fine at the beginning. I tried using my old obd2 adapter, couldnt get it to work again, i tried a brand new raspberrypi 4 with 4gb ram with old and new obd2 adapter still not working. I work on this several days now but im not a professional and i have no ideas left. Im grateful for any help. Maybe someone got the time to (voice) chat with me so i can finally figure out the problem.  Not sure if you need my python script since it worked perfectly fine before. 

`
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/rfcomm0']
[obd.obd] Attempting to use port: /dev/rfcomm0
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=38400 PROTOCOL=auto
[obd.elm327] write: 'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'ELM327 v2.1\r>'
[obd.elm327] write: 'ATE0\r'
[obd.elm327] read: b'OK\r>'
[obd.elm327] write: 'ATH1\r'
[obd.elm327] read: b'OK\r>'
[obd.elm327] write: 'ATL0\r'
[obd.elm327] read: b'OK\r>'
[obd.elm327] write: 'AT RV\r'
[obd.elm327] read: b'12.0V\r>'
[obd.elm327] write: 'ATSP0\r'
[obd.elm327] read: b'OK\r>'
[obd.elm327] write: '0100\r'
[obd.elm327] read: b'SEARCHING...\r7E8 06 41 00 BE 3E A8 13 \r\r>'
[obd.elm327] write: 'ATDPN\r'
[obd.elm327] read: b'A6\r>'
[obd.protocols.protocol] map ECU 0 --> ENGINE
[obd.elm327] Connected Successfully: PORT=/dev/rfcomm0 BAUD=38400 PROTOCOL=6
[obd.obd] querying for supported commands
[obd.obd] Sending command: 0100: Supported PIDs [01-20]
[obd.elm327] write: '0100\r'
[obd.elm327] read: b'7E8 06 41 00 BE 3E A8 13 \r\r>'
[obd.obd] Sending command: 0120: Supported PIDs [21-40]
[obd.elm327] write: '0120\r'
[obd.elm327] read: b'7E8 06 41 20 90 07 B0 01 \r\r>'
[obd.obd] Sending command: 0140: Supported PIDs [41-60]
[obd.elm327] write: '0140\r'
[obd.elm327] read: b'7E8 06 41 40 FE D2 00 11 \r\r>'
[obd.obd] Sending command: 0600: Supported MIDs [01-20]
[obd.elm327] write: '0600\r'
[obd.elm327] read: b'7E8 06 46 00 C0 00 00 01 \r\r>'
[obd.obd] Sending command: 0620: Supported MIDs [21-40]
[obd.elm327] write: '0620\r'
[obd.elm327] read: b'7E8 06 46 20 80 00 08 01 \r\r>'
[obd.obd] Sending command: 0640: Supported MIDs [41-60]
[obd.elm327] write: '0640\r'
[obd.elm327] read: b'7E8 06 46 40 00 00 00 01 \r\r>'
[obd.obd] Sending command: 0660: Supported MIDs [61-80]
[obd.elm327] write: '0660\r'
[obd.elm327] read: b'7E8 06 46 60 00 00 00 01 \r\r>'
[obd.obd] Sending command: 0680: Supported MIDs [81-A0]
[obd.elm327] write: '0680\r'
[obd.elm327] read: b'7E8 06 46 80 80 00 00 01 \r\r>'
[obd.obd] Sending command: 06A0: Supported MIDs [A1-C0]
[obd.elm327] write: '06A0\r'
[obd.elm327] read: b'7E8 06 46 A0 80 00 00 01 \r\r>'
[obd.obd] finished querying with 94 commands supported
[obd.obd] ===================================================================
[obd.obd] Sending command: 010C: Engine RPM
[obd.elm327] write: '010C\r'
[obd.elm327] read: b'7E8 04 41 0C 00 00 \r\r>'
[obd.obd] Sending command: 03: Get DTCs
[obd.elm327] write: '03\r'
[obd.elm327] read: b'7E8 02 43 00 \r\r>'
[obd.obd] Sending command: 012F: Fuel Level Input
[obd.elm327] write: '012F\r'
[obd.elm327] read: b'7E8 03 41 2F 9C \r\r>'
[obd.obd] Sending command: 010C: Engine RPM
[obd.elm327] write: '010C1\r'
[obd.elm327] read: b'?\r>'
[obd.OBDCommand] 010C: Engine RPM did not receive any acceptable messages`

alistair23, 2019-10-20:
So it used to work on an old Pi, but the same SD card with the same adapter (and a new one) on a new Pi doesn't work?

Fernien, 2019-10-20:
> So it used to work on an old Pi, but the same SD card with the same adapter (and a new one) on a new Pi doesn't work?

exactly. Also the first value is working and when connecting to dev/rfcomm0 its working all the time

alistair23, 2019-10-21:
Is it possible it's a hardware issue then? If that's all that changed between it working and not then it seems like the most likely option.

I don't have a Pi, so I can't reproduce this. I'm not really sure how to help.

Fernien, 2019-10-21:
I thought the same, thats why I tried another RaspberryPi. Didn't work either. I tried the RaspberryPi 3B+ and 4 (with 4GB Ram) with both my obd2 adapters. 
But i got different Problems with trying these combinations. I hope its not a hardware problem since i don't know how to figure that out and besides from that i'd be surprised if the newer hardware made problems whilst an older version worked perfectly fine the first time I tried it.

alistair23, 2019-11-09:
Any luck?

Fernien, 2019-11-10:
Not at all. I spent so much time with it now and tried so many things i can't figure out where exactly the problem is. I tried different Raspi's and Linux Versions and Adapters with every combination i was able to do. It gets depressing and expensive at this point since i don't know what to work on anymore. I'm not a developer. I think i have to end my project unless i can find someone who actually can take a look at it and maybe give me a hint or something.

alistair23, 2019-11-10:
Have you tried the latest master? The new read back delay might help.

Fernien, 2019-11-10:
I'm not sure what you mean to be honest.

alistair23, 2019-11-10:
Use the latest source from the git master branch.

Fernien, 2019-11-10:
You gotta help me out with that. I'ts my first time working with git and all this stuff. I got the latest version of the obd2 module installed via pip. If i have to install it directly from github i have to figure out how.