brendan-w/python-OBD #163: OBD - Commands not supported

JakubJD, 2019-11-02:
Hey,

I would like to ask  you for the help. 

I try to communicate with OBD Adapter OBD Link LX101 with Windows 10 and Python 3.7.4.
I wrote very simple code, where I test the communication and data receiving. But few commands are not supported, but I can not imagine why. 
Can you give me the advice?
Thanks
  
**My code is:**

import obd
import time
connection = obd.Async("COM4",baudrate=38400, fast=False)

# a callback that prints every new value to the console
#RPM
def new_rpm(r):
    print("RPM: ",r.value)
    print("\t")
    
#SPEED
def new_speed(r):
    print("SPEED: ",r.value)
    print("\t")
    
#COOLANT TEMPERATURE
def new_coolant_temperature(r):
    print("Coolant temperature: ",r.value)
    print("\t")
    
#THROTTLE POSITION
def new_throttle_position(r):
    print("THROTTLE POSITION: ",r.value)
    print("\t")
    
#FUEL_LEVEL
    
def new_fuel_level(r):
    print("Fuel level: ",r.value)
    print("\t")

cmd = obd.commands.FUEL_PRESSURE
r = connection.query(cmd, force=True)
print("Fuel level: ",r.value)
    
connection.watch(obd.commands.RPM, callback=new_rpm)
connection.watch(obd.commands.SPEED, callback=new_speed)
connection.watch(obd.commands.COOLANT_TEMP, callback=new_coolant_temperature)
connection.watch(obd.commands.THROTTLE_POS, callback=new_throttle_position)
connection.watch(obd.commands.FUEL_PRESSURE, callback=new_fuel_level)
connection.start()

# the callback will now be fired upon receipt of new values
time.sleep(600)
connection.stop()

**Output:**

Fuel level:  None
[obd.obd] 'b'0111': Throttle Position' is not supported
[obd.obd] 'b'010A': Fuel Pressure' is not supported
RPM:  0.0 revolutions_per_minute
	
SPEED:  0.0 kph
	
Coolant temperature:  68 degC

alistair23, 2019-11-03:
Not every car supports all the queries. Are you sure your car support them?

JakubJD, 2019-11-05:
Hi, 
thanks for your answer. Yes, may be, you are right. I have to check it.

Oswaldoo97, 2019-11-09:
As alistair23 says, if you get "x is not suported", your car doesn't support the commands. I got the same issue with some parameters, and I tried with a different car and I got all the parameters I asked.

alistair23, 2019-11-09:
I'm going to close this issue, feel free to re-open if it isn't fixed.

closed 2019-11-09 by alistair23