brendan-w/python-OBD #164: Connection issues

jevgienij123, 2019-11-03:
I've built a small car PC based on Orange Pi and python-OBD which works very well in general. Still there are some issues with establishing the connection itself:

`[s] [~] $ ./main.py
[obd.obd] ======================= python-OBD (v0.7.0) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=/dev/ttyS3 BAUD=38400 PROTOCOL=5
[obd.elm327] write: b'ATZ\r\n'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'\xfc\r\rELM327 v1.5\r\r>'
[obd.elm327] write: b'ATE0\r\n'
[obd.elm327] read: b'ATE0\rOK\r\r>'
[obd.elm327] write: b'ATH1\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'ATL0\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'ATTP5\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'0100\r\n'
[obd.elm327] read: b'BUS INIT: ERROR\r\r>'
[obd.elm327] Connected Successfully: PORT=/dev/ttyS3 BAUD=38400 PROTOCOL=5
[obd.obd] querying for supported commands
[obd.obd] Sending command: b'0100': Supported PIDs [01-20]
[obd.elm327] write: b'0100\r\n'
[obd.elm327] read: b'BUS INIT: ERROR\r\r>'
[obd.OBDCommand] b'0100': Supported PIDs [01-20] did not recieve any acceptable messages
[obd.obd] No valid data for PID listing command: b'0100': Supported PIDs [01-20]
[obd.obd] finished querying with 7 commands supported
[obd.obd] ===================================================================
[obd.obd] ======================= python-OBD (v0.7.0) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=/dev/ttyS3 BAUD=38400 PROTOCOL=5
[obd.elm327] write: b'ATZ\r\n'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'\xfc\r\rELM327 v1.5\r\r>'
[obd.elm327] write: b'ATE0\r\n'
[obd.elm327] read: b'ATE0\rOK\r\r>'
[obd.elm327] write: b'ATH1\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'ATL0\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'ATTP5\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'0100\r\n'
[obd.elm327] read: b'BUS INIT: OK\r86 F1 10 41 00 BE 3E B8 11 8D \r\r>'
[obd.protocols.protocol] map ECU 16 --> ENGINE
[obd.elm327] Connected Successfully: PORT=/dev/ttyS3 BAUD=38400 PROTOCOL=5
[obd.obd] querying for supported commands
[obd.obd] Sending command: b'0100': Supported PIDs [01-20]
[obd.elm327] write: b'0100\r\n'
[obd.elm327] read: b'86 F1 10 41 00 BE 3E B8 11 8D \r\r>'
[obd.obd] Sending command: b'0120': Supported PIDs [21-40]
[obd.elm327] write: b'0120\r\n'
[obd.elm327] read: b'86 F1 10 41 20 80 00 00 00 68 \r\r>'
[obd.obd] finished querying with 43 commands supported
[obd.obd] ===================================================================
[obd.asynchronous] Watching command: b'0105': Engine Coolant Temperature
[obd.asynchronous] subscribing callback for command: b'0105': Engine Coolant Temperature
[obd.asynchronous] Watching command: b'010F': Intake Air Temp
[obd.asynchronous] subscribing callback for command: b'010F': Intake Air Temp
[obd.asynchronous] Watching command: b'ATRV': Voltage detected by OBD-II adapter
[obd.asynchronous] subscribing callback for command: b'ATRV': Voltage detected by OBD-II adapter
[obd.asynchronous] Watching command: b'010D': Vehicle Speed
[obd.asynchronous] subscribing callback for command: b'010D': Vehicle Speed
[obd.asynchronous] Watching command: b'0104': Calculated Engine Load
[obd.asynchronous] subscribing callback for command: b'0104': Calculated Engine Load
[obd.asynchronous] Starting async thread
[obd.obd] Sending command: b'0105': Engine Coolant Temperature
[obd.elm327] write: b'0105\r\n'
[obd.elm327] read: b'83 F1 10 41 05 87 51 \r\r>'
[obd.obd] Sending command: b'010F': Intake Air Temp
[... 118 lines cut ...]
[obd.elm327] write: b'ATH1\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'ATL0\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'AT RV\r'
[obd.elm327] read: b'13.5V\r\r>'
[obd.elm327] write: b'ATTP5\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'BUS INIT: OK\r'
[obd.elm327] Connected Successfully: PORT=/dev/ttyS3 BAUD=38400 PROTOCOL=5
[obd.obd] querying for supported commands
[obd.obd] Sending command: b'0100': Supported PIDs [01-20]
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'86 F1 10 41 00 BE 3E B8 11 8D \r\r>'
[obd.OBDCommand] b'0100': Supported PIDs [01-20] did not receive any acceptable messages
[obd.obd] No valid data for PID listing command: b'0100': Supported PIDs [01-20]
[obd.obd] finished querying with 7 commands supported
[obd.obd] ===================================================================
^CKeyboard Interrupt (ID: 2) has been caught. Cleaning up...
[s] [~] $`

Every connection attempt ends at "finished querying with 7 commands supported". So to put it simply the v0.7.1 doesn't work for me at all.

Any ideas?

alistair23, 2019-11-03:
Can you run a git bisect to figure out what commit caused the regression? That would be the best place to start. It will take more digging to debug the bus init error, maybe you can try with a delay and see if that helps?

jevgienij123, 2019-11-03:
I'll try bisect soon but where I could add a delay?

alistair23, 2019-11-04:
You can try this: https://github.com/brendan-w/python-OBD#bluetooth-obd-ii-adapters

Robbie98, 2020-01-12:
I'm having exact same issue, but its on the third connection with v0.7.1
it will find 7 commands then the third try i get around 50 or so? Has anyone found a solution?

jevgienij123, 2020-01-12:
On my another car it connects on the third try as well.

I'm not using bluetooth nor the USB. I have a direct serial connection. Do I still need to try the delay?

Robbie98, 2020-01-13:
I'm using Bluetooth, I tried a usb adapter but believe it's not suitable for the application

I didn't try the delay but added in the "fast=False"

Will have another try tonight

amirmhp98, 2021-08-31:
any update yet? I'm really stuck on this problem :(
I got the same results except that mine cannot connect at the 2nd or 3rd try.
it remains on that "only 8 commands" state.
my logs:

`
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=\.\COM3 BAUD=38400 PROTOCOL=5
[obd.elm327] write: b'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'ELM327 v2.1\r>'
[obd.elm327] write: b'ATE0\r'
[obd.elm327] read: b'OK\r>'
[obd.elm327] write: b'ATH1\r'
[obd.elm327] read: b'OK\r>'
[obd.elm327] write: b'ATL0\r'
[obd.elm327] read: b'OK\r>'
[obd.elm327] write: b'AT RV\r'
[obd.elm327] read: b'13.9V\r>'
[obd.elm327] write: b'ATTP5\r'
[obd.elm327] read: b'OK\r>'
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'BUS INIT: OK\r'
[obd.protocols.protocol] map ECU 16 --> ENGINE
[obd.elm327] Connected Successfully: PORT=\.\COM3 BAUD=38400 PROTOCOL=5
[obd.obd] querying for supported commands
[obd.obd] Sending command: b'0100': Supported PIDs [01-20]
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'86 F1 11 41 00 BE 3E B8 11 8E \r\r>'
[obd.OBDCommand] b'0100': Supported PIDs [01-20] did not receive any acceptable messages
[obd.obd] No valid data for PID listing command: b'0100': Supported PIDs [01-20]
[obd.obd] Sending command: b'0900': Supported PIDs [01-20]
[obd.elm327] write: b'0900\r'
[obd.elm327] read: b'87 F1 11 49 00 01 FC 00 00 00 CF \r\r>'
[obd.obd] finished querying with 8 commands supported
[obd.obd] ===================================================================
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=\.\COM3 BAUD=38400 PROTOCOL=5
[obd.elm327] could not open port '\\.\\COM3': PermissionError(13, 'Access is denied.', None, 5)
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=\.\COM3 BAUD=38400 PROTOCOL=5
[obd.elm327] Cannot configure port, something went wrong. Original message: OSError(22, 'No process is on the other end of the pipe.', None, 233)
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=\.\COM3 BAUD=38400 PROTOCOL=5
[obd.elm327] write: b'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'ELM327 v2.1\r>'
[obd.elm327] write: b'ATE0\r'
[obd.elm327] read: b'OK\r>'
[obd.elm327] write: b'ATH1\r'
[obd.elm327] read: b'OK\r>'
[obd.elm327] write: b'ATL0\r'
[obd.elm327] read: b'OK\r>'
[obd.elm327] write: b'AT RV\r'
[obd.elm327] read: b'14.2V\r>'
[obd.elm327] write: b'ATTP5\r'
[obd.elm327] read: b'OK\r>'
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'BUS INIT: UNABLE TO CONNECT\r>'
[obd.elm327] Adapter connected, but the ignition is off
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=\.\COM3 BAUD=38400 PROTOCOL=5
[obd.elm327] could not open port '\\.\\COM3': PermissionError(13, 'Access is denied.', None, 5)
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=\.\COM3 BAUD=38400 PROTOCOL=5
[obd.elm327] Cannot configure port, something went wrong. Original message: OSError(22, 'No process is on the other end of the pipe.', None, 233)
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=\.\COM3 BAUD=38400 PROTOCOL=5
[obd.elm327] write: b'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'ELM327 v2.1\r>'
[obd.elm327] write: b'ATE0\r'
[obd.elm327] read: b'OK\r>'
[obd.elm327] write: b'ATH1\r'
[obd.elm327] read: b'OK\r>'
[obd.elm327] write: b'ATL0\r'
[obd.elm327] read: b'OK\r>'
[obd.elm327] write: b'AT RV\r'
[obd.elm327] read: b'14.1V\r>'
[obd.elm327] write: b'ATTP5\r'
[obd.elm327] read: b'OK\r>'
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'BUS INIT: OK\r'
[obd.protocols.protocol] map ECU 16 --> ENGINE
[obd.elm327] Connected Successfully: PORT=\.\COM3 BAUD=38400 PROTOCOL=5
[obd.obd] querying for supported commands
[obd.obd] Sending command: b'0100': Supported PIDs [01-20]
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'86 F1 11 41 00 BE 3E B8 11 8E \r\r>'
[obd.OBDCommand] b'0100': Supported PIDs [01-20] did not receive any acceptable messages
[obd.obd] No valid data for PID listing command: b'0100': Supported PIDs [01-20]
[obd.obd] Sending command: b'0900': Supported PIDs [01-20]
[obd.elm327] write: b'0900\r'
[obd.elm327] read: b'87 F1 11 49 00 01 FC 00 00 00 CF \r\r>'
[obd.obd] finished querying with 8 commands supported
[obd.obd] ===================================================================
[obd.obd] ======================= python-OBD (v0.7.1) =======================
`

jankovalski, 2021-08-31:
I'm sorry, I never solved the problem. I used this thing mainly for reading basic things like coolant temp. Now I've made my own device based on an ESP32 and a TFT screen. Using my own sensors. It boots up in an instant unlike the raspberry pi. I'm all done with issues related to reading 20 year old ECU.