brendan-w/python-OBD #166: I am getting "query failed, no connection available"

Oswaldoo97, 2019-11-09:
(This is a post that I was going to post on the Raspberry forums but I couldn't post it for problems with mi IP. Sorry if it causes discomfort)

Hello everyone, this is my first post on the Raspberry forums and I would really appreciate your help with this issue that I have.

**[THE PROJECT]**

I am trying to do a Head-Up Display using the ELM327 Bluetooth and the Raspberry PI 3B+

**[THE PROCEDURE]**

- The first step is make a bluetooth connection between the ELM327 and the Raspberry. For this step I am using the following commands:

$ bluetoothctl
# power on
# pairable on
# agent on
# default-agent
# scan on

In this part, I get the ADRESS of my ELM327 A2:2A:19:04:00:00

- The second step is pair the Raspberry and the ELM327. For this step I am using the following commands:

# pair A2:2A:19:04:00:00
# trust A2:2A:19:04:00:00
# quit

At this point the bluetooth connection is done with no problems.

- In the third step I locate the ELM327 in the port0 with the following command:

$ sudo rfcomm bind rfcomm0 A2:2A:19:04:00:00

- And finally in the last step I launch the following command to start sending AT commands to enable some functions of the ELM327

$ screen /dev/rfcomm0

In this part the screen is openned and I send the following AT commands.

- ATZ (I get the  ELM327 version, that is 2.1)
- ATL1 (I enable the line feeds)
- ATH1 (The headers are ON)
- ATSP0 (I set the protocol of the car, in this case 0 = automatic detection)

After sending the last 4 commands, I can request for the vehicle parameters like speed, rpm, coolant temperature, etc, with PID commands.

010C (For RPM)
010D (For speed)
0105 (For coolant temprature)

and I get the measure of the parameters but in a hexadecimal code, therefore I have to interpet that hexadecimal code in a decimal code, and for this part, there are a obd library that make the job, the library is called python-obd.

Here start the issue

[THE ISSUE]

In the link of the library there is a simple code that you can use to get the RPM in a decimal value. The code is the following:

import obd

connection = obd.OBD() # auto-connects to USB or RF port

cmd = obd.commands.RPM # select an OBD command (sensor)
response = connection.query(cmd) # send the command, and parse the response
print(response.value) # returns unit-bearing values thanks to Pint

I try to run that code but I get the following responses

[obd.obd] No OBD-II adapters found
[obd.obd] Cannot load commands: No connection to car
[obd.obd] Query failed, no connection available
None

And this is the issue that I am getting, please help me, I can't find the solution to this problem. I don't know if the problem is the ELM327, the Raspberry, the car or another thing.

Something that I would like to add is that two weeks ago my project was working with no problems, I had an ELM327 that was working fine and I was getting the vehicle parameters when I asked for it, but unfortunately my partner damaged the ELM327 and we had to bought a new one, but it didn't work (I think). :( After that I was analyzing the damaged ELM327 and the new one and I realized that the microcontrollers were different and today I bought another one with the same microcontroller than the damaged one but I am still getting the same issue. :(

Thank you!

alistair23, 2019-11-09:
This line explains it:

```
[obd.obd] No OBD-II adapters found
```

You don't have any adapters. Do you see your adapter under `/dev/` or in `dmesg`?

Oswaldoo97, 2019-11-09:
But I have connected my ELM327 in the OBD port, and I am getting the data of RPM, SPEED, COOLANT TEMPERATURE, but in hexadecimal and when I try to use the library to interpret the hexadecimal code to decimal code, I get the issues I posted.

Sorry, can you tell me exactly how can I see my device? Or how can I run the commands you told me? (/dev/ and dmesg), thank you!

alistair23, 2019-11-09:
Ah, I missed that you can connect to `/dev/rfcomm0`. Can you specify that as the port?

Oswaldoo97, 2019-11-09:
Yes I have sepecified the device in the port 0, and also I did the same with the first ELM327 I bought (with the damaged device which I was using to get the parameters). But with the 2 new ones I am getting this issue, but I don't know why :( 
Tomorrow I will try with another car, but today I wanted to know if the issue isn't caused by the ELM327.

alistair23, 2019-11-09:
I don't think it's the car, it can't connect to the port. It seems strange that you don't get any more logs, have you turned debugging on?

Oswaldoo97, 2019-11-09:
When I put the following line in my code: obd.logger.setLevel(obd.logging.DEBUG) 

I get the following lines:

[obd.obd] ================python-OBD (v0.7.1)===================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: []
[obd.obd] No OBD-II adapters found
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ================================================
[obd.obd] Query failed, no connection available
None

alistair23, 2019-11-09:
Manually specify the port

Oswaldoo97, 2019-11-09:
Sorry man, can you tell me how can I specify manually the port?

Also I have launched this code to see the available ports and I only get this "[]"

import obd
ports = obd.scan_serial()   
print ports

alistair23, 2019-11-09:
The documentation explains it: https://python-obd.readthedocs.io/en/latest/Connections/

`connection = obd.OBD("/dev/rfcomm0")`

Oswaldoo97, 2019-11-09:
Now I get the following response:

[obd.elm327] [Errno 16] could not open port /dev/rfcomm0: [Errno 16] Device or resource busy: '/dev/rfcomm0'

alistair23, 2019-11-09:
You will have to see if something else has opened the port.

Oswaldoo97, 2019-11-09:
I was typed "sudo rfcomm bind rfcomm0 00:00:00:33:33:33", because I tought that I had to write it before putting the port manually, therefore I was getting the device was busy. I have rebooted the Raspberry and I launched the code and now I get the following responses:

[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=auto PROTOCOL=auto
[obd.elm327] [Errno 2] could not open port /dev/rfcomm0: [Errno 2] No such file or directory: '/dev/rfcomm0'
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
[obd.obd] Query failed, no connection available
None

alistair23, 2019-11-09:
Now the file doesn't exist, that seems to be worse.

Oswaldoo97, 2019-11-17:
I have tried and tried but unsuccessfully :( Do you think that maybe could be a problem with thethe programation of thethe microcontroller that I could fix with the AT commands?

alistair23, 2019-11-18:
If you can't connect then I don't think so. That seems like a problem on your host side.

Oswaldoo97, 2019-11-18:
What do you mean with "host side"?
To keep us in touch, I have bought 3 new ELM327 on Amazon which get to my home on wednesday. I have bought that with the purpose of discarding issues with the scanner. By this time on wednesday I will be posting my result and I hope success im that process.

alistair23, 2019-11-20:
Host side is the machine that is connecting to the OBD connector.

Oswaldoo97, 2019-11-25:
Hello again, the issue has been solved. I was getting issues due to the automatic connection that I was doing between the raspberry and the obd2 port/elm327, because I had configured the raspberry to connect automatically to the scanner once rebooted the raspberry, but to do this procces, I had type this code line "sudo rfcomm bind rfcomm99 (MAC address)" instead of "rfcomm bind rfcomm99 (MAC adress)". 

Thank you anyway!

By the way, do you know if I can get the fuel level of the ELM327? Because I have tried with torque and with the raspberry and I can't get that.

closed 2019-11-26 by alistair23

ksrm0024, 2020-06-30:
Hi there,

I recognize that this is an old thread, but I am having similar problems. I have:

`import obd
connection = obd.OBD("/dev/rfcomm99")
response = connection.query(obd.commands.RPM)
print(response.value)
connection.close()`

Which returns

`[obd.elm327] [Errno 16} could not open port /dev/rfcomm99: [Errno 16] device or resource busy: '/dev/rfcomm99'
[obd.obd] Cannot load commands: No connection to car
[obd.obd] Query failed, no connection available
None`

I have tried this with both rfcomm bind rfcomm99 <mac> and sudo rfcomm bind rfcomm99 but I cannot seem to get it to work, despite being able to get values from Screen. How exactly did you get it to work?