brendan-w/python-OBD #167: Only works when debug is enabled?

romwhite, 2019-11-14:
Not seeing this specific issue mentioned. I'm using a Windows 10 PC and a obdator Mini Bluetooth OBD2 Scanner ELM327 and coding in Python3. Also I've seen threads talking about knockoff ELM chips... but I don't see anyone stating what OBDII readers have real ones.

I can only seem to get things to work when I've uncommented out the diagnostics line at the top. 

import obd
import time

obd.logger.setLevel(obd.logging.DEBUG)

connection = obd.OBD("COM8", fast=False, baudrate=38400)

while True:
    command = obd.commands.SPEED
    print("\n")

    r = connection.query(command)
    print(r.value)
    time.sleep(5)

[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=COM8 BAUD=38400 PROTOCOL=auto
[obd.elm327] write: b'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'\r\rELM327 v1.5\r\r>'
[obd.elm327] write: b'ATE0\r'
[obd.elm327] read: b'ATE0\rOK'
[obd.elm327] write: b'ATH1\r'
[obd.elm327] read: b'OK\r'
[obd.elm327] write: b'ATL0\r'
[obd.elm327] read: b'OK\r'
[obd.elm327] write: b'AT RV\r'
[obd.elm327] read: b'13.3V\r\r>'
[obd.elm327] write: b'ATSP0\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'SEARCHING...\r7E8 06 41 00 BE 1F A8 13 \r\r>'
[obd.elm327] write: b'ATDPN\r'
[obd.elm327] read: b'A6\r\r>'
[obd.protocols.protocol] map ECU 0 --> ENGINE
[obd.elm327] Connected Successfully: PORT=COM8 BAUD=38400 PROTOCOL=6
[obd.obd] querying for supported commands
[obd.obd] Sending command: b'0100': Supported PIDs [01-20]
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'7E8 06 41 00 BE 1F A8 13 \r\r>'
[obd.obd] Sending command: b'0120': Supported PIDs [21-40]
[obd.elm327] write: b'0120\r'
[obd.elm327] read: b'7E8 06 41 20 90 05 B0 15 \r\r>'
[obd.obd] Sending command: b'0140': Supported PIDs [41-60]
[obd.elm327] write: b'0140\r'
[obd.elm327] read: b'7E8 06 41 40 FA DC 20 00 \r\r>'
[obd.obd] Sending command: b'0600': Supported MIDs [01-20]
[obd.elm327] write: b'0600\r'
[obd.elm327] read: b'7E8 06 46 00 C0 00 00 01 \r\r>'
[obd.obd] Sending command: b'0620': Supported MIDs [21-40]
[obd.elm327] write: b'0620\r'
[obd.elm327] read: b'7E8 06 46 20 80 00 00 09 \r\r>'
[obd.obd] Sending command: b'0640': Supported MIDs [41-60]
[obd.elm327] write: b'0640\r'
[obd.elm327] read: b'7E8 06 46 40 40 00 00 01 \r\r>'
[obd.obd] Sending command: b'0660': Supported MIDs [61-80]
[obd.elm327] write: b'0660\r'
[obd.elm327] read: b'7E8 06 46 60 00 00 00 01 \r\r>'
[obd.obd] Sending command: b'0680': Supported MIDs [81-A0]
[obd.elm327] write: b'0680\r'
[obd.elm327] read: b'7E8 06 46 80 00 00 00 01 \r\r>'
[obd.obd] Sending command: b'06A0': Supported MIDs [A1-C0]
[obd.elm327] write: b'06A0\r'
[obd.elm327] read: b'7E8 06 46 A0 F8 00 00 00 \r\r>'
[obd.obd] finished querying with 100 commands supported
[obd.obd] ===================================================================

[obd.obd] Sending command: b'010D': Vehicle Speed
[obd.elm327] write: b'010D\r'
[obd.elm327] read: b'7E8 03 41 0D 00 \r\r>'
0.0 kph

alistair23, 2019-11-14:
Are you using the master branch?

romwhite, 2019-11-14:
I'm using the version installed with pip. Should I have installed it differently?

alistair23, 2019-11-14:
Can you try using the master branch? It has some fixes that might fix the issue for you.

romwhite, 2019-11-14:
Yes, I'll give this a shot. I've also ordered a hard wired OBDII reader as well. I don't know that I trust this bluetooth one.

romwhite, 2019-11-15:
Wanted to mention that this resolved the issue. Not sure why... the version number didn't change.

closed 2019-11-15 by romwhite

alistair23, 2019-11-15:
Great!