brendan-w/python-OBD #168: Custom Command - only HEX, no PID

vitali-mueller, 2019-11-20:
Hello together,

right now I am trying to get the SoC (state of charge) from an electric car via OBD ELM327 Adapter (wired).
The issue I already had, that the connection can't get established with ELM Scan Master. As the electric car doesn't have an ECU, I think that's the problem. With a combustion engine the connection could be established, so the chip seems to work.
With python I can connect to the car, but reading the log, I get an error when reading "0100" for the supported PIDs, so I can't get any data over PIDs.
But my actual approach is to send a HEX code (8 byte) with the ELM327 adapter, independently of PIDs. I know the exact HEX code by listening to the codes sent and received from a diagnosis module over OBD.
So as the diagnosis module is able to read the data over OBD, I was thinking about getting this data with our RPI and the python-OBD package, but I'm stuck now how to make an approach without PID queries. Which command is even used for the HEX query? 
And as I'm an beginner in python, this isn't getting easier...

I hope you understand my problem and can help me out.
Thanks and best regards.

alistair23, 2019-11-22:
I'm not really sure what you are asking.

The send command can be used to send data: https://github.com/brendan-w/python-OBD/blob/b2d65f6e93161b1461450af14c3ea647fe9d0760/obd/elm327.py#L467

Oswaldoo97, 2019-11-25:
I was reading that the ELM327 can't support electric cars, diesel cars and hybrid cars. Maybe your problem starts there.

vitali-mueller, 2019-11-28:
Thanks for your messages.
I changed to PiCAN2 using python-can.
This seems to work for my purpose.

closed 2019-12-04 by alistair23