brendan-w/python-OBD #169: Is there a way to determinate with the library if the bluetooth connection is good?

Oswaldoo97, 2019-11-27:
Hello everyone,

I am using this library to obtain some parameters of the vehicle to display it on a graphical interface programmed in Java and that is working fine, but I have an issue and not precisely with the library (well I think so).

The issue that I am getting is that I am lossing connection between the raspberry and the elm327 bluetooth, and the transmission of the parameters is paused for a while, then after a few seconds, the connection starts again and so on. (When this happens in the serial console of the raspberry I get the following text "failed to read the port")

I don't know if the issue is the scanner, the car, the rasp, the transmission speed or another thing.

As this is a school project, my teacher told me that it is normal between bluetooth connections (I am not completely sure), but he told me that I had to put in the graphical interface and indicator if the connection is good (a green image of a bluetooth symbol) and if the connection is bad (a red image of a bluetooth symbol), but I was wondering if the library has a command to do this action.

Thank you!

alistair23, 2019-12-04:
I don't think this library does, you should check PySerial or add support to this library.

alistair23, 2020-03-06:
Closing due to lack of activity

closed 2020-03-06 by alistair23

dia986, 2021-07-27:
@Oswaldoo97 how did you get the .py scripts to work with Java?