brendan-w/python-OBD #175: compatible adapters

Miky94, 2020-01-01:
this is not properly a issue, is a clarification, is this library only compatible with ELM327 OBD-II adapters? the reason is that there are tons of clones which doesn't work properly and I was wondering if there is a way to recognize them or  more performing tools to get data at a faster refresh rate

alistair23, 2020-01-01:
The library also works with the STN1110 chips as they are ELM compatible.

As for the clones, that is up to the clone. If they correctly follow the protocol then it will work

Miky94, 2020-01-01:
can you tell me a tool with that chip?
maybe [this](https://www.scantool.net/obdlink-sx/)?
I already have [this](https://i.ibb.co/Rzwq8sV/IMG-20200102-001243.jpg) but I don't know if it is compatible

alistair23, 2020-01-03:
I'm not sure, I have used this which works: https://www.sparkfun.com/products/9555

and now I use a custom designed board to interface with the car.

Miky94, 2020-01-03:
I asked the seller and they told me that all their tools use STN Technology but I don't know if it works with all STN chips or only with the STN1110. I tried with mine but it doesn't connect to the ecu, its seller told me it has a FTDI chip, I don't know if it isn't compatible or I just have to specify some parameter

alistair23, 2020-01-03:
It looks like all STN chips should be ELM327 compatible: https://www.obdsol.com/solutions/chips/

FTDI converts the UART to USB, that should work with this library.

Do you get any errors?

Miky94, 2020-01-03:
yes with that adapter I couldn't connect to the ecu, maybe I have to specify baud or other params?

alistair23, 2020-01-03:
Can you share the debug log?

Miky94, 2020-01-04:
here's  the code
```
import obd
connection=obd.OBD('/dev/ttyUSB0')
c=obd.commands.SPEED#SPEED#RPM#PIDS_A
response=connection.query(c)#, force=True
print(response.value)
connection.close()
```

without debug
```
pi@raspberrypi:~ $ python Desktop/ecu.py 
[obd.elm327] Failed to set baudrate
[obd.obd] Cannot load commands: No connection to car
[obd.obd] Query failed, no connection available
None
```

specifying a baudrate (don't know if it is the correct one) with `connection=obd.OBD('/dev/ttyUSB0',9600)`
```
pi@raspberrypi:~ $ python Desktop/ecu.py 
[obd.elm327] Failed to read port
[obd.elm327] Failed to read port
[obd.elm327] ATE0 did not return 'OK'1
[obd.obd] Cannot load commands: No connection to car
[obd.obd] Query failed, no connection available
None
```

with debug `obd.logger.setLevel(obd.logging.DEBUG)` and without baudrate:
```
pi@raspberrypi:~ $ python Desktop/ecu.py 
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=/dev/ttyUSB0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: b'\x7f\x7f\r'
[obd.elm327] Response from baud 9600: b'\x7f\x7f\r'
[obd.elm327] Response from baud 230400: b'\x7f\x7f\r'
[obd.elm327] Response from baud 115200: b'\x7f\x7f\r'
[obd.elm327] Response from baud 57600: b'\x7f\x7f\r'
[obd.elm327] Response from baud 19200: b'\x7f\x7f\r'
[obd.elm327] Failed to choose baud
[obd.elm327] closing port
[obd.elm327] write: b'ATZ\r'
[obd.elm327] Failed to set baudrate
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
[obd.obd] Query failed, no connection available
None
```

with debug `obd.logger.setLevel(obd.logging.DEBUG)` and with baudrate:
```
pi@raspberrypi:~ $ python Desktop/ecu.py 
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=/dev/ttyUSB0 BAUD=9600 PROTOCOL=auto
[obd.elm327] write: b'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] Failed to read port
[obd.elm327] read: b'ATZ\r'
[obd.elm327] write: b'ATE0\r'
[obd.elm327] Failed to read port
[obd.elm327] read: b'ATE0\r'
[obd.elm327] closing port
[obd.elm327] write: b'ATZ\r'
[obd.elm327] ATE0 did not return 'OK'
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
[obd.obd] Query failed, no connection available
None

```

alistair23, 2020-01-04:
Can you try using the latest master?

Miky94, 2020-01-04:
the v0.7.1 isn't the latest?

alistair23, 2020-01-04:
There are changes on master that haven't been released.

Miky94, 2020-01-04:
how can I install it?

alistair23, 2020-01-04:
A quick search led me to this: https://packaging.python.org/tutorials/installing-packages/#installing-from-vcs
or this should work as well: https://packaging.python.org/tutorials/installing-packages/#installing-from-a-local-src-tree

Miky94, 2020-01-04:
but where do I find the source?

alistair23, 2020-01-06:
This git repo

Miky94, 2020-01-07:
I'm not sure I understand, I did:
`pi@raspberrypi:~ $ sudo pip uninstall obd`
`pi@raspberrypi:~ $ sudo pip install -e git+https://github.com/brendan-w/python-OBD.git#egg=python-OBD
`
is it right? later I'll try it with the car
before:
```
pi@raspberrypi:~ $ sudo pip show obd
Name: obd
Version: 0.7.1
Summary: Serial module for handling live sensor data from a vehicle's OBD-II port
Home-page: http://github.com/brendan-w/python-OBD
Author: Brendan Whitfield
Author-email: brendanw@windworksdesign.com
License: GNU GPLv2
Location: /usr/local/lib/python3.5/dist-packages
Requires: pint, pyserial
Required-by: 
```
after:
```
pi@raspberrypi:~ $ sudo pip show obd
Name: obd
Version: 0.7.1
Summary: Serial module for handling live sensor data from a vehicle's OBD-II port
Home-page: http://github.com/brendan-w/python-OBD
Author: Brendan Whitfield
Author-email: brendanw@windworksdesign.com
License: GNU GPLv2
Location: /home/pi/src/python-obd
Requires: pyserial, pint
Required-by: 
```

alistair23, 2020-01-07:
That looks right, although looking more at master I don't seem much that should fix your issue.

Are you sure you have connected to the right device? You don't seem to be getting any responses.

You could try using screen (or something else) to manually query the chip to see if anything works.

Miky94, 2020-01-07:
I'm trying on a Raspberry Pi Zero W with the reader on the only USB port so /dev/ttyUSB0 should be right. I tried both screen and minicom (I'm not familiar with neither) but when I type e.g. `atz` it doesn't return me anything. the reader works using his software on Windows

alistair23, 2020-01-07:
If python-obd works on Windows with the same reader then it must be an issue with the setup on the Pi. It looks like you aren't communicating with the device. If it doesn't work manually as well then I'd guess somehow you can't comunicate. I'm not really sure how to help though.

Miky94, 2020-01-07:
No I meant on Windows it works with the scanner software, anyway I'll try using python-obd too.

Miky94, 2020-01-07:
Here's the attempt on Windows with and without debug:
```
C:\Users\mikyb\AppData\Local\Programs\Python\Python36-32>python.exe ecu.py
[obd.elm327] Failed to set baudrate
[obd.obd] Cannot load commands: No connection to car
[obd.obd] Query failed, no connection available
None
```
```
C:\Users\mikyb\AppData\Local\Programs\Python\Python36-32>python.exe ecu.py
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=COM3 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: b'\x7f\x7f\r'
[obd.elm327] Response from baud 9600: b'\x7f\x7f\r'
[obd.elm327] Response from baud 230400: b'\x7f\x7f\r'
[obd.elm327] Response from baud 115200: b'\x7f\x7f\r'
[obd.elm327] Response from baud 57600: b'\x7f\x7f\r'
[obd.elm327] Response from baud 19200: b'\x7f\x7f\r'
[obd.elm327] Failed to choose baud
[obd.elm327] closing port
[obd.elm327] write: b'ATZ\r'
[obd.elm327] Failed to set baudrate
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
[obd.obd] Query failed, no connection available
None
```
Maybe this adapter isn't compatible?
I also tried python-obd using a €2 ELM327 OBD2 Bluetooth V2.1 adapter (probably a clone) on the same Pi and it connects, even if very slow and most of commands are unsupported and work intermittently.
I want to buy this ScanTool but before spending €50 I need to be sure that with it this module will work like a clock: https://www.amazon.it/ScanTool-425801-Strumento-diagnostica-professionale/dp/B005ZWM0R4

alistair23, 2020-01-09:
I'm not really sure how to help unfortunately. It looks like you can't communicate with the device. I would try manually sending commands and see if that works.

Miky94, 2020-01-09:
when I send commands after the enter nothing happens, anyway I order a scantool I let you know when it arrives

TDubz89, 2020-01-22:
> wscantool I let you know when it arrives

> Here's the attempt on Windows with and without debug:
>
> ```
> C:\Users\mikyb\AppData\Local\Programs\Python\Python36-32>python.exe ecu.py
> [obd.elm327] Failed to set baudrate
> [obd.obd] Cannot load [...]

> BAUD=auto PROTOCOL=auto

> [obd.obd] Closing connection
> [obd.obd] Cannot load commands: No connection to car
> [obd.obd] ===================================================================
> [obd.obd] Query failed, no connection available
> None
--------------------------------------------------------------------------
It's telling you specify the baud manually. So change it from Auto select, manual input the correct baud per whatever ISO protocol you're supposed to be using.  Also, those bauds look wrong.. refer here or search common baud rates for obd2.

 My experience is limited to a 1996 HVAC system that had super basic functions using rs232 on like, windows 95 with a really low baud rate... But it's most definitely telling you set the baud manually. I'd be checking voltage is sufficient too, and depending what your car uses, be it K line, Lowspeed, High speed etc. Make sure it's got power.

All the chips are clones. Any elm style chip, most are counterfeit, the ones that aren't, use unbranded capacitors and regulators or Bluetooth chips to offset the price. I had a "BAFX " and Carista. Carista was ungodly slow, kinda worked, first which I just broke, the BAFX had a good setup. Pretty fast, on torque pro It spec was 2nd or third fastest at the time.  It used a PIC18f2480 micro controller with a BEKEN wireless reciever. BK3231Q / CG7357 mostly undocumented but it did the trick for my 2010 Mazda that was virtually the same dictate software VAGcom uses, IDs, VCM, any of those

Miky94, 2020-01-22:
from the log, with Auto select it tries a set of baud rates without luck, so I don't know what else to try.
anyway it is probably the fault of the $2 chinese ELM327, which anyway works with his app, I'm waiting a OBDLink SX.
I'll let you know

Miky94, 2020-01-25:
Ok with the OBDLink SX I tried the [Basic Usage](https://github.com/brendan-w/python-OBD#basic-usage) with the PIDS_A command 
on my car (Magneti Marelli ME100T):
`10011000001110110000000000010001`
on my motorcycle (Magneti Marelli IAW 59M):
`00010001000000000000000000000000`
So ok for the car but is it possible that on the motorcycle I can only read "Calculated engine load" and "Short term fuel trim—Bank 2"?
also which frequency do you recommend between one command and another on for example Raspberry Pi Zero and Raspberry Pi 3?

Miky94, 2020-01-28:
It seems strange to me that only 2 PIDs are supported, maybe I could just try to force each protocol from 1 to A to see which has more supported PIDs?

Miky94, 2020-01-29:
update:
I tried the `PIDS_A` command using each protocol from `connection=obd.OBD("/dev/ttyUSB0",None,"1")` to `connection=obd.OBD("/dev/ttyUSB0",None,"A")` but I always get `None` response
so I tried using protocol `auto` enabling debug and it uses protocol `5`
I don't understand this
anyway I can't read nothing else than those two values but I'm sure it is possible because with his program I can
```
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=/dev/ttyUSB0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: b'\xfe\xfe'
[obd.elm327] Response from baud 9600: b''
[obd.elm327] Response from baud 230400: b''
[obd.elm327] Response from baud 115200: b'\x7f\x7f\r?\r\r>'
[obd.elm327] Choosing baud 115200
[obd.elm327] write: b'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'ATZ\r\r\rELM327 v1.3a\r\r>'
[obd.elm327] write: b'ATE0\r'
[obd.elm327] read: b'ATE0\rOK\r\r>'
[obd.elm327] write: b'ATH1\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'ATL0\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'AT RV\r'
[obd.elm327] read: b'13.6V\r\r>'
[obd.elm327] write: b'ATSP0\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'SEARCHING...\r83 F1 10 7F 01 11 15 \r\r>'
[obd.elm327] write: b'ATDPN\r'
[obd.elm327] read: b'A5\r\r>'
[obd.protocols.protocol] map ECU 16 --> ENGINE
[obd.elm327] Connected Successfully: PORT=/dev/ttyUSB0 BAUD=115200 PROTOCOL=5
[obd.obd] querying for supported commands
[obd.obd] Sending command: b'0100': Supported PIDs [01-20]
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'83 F1 10 7F 01 11 15 \r\r>'
[obd.OBDCommand] Message was shorter than expected (3<6). Padded message: bytearray(b'\x7f\x01\x11\x00\x00\x00')
[obd.obd] finished querying with 11 commands supported
[obd.obd] ===================================================================
[obd.obd] Sending command: b'0100': Supported PIDs [01-20]
[obd.elm327] write: b'01001\r'
[obd.elm327] read: b'83 F1 10 7F 01 11 15 \r\r>'
[obd.OBDCommand] Message was shorter than expected (3<6). Padded message: bytearray(b'\x7f\x01\x11\x00\x00\x00')
00010001000000000000000000000000
[obd.obd] Closing connection
[obd.elm327] closing port
[obd.elm327] write: b'ATZ\r'

```

alistair23, 2020-03-06:
Any update on this?

Miky94, 2020-03-06:
yes, see [this](https://github.com/brendan-w/python-OBD/issues/180)
I close this that was not properly a issue

closed 2020-03-06 by Miky94