brendan-w/python-OBD #177: Data does not update on Async connection

saintkazu, 2020-01-06:
I saw this tutorial by Tinkernut on Youtube ([https://youtu.be/VCU81PfSlcI](url)) and used the code he posted on hackster.io. I initially got the "ATH1 did not return 'OK', or echoing is still ON" error. Granted, I am using a counterfeit V1.5 ELM327 usb adapter as mentioned in other issues under this project. However, when I added the obd.logger.setLevel(obd.logging.DEBUG) line to the code, the connection to the car's obd port became successful, and I was able to obtain RPM and Speed data. The problem I am now experiencing is that the data does not update or change at all, meaning the RPM and Speed data shown on the screen are the only data obtained upon connection and no longer updates. If I use this same adapter on the Scanmaster program running on a Windows 10 computer, however, the adapter works as it should, displaying real-time data. Here are the code and log outputs as well as the gui output using pygame: 

pygame 1.9.4.post1
Hello from the pygame community. https://www.pygame.org/contribute.html
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/ttyUSB0']
[obd.obd] Attempting to use port: /dev/ttyUSB0
[obd.elm327] Initializing ELM327: PORT=/dev/ttyUSB0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: b'\x7f\x7f\r?\r\r>'
[obd.elm327] Choosing baud 38400
[obd.elm327] write: b'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'ATZ\r\xfc\r\rELM327 v1.5\r\r>'
[obd.elm327] write: b'ATE0\r'
[obd.elm327] read: b'ATE0\rOK'
[obd.elm327] write: b'ATH1\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'ATL0\r'
[obd.elm327] read: b'OK\r'
[obd.elm327] write: b'AT RV\r'
[obd.elm327] read: b'14.2V\r\r>'
[obd.elm327] write: b'ATSP0\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'\r\r>'
[obd.elm327] write: b'ATDPN\r'
[obd.elm327] read: b'A0\r\r>'
[obd.elm327] ELM responded with unknown protocol. Trying them one-by-one
[obd.elm327] write: b'ATTP6\r'
[obd.elm327] read: b'OK\r\r'
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'7E8 06 41 00 BE 1F B8 13 \r\r>'
[obd.protocols.protocol] map ECU 0 --> ENGINE
[obd.elm327] Connected Successfully: PORT=/dev/ttyUSB0 BAUD=38400 PROTOCOL=6
[obd.obd] querying for supported commands
[obd.obd] Sending command: b'0100': Supported PIDs [01-20]
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'7E8 06 41 00 BE 1F B8 13 \r\r>'
[obd.obd] Sending command: b'0120': Supported PIDs [21-40]
[obd.elm327] write: b'0120\r'
[obd.elm327] read: b'7E8 06 41 20 80 05 A0 01 \r\r>'
[obd.obd] Sending command: b'0140': Supported PIDs [41-60]
[obd.elm327] write: b'0140\r'
[obd.elm327] read: b'7E8 06 41 40 7A DC 00 00 \r\r>'
[obd.obd] Sending command: b'0600': Supported MIDs [01-20]
[obd.elm327] write: b'0600\r'
[obd.elm327] read: b'7E8 06 46 00 C0 00 00 01 \r\r>'
[obd.obd] Sending command: b'0620': Supported MIDs [21-40]
[obd.elm327] write: b'0620\r'
[obd.elm327] read: b'7E8 06 46 20 00 00 00 01 \r\r>'
[obd.obd] Sending command: b'0640': Supported MIDs [41-60]
[obd.elm327] write: b'0640\r'
[obd.elm327] read: b'7E8 06 46 40 C0 00 00 00 \r\r>'
[obd.obd] finished querying with 81 commands supported
[obd.obd] ===================================================================
[obd.asynchronous] Watching command: b'010D': Vehicle Speed
[obd.asynchronous] subscribing callback for command: b'010D': Vehicle Speed
[obd.asynchronous] Watching command: b'010C': Engine RPM
[obd.asynchronous] subscribing callback for command: b'010C': Engine RPM
[obd.asynchronous] Watching command: b'0104': Calculated Engine Load
[obd.asynchronous] subscribing callback for command: b'0104': Calculated Engine Load
[obd.asynchronous] Starting async thread
[obd.obd] Sending command: b'010D': Vehicle Speed
[obd.elm327] write: b'010D\r'
[obd.elm327] read: b'7E8 03 41 0D 00 \r\r>'
[obd.obd] Sending command: b'010C': Engine RPM
[obd.elm327] write: b'010C\r'
[obd.elm327] read: b'7E8 04 41 0C 0B 9B \r\r>'
[obd.obd] Sending command: b'0104': Calculated Engine Load
[obd.elm327] write: b'0104\r'
[obd.elm327] read: b'7E8 03 41 04 4C \r\r>'
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/pint/quantity.py", line 1107, in __getattr__
    return getattr(self._magnitude, item)
AttributeError: 'float' object has no attribute 'mangitude'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/obd/asynchronous.py", line 222, in run
    callback(r)
  File "/home/pi/Desktop/thesis2.py", line 56, in get_load
    load = int(l.value.mangitude)
  File "/home/pi/.local/lib/python3.7/site-packages/pint/quantity.py", line 1110, in __getattr__
    "has attribute '{1}'".format(self._magnitude, item))
AttributeError: Neither Quantity object nor its magnitude (29.80392156862745) has attribute 'mangitude'

Here's the code: 
import pygame
from pygame.locals import *
import obd
obd.logger.setLevel(obd.logging.DEBUG)
pygame.init()
#connection = obd.OBD()
connection = obd.Async(fast=True)
screen = pygame.display.set_mode((848,480),pygame.RESIZABLE)
screen_w = screen.get_width()
screen_h = screen.get_height()
circle_y = screen_h/2
circle1_x = screen_w * .25
circle2_x = screen_w * .5
circle3_x = screen_w * .75
circle_rad = (circle2_x - circle1_x)/2
speed_text_x = screen_w * .25
speed_text_y = screen_h * .25
rpm_text_x = screen_w * .5
rpm_text_y = screen_h * .25
load_text_x = screen_w * .75
load_text_y = screen_h * .25
headerFont = pygame.font.SysFont("Arial", 50)
digitFont = pygame.font.SysFont("Arial", 50)
white = (255,255,255)
black = (0,0,0)
grey = (112, 128, 144)
speed = 0
rpm = 0
load = 0
def draw_hud():
    screen.fill(grey)
    pygame.draw.circle(screen, black, (int(circle1_x), int(circle_y)), int(circle_rad), 5)
    pygame.draw.circle(screen, black, (int(circle2_x), int(circle_y)), int(circle_rad), 5)
    pygame.draw.circle(screen, black, (int(circle3_x), int(circle_y)), int(circle_rad), 5)
    speed_text = headerFont.render("SPEED", True, black)
    rpm_text = headerFont.render("RPM", True, black)
    load_text = headerFont.render("LOAD", True, black)
    speed_text_loc = speed_text.get_rect(center=(speed_text_x, speed_text_y))
    rpm_text_loc = rpm_text.get_rect(center=(rpm_text_x, rpm_text_y))
    load_text_loc = load_text.get_rect(center=(load_text_x, load_text_y))
    screen.blit(speed_text, speed_text_loc)
    screen.blit(rpm_text, rpm_text_loc)
    screen.blit(load_text, load_text_loc)
def get_speed(s):
    global speed
    if not s.is_null():
        speed = int(s.value.magnitude) #for kph
        #speed = int(s.value.magnitude * .060934) #for mph
def get_rpm(r):
    global rpm
    if not r.is_null():
        rpm = int(r.value.magnitude)
def get_load(l):
    global load
    if not l.is_null():
        load = int(l.value.mangitude)
connection.watch(obd.commands.SPEED, callback=get_speed)
connection.watch(obd.commands.RPM, callback=get_rpm)
connection.watch(obd.commands.ENGINE_LOAD, callback=get_load)
connection.start()
running = True
while running:
    for event in pygame.event.get():
        if event.type == KEYDOWN:
            if event.key == K_ESCAPE:
                connection.stop()
                connection.close()
                running = False
            elif event.type == QUIT:
                connection.stop()
                connection.close()
                running = False
    draw_hud()
    speedDisplay = digitFont.render(str(speed), 3, white)
    rpmDisplay = digitFont.render(str(rpm), 3, white)
    loadDisplay = digitFont.render(" " + str(load) + " %", 3, white)
    screen.blit(loadDisplay, (circle3_x-(circle3_x/8), circle_y-45))
    screen.blit(rpmDisplay, (circle2_x-(circle2_x/8), circle_y-45))
    screen.blit(speedDisplay,(circle1_x-(circle1_x/8), circle_y-45))
    pygame.display.update()
    pygame.display.flip()

Here's the output gui. You can see the program is successfully able to output the RPM but when the revs change, the output remains static. I could also see the indicator lights on the OBD adapter not lighting up, meaning that the data is not being fetched. 
[image]

alistair23, 2020-01-06:
Did you try disabling fast?

saintkazu, 2020-01-06:
> Did you try disabling fast?

Yes I have but I got the same results. I also tried with a bluetooth and wifi obd adapter but for some reason they couldn't be detected

closed 2020-01-08 by saintkazu