brendan-w/python-OBD #180: commands not supported

Miky94, 2020-02-16:
running the [basic usage](https://github.com/brendan-w/python-OBD#basic-usage) with the `PIDS_A` command using a [OBDLink SX](https://www.scantool.net/obdlink-sx/) on a **Magneti Marelli IAW 59M** ECU I get:
`00010001000000000000000000000000`
So it can only read "Calculated engine load" and "Short term fuel trim—Bank 2".
I tried the `PIDS_A` command using each protocol from `connection=obd.OBD("/dev/ttyUSB0",None,"1")` to `connection=obd.OBD("/dev/ttyUSB0",None,"A")` but I always get `None` response.
So I tried again with protocol `auto` and debug shows it uses the `ISO 14230-4 (KWP FAST)` protocol, when `connection=obd.OBD("/dev/ttyUSB0",None,"5")` responses `None`, I don't understand this, and the "finished querying with 11 commands supported" message doesn't make sense to me.
With another program that uses more or less Fiat protocol (based on KWP) I also can read other values.
```
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=/dev/ttyUSB0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: b'\xfe\xfe'
[obd.elm327] Response from baud 9600: b''
[obd.elm327] Response from baud 230400: b''
[obd.elm327] Response from baud 115200: b'\x7f\x7f\r?\r\r>'
[obd.elm327] Choosing baud 115200
[obd.elm327] write: b'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'ATZ\r\r\rELM327 v1.3a\r\r>'
[obd.elm327] write: b'ATE0\r'
[obd.elm327] read: b'ATE0\rOK\r\r>'
[obd.elm327] write: b'ATH1\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'ATL0\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'AT RV\r'
[obd.elm327] read: b'13.6V\r\r>'
[obd.elm327] write: b'ATSP0\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'SEARCHING...\r83 F1 10 7F 01 11 15 \r\r>'
[obd.elm327] write: b'ATDPN\r'
[obd.elm327] read: b'A5\r\r>'
[obd.protocols.protocol] map ECU 16 --> ENGINE
[obd.elm327] Connected Successfully: PORT=/dev/ttyUSB0 BAUD=115200 PROTOCOL=5
[obd.obd] querying for supported commands
[obd.obd] Sending command: b'0100': Supported PIDs [01-20]
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'83 F1 10 7F 01 11 15 \r\r>'
[obd.OBDCommand] Message was shorter than expected (3<6). Padded message: bytearray(b'\x7f\x01\x11\x00\x00\x00')
[obd.obd] finished querying with 11 commands supported
[obd.obd] ===================================================================
[obd.obd] Sending command: b'0100': Supported PIDs [01-20]
[obd.elm327] write: b'01001\r'
[obd.elm327] read: b'83 F1 10 7F 01 11 15 \r\r>'
[obd.OBDCommand] Message was shorter than expected (3<6). Padded message: bytearray(b'\x7f\x01\x11\x00\x00\x00')
00010001000000000000000000000000
[obd.obd] Closing connection
[obd.elm327] closing port
[obd.elm327] write: b'ATZ\r'
```

zehpavora, 2020-03-11:
As far as I can see, the problem is how the protocol is being interpreted:
`[obd.OBDCommand] Message was shorter than expected (3<6). Padded message: bytearray(b'\x7f\x01\x11\x00\x00\x00')`

@alistair23 does it mean the protocol is not fully implemented or is in need of an update?

alistair23, 2020-03-11:
AFAIK it's supported, there is always a possibility of a bug though.

The best bet would be to print everything you can inside PythonODB to get a better idea to what is going on. Then compare that to the protocol and see what's wrong.

Miky94, 2020-03-11:
If you tell me what to print I do another try and write the output to you to debug

alistair23, 2020-03-11:
grep for "Message was shorter than expected" and print everything you can around it. You need to figure out why it's not what we are expecting.

Miky94, 2020-03-21:
I read [here](https://github.com/brendan-w/python-OBD/issues/132) and in **/home/pi/src/python-obd/obd/protocols/protocol_legacy.py** I replaced `frame.data = raw_bytes[3:-1]` with `frame.data = raw_bytes[3:]`, here is the output:
```
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=/dev/ttyUSB0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: b'\xfe\xfe'
[obd.elm327] Response from baud 9600: b''
[obd.elm327] Response from baud 230400: b''
[obd.elm327] Response from baud 115200: b'\x7f\x7f\r?\r\r>'
[obd.elm327] Choosing baud 115200
[obd.elm327] write: b'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'ATZ\r\r\rELM327 v1.3a\r\r>'
[obd.elm327] write: b'ATE0\r'
[obd.elm327] read: b'ATE0\rOK\r\r>'
[obd.elm327] write: b'ATH1\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'ATL0\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'AT RV\r'
[obd.elm327] read: b'14.8V\r\r>'
[obd.elm327] write: b'ATSP0\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'SEARCHING...\r83 F1 10 7F 01 11 15 \r\r>'
[obd.elm327] write: b'ATDPN\r'
[obd.elm327] read: b'A5\r\r>'
[obd.protocols.protocol] map ECU 16 --> ENGINE
[obd.elm327] Connected Successfully: PORT=/dev/ttyUSB0 BAUD=115200 PROTOCOL=5
[obd.obd] querying for supported commands
[obd.obd] Sending command: b'0100': Supported PIDs [01-20]
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'83 F1 10 7F 01 11 15 \r\r>'
[obd.OBDCommand] Message was shorter than expected (4<6). Padded message: bytearray(b'\x7f\x01\x11\x15\x00\x00')
[obd.obd] finished querying with 17 commands supported
[obd.obd] ===================================================================
[obd.obd] Sending command: b'0100': Supported PIDs [01-20]
[obd.elm327] write: b'01001\r'
[obd.elm327] read: b'83 F1 10 7F 01 11 15 \r\r>'
[obd.OBDCommand] Message was shorter than expected (4<6). Padded message: bytearray(b'\x7f\x01\x11\x15\x00\x00')
00010001000101010000000000000000
[obd.obd] Closing connection
[obd.elm327] closing port
[obd.elm327] write: b'ATZ\r'
```
there is still inconsistency between `finished querying with 17 commands supported` and `00010001000101010000000000000000` (ENGINE_LOAD, SHORT_FUEL_TRIM_2, RPM, TIMING_ADVANCE, MAF), I tried the RPM command but in 3 attempts it always returned **1093.25 rpm**:
```
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=/dev/ttyUSB0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: b'\xfe\xfe'
[obd.elm327] Response from baud 9600: b''
[obd.elm327] Response from baud 230400: b''
[obd.elm327] Response from baud 115200: b'\x7f\x7f\r?\r\r>'
[obd.elm327] Choosing baud 115200
[obd.elm327] write: b'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'ATZ\r\r\rELM327 v1.3a\r\r>'
[obd.elm327] write: b'ATE0\r'
[obd.elm327] read: b'ATE0\rOK\r\r>'
[obd.elm327] write: b'ATH1\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'ATL0\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'AT RV\r'
[obd.elm327] read: b'14.1V\r\r>'
[obd.elm327] write: b'ATSP0\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'SEARCHING...\r83 F1 10 7F 01 11 15 \r\r>'
[obd.elm327] write: b'ATDPN\r'
[obd.elm327] read: b'A5\r\r>'
[obd.protocols.protocol] map ECU 16 --> ENGINE
[obd.elm327] Connected Successfully: PORT=/dev/ttyUSB0 BAUD=115200 PROTOCOL=5
[obd.obd] querying for supported commands
[obd.obd] Sending command: b'0100': Supported PIDs [01-20]
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'83 F1 10 7F 01 11 15 \r\r>'
[obd.OBDCommand] Message was shorter than expected (4<6). Padded message: bytearray(b'\x7f\x01\x11\x15\x00\x00')
[obd.obd] finished querying with 17 commands supported
[obd.obd] ===================================================================
[obd.obd] Sending command: b'010C': Engine RPM
[obd.elm327] write: b'010C\r'
[obd.elm327] read: b'83 F1 10 7F 01 11 15 \r\r>'
1093.25 revolutions_per_minute
[obd.obd] Closing connection
[obd.elm327] closing port
[obd.elm327] write: b'ATZ\r'

```

beardedone55, 2020-03-21:
The engine computer is returning a negative response code (0x7F) for service 0x01 with a value of 0x11 (service not supported).  So it seems like your car doesn't support reading any live data PIDs.  Do you have any reason to believe it should?  Do you have another scan tool that will display live data for this car?

It also looks like python-obd isn't handling negative response codes, and it is treating whatever is returned by the car as expected data.

Miky94, 2020-03-22:
the ECU is of a 2003 Ducati motorbike but it also mounted on Fiat cars, as written in the first post using his program that uses a KWP-like protocol I can read live data
with the same tool I can read live data from my car

beardedone55, 2020-03-22:
The ECU at address 0x10 responded with service not supported.  Maybe there are multiple ECUs and the wrong one responded?  The ELM327 allows you to set the message header to target specific ECUs, but I don't know how you would do that with this library.  There's probably a way to do it with a custom command.

beardedone55, 2020-03-22:
For your reference, if you figure out how to change the message header, the format of the header would be '80 xx F1', where xx would be the ECU address in hexadecimal.

Miky94, 2020-03-22:
No there is only one ECU

beardedone55, 2020-03-22:
Did you try other services like read trouble codes or reading out the VIN?  Maybe this ECU uses manufacturer specific PIDs to get the live data....

Miky94, 2020-03-22:
I'll try the GET_DTC command as FREEZE_DTC doesn't seem to be supported.
How do I read the VIN? On [Wikipedia](https://en.wikipedia.org/wiki/OBD-II_PIDs#Service_01) I can request it using Service 09 which I dind't find on the [documentation](https://python-obd.readthedocs.io/en/latest/Command%20Tables/).
How can I find out if the ECU uses other PIDs to get the live data? Is there a command to print all the available data?

beardedone55, 2020-03-23:
If you got your copy of python-obd from github, and you've pulled down the latest version of the master branch, there is a command called 'VIN' to get the VIN.  Otherwise, you will have to create a custom command to get mode 09, PID 02.

You can create custom commands to see what services your ECU supports.  Each manufacturer might use a different service for its "enhanced" manufacturer specific data.  Toyota uses service $21.  I think other manufacturers use $22.  Even if you find a service that the ECU supports, it's really hard to find out what the PIDs on these services do, however, because the manufacturers keep this information secret and force the manufacturers of the high-end scan tools to do the same.

Ahghaffari, 2022-04-19:
@Miky94 is this problem solved? I get a message exactly like you using protocol `5` and `ISO 14230-4 (KWP FAST)` ...
@alistair23 reply `read: b'83 F1 10 7F 01 11 15 \r\r>'` that we get is from ELM327 PIC18F2480 IC or it is from our car ECU? I mean the IC generates that reply or IC gets it from ECU? how can we know this?