brendan-w/python-OBD #182: Unable to connect via Bluetooth on OSX

chainhead, 2020-02-18:
I enabled debug as described in #10 to connect my Macbook to OBDII dongle via Bluetooth. Any suggestions why am I not able to connect?

```python
import obd
obd.logger.setLevel(obd.logging.DEBUG)
connection = obd.OBD("/dev/tty.OBDII-Port",baudrate=460800)
cmd = obd.commands.RPM
response = connection.query(cmd)
print(response.value)
```

```bash
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=/dev/tty.OBDII-Port BAUD=460800 PROTOCOL=auto
[obd.elm327] write: 'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] Failed to read port
[obd.elm327] read: b''
[obd.elm327] write: 'ATE0\r'
[obd.elm327] Failed to read port
[obd.elm327] read: b''
[obd.elm327] closing port
[obd.elm327] write: 'ATZ\r'
[obd.elm327] ATE0 did not return 'OK'
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
[obd.obd] Query failed, no connection available
None
```

Hanlon2222, 2020-02-20:
It may be the baud rate. What kind of bluetooth OBDII dongle are you using? ELM327? If so i believe the baud rate i used for mine was 38400.

chainhead, 2020-02-26:
Yes, it is ELM327 as seen in the debug log above. I even tried with `sudo python2`.

alistair23, 2020-03-06:
Can you check connecting to the device with screen or some serial program? It looks like the device isn't connected.

closed 2020-03-06 by alistair23