brendan-w/python-OBD #184: can't connect to car

alecoexposito, 2020-03-09:
1

I'm trying to connect my linux with the car using an USB OBD-II device, it connects to the OBD-II device but not to the car. The software OBD Auto Doctor https://www.obdautodoctor.com/ connects just fine, look what the python-obd in DEBUG mode says when trying to connect and execute a query (RPM):

```
>>> connection = obd.OBD()
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/ttyUSB0']
[obd.obd] Attempting to use port: /dev/ttyUSB0
[obd.elm327] Initializing ELM327: PORT=/dev/ttyUSB0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: '?\r\r>'
[obd.elm327] Choosing baud 38400
[obd.elm327] write: 'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'\r\rELM327 v1.5\r\r>'
[obd.elm327] write: 'ATE0\r'
[obd.elm327] read: b'ATE0\rOK'
[obd.elm327] write: 'ATH1\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: 'ATL0\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: 'AT RV\r'
[obd.elm327] read: b'13.9V\r\r>'
[obd.elm327] write: 'ATSP0\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: '0100\r'
[obd.elm327] read: b'\r\r>'
[obd.elm327] write: 'ATDPN\r'
[obd.elm327] Failed to read port
[obd.elm327] read: b''
[obd.elm327] Failed to retrieve current protocol
[obd.elm327] Adapter connected, but the ignition is off
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
>>> connection.status()
'OBD Connected'
>>> response = connection.query(obd.commands.RPM, force=True)
[obd.obd] Sending command: 010C: Engine RPM
[obd.elm327] write: '010C\r'
[obd.elm327] read: b'010C\r41 0C 0D 4D \r\r>'
[obd.OBDCommand] 010C: Engine RPM did not receive any acceptable messages
>>>
```

toojayzee, 2020-03-10:
You may need to specify a protocol in the constructor. What kind of vehicle is it? What kind of cable are you using? Your cable needs to support the protocol too via the pin outs and serial to USB elm chipset.

gmaniak, 2021-06-24:
I was seeing the same problem. You can try my branch to see if it fixes the problem. 
After you do `write: '0100\r'` a `SEARCHING` output should be read. Because of a sync issue we fail to read the protocol even if the adapter can detect it successfully. 

https://github.com/brendan-w/python-OBD/pull/220