brendan-w/python-OBD #187: write: '0100\r' results in read: b'BUS INIT: ERROR\r\r>' or b'NO DATA\r\r>'

LVMJohnson1996, 2020-04-19:
Previously I had my Raspberry PI 3B+ hooked up to my 2008 Toyota Yaris and was receiving data just fine. The SD card died and I reinstalled python OBD on my new SD card and haven't been able to get it to work since. Running the debug it seems that when 0100 is sent to the adapter while trying different protocols I either get "NO DATA" or "BUS INIT ERROR".

I have specified the portstr="/dev/ttyUSB0", baudrate=38400, protocol="1-A", fast=False, timeout=1, check_voltage=False. I have also added a delay to "r0100 = self.__send(b"0100",delay=1)" just to see if it was an error in reading. 

This is the result that I get when attemting the protocol that should work with my vehilcle ("3" ISO_9141_2):

>>>import obd
>>> connection = obd.OBD(portstr="/dev/ttyUSB0", baudrate=38400, protocol="3", fast=False, timeout=1, check_voltage=False)
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=/dev/ttyUSB0 BAUD=38400 PROTOCOL=3
[obd.elm327] write: 'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'\xfc\r\rELM327 v1.5\r\r>'
[obd.elm327] write: 'ATE0\r'
[obd.elm327] read: b'ATE0\rOK'
[obd.elm327] write: 'ATH1\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: 'ATL0\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: 'ATTP3\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: '0100\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'BUS INIT: ERROR\r\r>'
[obd.elm327] Connected Successfully: PORT=/dev/ttyUSB0 BAUD=38400 PROTOCOL=3
[obd.obd] querying for supported commands
[obd.obd] Sending command: 0100: Supported PIDs [01-20]
[obd.elm327] write: '0100\r'
[obd.elm327] read: b'BUS INIT: ERROR\r\r>'
[obd.OBDCommand] 0100: Supported PIDs [01-20] did not receive any acceptable messages
[obd.obd] No valid data for PID listing command: 0100: Supported PIDs [01-20]
[obd.obd] finished querying with 7 commands supported
[obd.obd] ===================================================================

and this is the result when using the protocol that is used first from the auto detect ("6" ISO_15765_4_11bit_500k):

>>>import obd
>>>connection = obd.OBD(portstr="/dev/ttyUSB0", baudrate=38400, protocol="6", fast=False, timeout=1, check_voltage=False)
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=/dev/ttyUSB0 BAUD=38400 PROTOCOL=6
[obd.elm327] write: 'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'\xfc\r\rELM327 v1.5\r\r>'
[obd.elm327] write: 'ATE0\r'
[obd.elm327] read: b'ATE0\rOK'
[obd.elm327] write: 'ATH1\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: 'ATL0\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: 'ATTP6\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: '0100\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'NO DATA\r\r>'
[obd.elm327] Connected Successfully: PORT=/dev/ttyUSB0 BAUD=38400 PROTOCOL=6
[obd.obd] querying for supported commands
[obd.obd] Sending command: 0100: Supported PIDs [01-20]
[obd.elm327] write: '0100\r'
[obd.elm327] read: b'NO DATA\r\r>'
[obd.OBDCommand] 0100: Supported PIDs [01-20] did not receive any acceptable messages
[obd.obd] No valid data for PID listing command: 0100: Supported PIDs [01-20]
[obd.obd] Sending command: 0600: Supported MIDs [01-20]
[obd.elm327] write: '0600\r'
[obd.elm327] read: b'NO DATA\r\r>'
[obd.obd] finished querying with 7 commands supported
[obd.obd] ===================================================================

Like I said before, I had this USB OBD2 module working with before with Python OBD and was able to get Coolant, Load, Voltage and Throttle Position. Of course none of those commands are working because none of the commands are supported.

Any help would be greatly appreciated.

alistair23, 2020-04-19:
Can you try bisecting and see if a recent commit breaks anything for you?

LVMJohnson1996, 2020-04-19:
Not too familiar with bisecting, as far as I am aware this means to use the most recent commit instead of the master? 
If this is correct when using commit:
https://github.com/brendan-w/python-OBD/commit/8f4a55cd04170d006eb7d1d774fb4bacb1c6282f
I ran into a different debug log when using the auto protocol. It seemed to get stuck until I used a KeyboardInterrupt
The log was as follows with only fast=False as the only argument:
>>> connection = obd.OBD(fast=False)
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/ttyUSB0']
[obd.obd] Attempting to use port: /dev/ttyUSB0
[obd.elm327] Initializing ELM327: PORT=/dev/ttyUSB0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: '?\r\r>'
[obd.elm327] Choosing baud 38400
[obd.elm327] write: 'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'\xfc\r\rELM327 v1.5\r\r>'
[obd.elm327] write: 'ATE0\r'
[obd.elm327] read: b'ATE0\rOK'
[obd.elm327] write: 'ATH1\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: 'ATL0\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: 'AT RV\r'
[obd.elm327] read: b'11.9V\r\r>'
[obd.elm327] write: 'ATSP0\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: '0100\r'
[obd.elm327] read: b'\r\r>'
[obd.elm327] no response; wait: 0.100000 seconds
[obd.elm327] Failed to read port
[obd.elm327] read: b''
[obd.elm327] no response; wait: 0.100000 seconds
[obd.elm327] Failed to read port
[obd.elm327] read: b''
[obd.elm327] no response; wait: 0.100000 seconds
[obd.elm327] Failed to read port
[obd.elm327] read: b''
[obd.elm327] no response; wait: 0.100000 seconds
KeyboardInterrupt

 but when protocol was specified I get the same results as I did before with the master.

alistair23, 2020-04-19:
I mean use [`git bisect`](https://git-scm.com/docs/git-bisect) to determine which commit breaks for you. Start with current master as the `bad` commit and then try an old commit (based on when you last updated) as a `good` commit. You will need to test every commit and mark them bad/good.

LVMJohnson1996, 2020-05-17:
Thank you for the recommendation, however I kept getting the same bus error. When I attempted to use pyOBD (which has also worked in the past) it gave me a similar error message stating that there was an issue after sending 0100. After this I assumed my reader somehow died. I bought a new one and after some configuration with the latest python OBD it is working again. I am having some latency issues though. But I will open a separate ticket to solve that. 

TL;DR - I bought a new adapter and fixed the issue.

closed 2020-05-17 by LVMJohnson1996