brendan-w/python-OBD #188: Device disconnects after a few seconds of getting results.

Elm000, 2020-04-26:
Hey,

I have a small piece of code that I use to read car speed and rpm. When I run it it returns the speed and rpm but after a few seconds the connection seems to drop. I don't know if it the issue with the elm327 adapter or something else. When I run the Torque app on my phone it works just fine and doesn't drop the connection so the for that part the adapter seems to be working fine. I'm using a bluetooth adapter.

I would be really happy if you could help me with this or guide me in the right direction.

This is the output I get when running the code:
0.0 kph
757.0 revolutions_per_minute
0.0 kph
764.0 revolutions_per_minute
0.0 kph
750.0 revolutions_per_minute
0.0 kph
772.0 revolutions_per_minute
0.0 kph
742.0 revolutions_per_minute
0.0 kph
763.0 revolutions_per_minute
0.0 kph
756.0 revolutions_per_minute
[obd.elm327] Failed to read port
None
[obd.elm327] Device disconnected while reading
None

Elm000, 2020-04-26:
This is the code I'm running.

import obd
import time
from time import sleep

#obd.logger.setLevel(obd.logging.DEBUG)

connection = obd.Async("/dev/rfcomm0")

def get_rpm(r):
	print r.value

def get_speed(r):
	print r.value
connection.watch(obd.commands.SPEED, callback=get_speed)
connection.watch(obd.commands.RPM, callback=get_rpm)

connection.start()
time.sleep(60)
connection.stop()

alistair23, 2020-04-26:
Can you include the debug information?

Elm000, 2020-04-26:
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: ''
[obd.elm327] Response from baud 9600: ''
[obd.elm327] Response from baud 230400: 'NO DATA\r\r>'
[obd.elm327] Choosing baud 230400
[obd.elm327] write: 'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'\rELM327 v2.1\r\r>'
[obd.elm327] write: 'ATE0\r'
[obd.elm327] read: b'ATE0\rOK\r\r>'
[obd.elm327] write: 'ATH1\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: 'ATL0\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: 'AT RV\r'
[obd.elm327] read: b'14.2V\r\r>'
[obd.elm327] write: 'ATSP0\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: '0100\r'
[obd.elm327] read: b'SEARCHING...\r86 F1 10 41 00 BE 3E F8 11 CD \r\r>'
[obd.elm327] write: 'ATDPN\r'
[obd.elm327] read: b'A5\r\r>'
[obd.protocols.protocol] map ECU 16 --> ENGINE
[obd.elm327] Connected Successfully: PORT=/dev/rfcomm0 BAUD=230400 PROTOCOL=5
[obd.obd] querying for supported commands
[obd.obd] Sending command: 0100: Supported PIDs [01-20]
[obd.elm327] write: '0100\r'
[obd.elm327] read: b'86 F1 10 41 00 BE 3E F8 11 CD \r\r>'
[obd.obd] Sending command: 0120: Supported PIDs [21-40]
[obd.elm327] write: '0120\r'
[obd.elm327] read: b'86 F1 10 41 20 80 00 00 00 68 \r\r>'
[obd.obd] finished querying with 45 commands supported
[obd.obd] ===================================================================
[obd.asynchronous] Watching command: 010D: Vehicle Speed
[obd.asynchronous] subscribing callback for command: 010D: Vehicle Speed
[obd.asynchronous] Watching command: 010C: Engine RPM
[obd.asynchronous] subscribing callback for command: 010C: Engine RPM
[obd.asynchronous] Starting async thread
[obd.obd] Sending command: 010D: Vehicle Speed
[obd.elm327] write: '010D\r'
[obd.elm327] read: b'83 F1 10 41 0D 00 D2 \r\r>'
0.0 kph
[obd.obd] Sending command: 010C: Engine RPM
[obd.elm327] write: '010C\r'
[obd.elm327] read: b'84 F1 10 41 0C 0E FC DC \r\r>'
959.0 revolutions_per_minute
[obd.obd] Sending command: 010D: Vehicle Speed
[obd.elm327] write: '010D1\r'
[obd.elm327] read: b'83 F1 10 41 0D 00 D2 \r\r>'
0.0 kph
[obd.obd] Sending command: 010C: Engine RPM
[obd.elm327] write: '010C1\r'
[obd.elm327] read: b'84 F1 10 41 0C 0E DC BC \r\r>'
951.0 revolutions_per_minute
[obd.obd] Sending command: 010D: Vehicle Speed
[obd.elm327] write: '010D1\r'
[obd.elm327] read: b'83 F1 10 41 0D 00 D2 \r\r>'
0.0 kph
[obd.obd] Sending command: 010C: Engine RPM
[obd.elm327] write: '010C1\r'
[obd.elm327] read: b'84 F1 10 41 0C 0F 1C FD \r\r>'
967.0 revolutions_per_minute
[obd.obd] Sending command: 010D: Vehicle Speed
[obd.elm327] write: '010D1\r'
[obd.elm327] read: b'83 F1 10 41 0D 00 D2 \r\r>'
0.0 kph
[obd.obd] Sending command: 010C: Engine RPM
[obd.elm327] write: '010C1\r'
[obd.elm327] read: b'84 F1 10 41 0C 0E 9C 7C \r\r>'
935.0 revolutions_per_minute
[obd.obd] Sending command: 010D: Vehicle Speed
[obd.elm327] write: '010D1\r'
[obd.elm327] read: b'83 F1 10 41 0D 00 D2 \r\r>'
0.0 kph
[obd.obd] Sending command: 010C: Engine RPM
[obd.elm327] write: '010C1\r'
[obd.elm327] read: b'84 F1 10 41 0C 0E F4 D4 \r\r>'
957.0 revolutions_per_minute
[obd.obd] Sending command: 010D: Vehicle Speed
[obd.elm327] write: '010D1\r'
[obd.elm327] read: b'83 F1 10 41 0D 00 D2 \r\r>'
0.0 kph
[obd.obd] Sending command: 010C: Engine RPM
[obd.elm327] write: '010C1\r'
[obd.elm327] read: b'84 F1 10 41 0C 0E 98 78 \r\r>'
934.0 revolutions_per_minute
[obd.obd] Sending command: 010D: Vehicle Speed
[obd.elm327] write: '010D1\r'
[obd.elm327] read: b'83 F1 10 41 0D 00 D2 \r\r>'
0.0 kph
[obd.obd] Sending command: 010C: Engine RPM
[obd.elm327] write: '010C1\r'
[obd.elm327] read: b'84 F1 10 41 0C 0F 10 F1 \r\r>'
964.0 revolutions_per_minute
[obd.obd] Sending command: 010D: Vehicle Speed
[obd.elm327] write: '010D1\r'
[obd.elm327] Failed to read port
[obd.elm327] read: b''
[obd.obd] No valid OBD Messages returned
None
[obd.obd] Sending command: 010C: Engine RPM
[obd.elm327] write: '010C1\r'
[obd.elm327] Failed to read port
[obd.elm327] read: b''
[obd.obd] No valid OBD Messages returned
None
[obd.obd] Sending command: 010D: Vehicle Speed
[obd.elm327] write: '010D1\r'
[obd.elm327] Device disconnected while reading
[obd.obd] No valid OBD Messages returned
None
[obd.asynchronous] Async thread terminated because device disconnected

alistair23, 2020-04-26:
Can you still communicate with the adaptor over Bluetooth after it drops out?

Elm000, 2020-04-26:
If I run the code again it works just like it did the last time. The time it takes to disconnect varies a little bit but it's usually within 20 seconds. I'm not sure if that is what you meant by communicating.

alistair23, 2020-04-26:
Hmm... So it's not a permanent disconnection. Have you tried adding a delay?

Also, if you aren't using the latest master update to that.

Elm000, 2020-04-26:
You mean a delay between the queries? I added 1 second delay between the queries before but it didn't seem to help. I'll have to try again tomorrow. 

The master is the latest.

I also suspect there might be something funky going on with the pi's bluetooth or the serial connection since I've tried connecting to the adapter using screen and other methods I found online but none of them seem to even connect. On the other hand when I tried connecting my phone to the adapter, it worked no problem and I was able to manually enter some obd commands.

alistair23, 2020-04-27:
Hmm.. I think the next step is to get screen working with the adaptor. If the connection is unreliable then you are going to have all sorts of issues.

Elm000, 2020-04-27:
I have to say I'm a bit lost with these issues. I've been trying to get this thing to work for like a week now. Google doesn't get me anywhere. It seems like other people are also having issues using screen to connect to rfcomm, no real answers anywhere. When I send commands from my phone to the adapter I always get replies no matter how many times I do it so the adapter seems to be working fine. I think the issue is somewhere in the rfcomm, bluetooth or the raspbian itself.

All I can think of right now would be trying to wipe the pi clean and try again or try a different OS.

alistair23, 2020-04-28:
I have seen lots of issues with the Pi's bluetooth.

I think the best step would be to use a more traditional distro instead of Raspbian and see if that helps. Try to get a recent kernel as well, as it could be a device driver bug. Check `dmesg` and see if it has any hints.

If there are also issues with screen then we can't really do much from a python-obd side.

Elm000, 2020-04-28:
I'll try to look into it, thanks for the help.

Elm000, 2020-05-01:
I finally managed to get it working. 
I plugged in a usb bluetooth dongle to the pi and that seems to work. It seems like there is some issue with the pi's own bluetooth device. I'm using the pi3 model b+. 

Now I can connect with screen and the bluetooth doesn't drop the connection anymore.

closed 2020-05-07 by Elm000