brendan-w/python-OBD #190: Consistent Low Latency Reads

LVMJohnson1996, 2020-05-17:
I plugged up my new scanner to my car and after manually setting the baudrate to 500000 it is working everytime I turn the key. Unfortunately, the response time of the data update is pretty slow. I had this problem intermittently with my last scanner, but this new one I can only get slow responses from the port everytime. I have set the port, baudrate, and fast=True but it appears that the data transmission is still laggy.

alistair23, 2020-05-17:
How old is the car? What protocol are you using? Is that just as fast as the car can respond?

LVMJohnson1996, 2020-05-17:
2008 Toyota Yaris, protocol 6, the car has responded faster with a different OBD scanner in the past. The Asynchronous connection is updating at a visually slow pace. 

I have linked a video showing the latency.
https://youtu.be/762GcOZd0HI

alistair23, 2020-05-18:
Have you checked the debug log?

LVMJohnson1996, 2020-05-18:
I have, but I don't see anything out of the ordinary that would cause such a slow response. Everything looks like it's responding properly until I visually see it on the screen.

> pi@raspberrypi:~ $ python Documents/HUD.py
pygame 1.9.4.post1
Hello from the pygame community. https://www.pygame.org/contribute.html
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=/dev/ttyUSB0 BAUD=500000 PROTOCOL=6
[obd.elm327] write: 'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'ATZ\r\r\rELM327 v1.4\r\r>'
[obd.elm327] write: 'ATE0\r'
[obd.elm327] read: b'ATE0\rOK\r\r>'
[obd.elm327] write: 'ATH1\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: 'ATL0\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: 'AT RV\r'
[obd.elm327] read: b'28.3V\r\r>'
[obd.elm327] write: 'ATTP6\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: '0100\r'
[obd.elm327] read: b'7E8 06 41 00 BE 1F A8 13 \r\r>'
[obd.protocols.protocol] map ECU 0 --> ENGINE
[obd.elm327] Connected Successfully: PORT=/dev/ttyUSB0 BAUD=500000 PROTOCOL=6
[obd.obd] querying for supported commands
[obd.obd] Sending command: 0100: Supported PIDs [01-20]
[obd.elm327] write: '0100\r'
[obd.elm327] read: b'7E8 06 41 00 BE 1F A8 13 \r\r>'
[obd.obd] Sending command: 0120: Supported PIDs [21-40]
[obd.elm327] write: '0120\r'
[obd.elm327] read: b'7E8 06 41 20 90 05 B0 15 \r\r>'
[obd.obd] Sending command: 0140: Supported PIDs [41-60]
[obd.elm327] write: '0140\r'
[obd.elm327] read: b'7E8 06 41 40 FA DC 20 00 \r\r>'
[obd.obd] Sending command: 0600: Supported MIDs [01-20]
[obd.elm327] write: '0600\r'
[obd.elm327] read: b'7E8 06 46 00 C0 00 00 01 \r\r>'
[obd.obd] Sending command: 0620: Supported MIDs [21-40]
[obd.elm327] write: '0620\r'
[obd.elm327] read: b'7E8 06 46 20 80 00 00 09 \r\r>'
[obd.obd] Sending command: 0640: Supported MIDs [41-60]
[obd.elm327] write: '0640\r'
[obd.elm327] read: b'7E8 06 46 40 40 00 00 01 \r\r>'
[obd.obd] Sending command: 0660: Supported MIDs [61-80]
[obd.elm327] write: '0660\r'
[obd.elm327] read: b'7E8 06 46 60 00 00 00 01 \r\r>'
[obd.obd] Sending command: 0680: Supported MIDs [81-A0]
[obd.elm327] write: '0680\r'
[obd.elm327] read: b'7E8 06 46 80 00 00 00 01 \r\r>'
[obd.obd] Sending command: 06A0: Supported MIDs [A1-C0]
[obd.elm327] write: '06A0\r'
[obd.elm327] read: b'7E8 06 46 A0 F8 00 00 00 \r\r>'
[obd.obd] finished querying with 100 commands supported
[obd.obd] ===================================================================
[obd.asynchronous] Watching command: 0105: Engine Coolant Temperature
[obd.asynchronous] subscribing callback for command: 0105: Engine Coolant Temperature
[obd.asynchronous] Watching command: 0142: Control module voltage
[obd.asynchronous] subscribing callback for command: 0142: Control module voltage
[obd.asynchronous] Watching command: 0104: Calculated Engine Load
[obd.asynchronous] subscribing callback for command: 0104: Calculated Engine Load
[obd.asynchronous] Watching command: 0111: Throttle Position
[obd.asynchronous] subscribing callback for command: 0111: Throttle Position
[obd.asynchronous] Starting async thread
[obd.obd] Sending command: 0142: Control module voltage
[obd.elm327] write: '0142\r'
[obd.elm327] read: b'7E8 04 41 42 31 83 \r\r>'
[obd.obd] Sending command: 0111: Throttle Position
[obd.elm327] write: '0111\r'
[obd.elm327] read: b'7E8 03 41 11 2F \r\r>'
[... 514 lines cut ...]
[obd.elm327] write: '01111\r'
[obd.elm327] read: b'7E8 03 41 11 2D \r\r>'
[obd.obd] Sending command: 0104: Calculated Engine Load
[obd.elm327] write: '01041\r'
[obd.elm327] read: b'7E8 03 41 04 00 \r\r>'
[obd.obd] Sending command: 0105: Engine Coolant Temperature
[obd.elm327] write: '01051\r'
[obd.elm327] read: b'7E8 03 41 05 7D \r\r>'
[obd.obd] Sending command: 0142: Control module voltage
[obd.elm327] write: '01421\r'
[obd.elm327] read: b'7E8 04 41 42 31 83 \r\r>'
[obd.obd] Sending command: 0111: Throttle Position
[obd.elm327] write: '01111\r'
[obd.elm327] read: b'7E8 03 41 11 2D \r\r>'
[obd.obd] Sending command: 0104: Calculated Engine Load
[obd.elm327] write: '01041\r'
[obd.elm327] read: b'7E8 03 41 04 00 \r\r>'
[obd.obd] Sending command: 0105: Engine Coolant Temperature
[obd.elm327] write: '01051\r'
[obd.elm327] read: b'7E8 03 41 05 7D \r\r>'
^CTraceback (most recent call last):
  File "Documents/HUD.py", line 177, in <module>
    pygame.display.flip()
KeyboardInterrupt
`

alistair23, 2020-05-18:
I'm not sure then sorry.

stefan31415, 2020-05-18:
Hello, 
I’m not expert, but I saw :

First :
[obd.elm327] write: 'AT RV\r'
[obd.elm327] read: b'28.3V\r\r>'
Have you really 28.3V ? (Voltage regulator KO?)

Second :
You use the protocol 6 - ISO 15765-4 CAN (11 bit ID, 500 Kbaud). I think the 
500Kbaud is the rate between the ELM327 IC and the ECU (defined by the protocol).

This is not the speed between the computer and your ELM327 device (your scanner).
I think you should not  communicate at this rate with your scanner. The IC / cable between ELM327 IC and your computer probably doesn’t support correctly this rate.
I hope this help you.

LVMJohnson1996, 2020-05-29:
I never saw that before. As far as I am aware the voltage shouldn't be 28.3V, but it's still weird that it showed up in the debug. Probing the port with a multi-meter I couldn't find a pin that high, only 12V max. 

When I used obd.OBD() to automatically set the baud rate I get an error stating that it couldn't be set automatically testing all the baud rates. I manually tried 9600 38400 and so on... and the only one that I get a response from is 500000 (for some odd reason).

This is the scanner I am using: https://www.amazon.com/gp/product/B01F0GVBWY/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

I didn't read into the Ford Diagnostic part when I got it and that could potentially be the issue. But as you can see I do get responses from the vehicle. I may just need to get another different scanner again. Anyone that you would recommend?

LVMJohnson1996, 2020-05-29:
The other one that I was using that no longer works is this one: https://www.amazon.com/gp/product/B071GQ8PRD/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1

But I am apprehensive to purchase this one again with the issues I experienced before.

csurf, 2023-07-03:
@stefan31415 is correct. The baud rate setting is for negotation between your elm327 device and the device that's running python-OBD. It has nothing to do with the baudrate that's being used between the elm327 and the vehicle.

In order to best judge your vehicle's response times, you'd have to directly connect to the elm327 via a serial connection, manually send raw OBD2 commands to the elm327, and measure the response times. This can easily be done in basic python without using python-OBD by sending a raw command to the serial port, reading back the response bytes, and measuring how long it took.

LVMJohnson1996, 2023-07-03:
@csurf Thanks for the info, but that was a problem 3 years ago. Issue has been fixed by throwing the old ELM327 out and moving to a Bluetooth setup.

csurf, 2023-07-03:
@LVMJohnson1996 go ahead and close the issue then.

closed 2023-07-03 by LVMJohnson1996