brendan-w/python-OBD #194: Odometer

scoopgracie, 2020-06-13:
Is odometer data available? I need it for a self driving car.

Lizardskins, 2021-02-12:
I would also like to know if this is already available in the code. Or would this be a custom command?

Lizardskins, 2021-02-12:
Just answered my own question. Odometer information is blocked by manufacturers for security reasons. It is not available through the ELM327 OBD interface.

cewbdex, 2021-06-08:
I do not think it is blocked, it is just not a part of the OBD standard, you might get the value, but you need to reverse engineer the communication with vehicle.

tkyanaga, 2022-07-13:
Bumping this up; added the odometer to mode1 of the commands.py file, but got null data when used.

PID reference (A6 found on both sites)
https://en.wikipedia.org/wiki/OBD-II_PIDs
https://www.csselectronics.com/pages/obd2-pid-table-on-board-diagnostics-j1979

Does this OBDCommand look correct?

    OBDCommand("ODOMETER"                   , "Odometer"        , b"01A6", 6, uas(0x25), ECU.ALL, True),

*I've also tried ECU.ENGINE but still get null data

Lizardskins, 2022-07-13:
Excellent Find @tkyanaga, I might have to resurrect this project and try it out.

Lizardskins, 2022-07-14:
Ah, yes I do remember that now that you mention it. I am also testing on a vehicle pre 2019 so I think that's why I wasn't able to get it to work for me as well. I'll post here if I figure anything out in the future. Thanks for reviving this project for me! :P