brendan-w/python-OBD #196: Can't connect to serial port

ksrm0024, 2020-06-30:
Hi there,

I have effectively the same issue as the person in this thread: https://github.com/brendan-w/python-OBD/issues/166

My code is as follows:

`import obd connection = obd.OBD("/dev/rfcomm99")' 
'response = connection.query(obd.commands.RPM)' 
'print(response.value) connection.close()`

Which returns the error:

'[obd.elm327] [Errno 16} could not open port /dev/rfcomm99: [Errno 16] device or resource busy: '/dev/rfcomm99'' 
'[obd.obd] Cannot load commands: No connection to car' 
'[obd.obd] Query failed, no connection available'  
'None'

The person who posted that thread said that automatically connecting with rfcomm bind rfcomm99 <mac address> as opposed to sudo rfcomm bind rfcomm99 <mac address> solved the issue, but that doesn't work in my case. Any ideas would be greatly appreciated - thanks in advance

alistair23, 2020-06-30:
It looks like some other process is accessing `/dev/rfcomm99`

ksrm0024, 2020-07-01:
I figured out the issue. I was using the screen application to test my bluetooth connection upon startup each time and I guess the application would hang and prevent the port from being accessed even after I closed the application. I tried it without testing beforehand and it worked fine.

closed 2020-07-01 by alistair23