brendan-w/python-OBD #198: ELM RESPONDS 7F 01 7F

SalvoQuattropani, 2020-07-20:
Hi everyone, I'm trying to read data from my mercedes sprinter 2015, I can connect to the vehicle but I always get for every command -> "... is not supported". If I set force = True, I get incorrect values because the elm always returns me 7F 01 7F.
Can you help me? I am working on a research project and need to read this information. Strangely, on my Fiat I can read all the information.

```
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['\\.\\COM3']
[obd.obd] Attempting to use port: \.\COM3
[obd.elm327] Initializing ELM327: PORT=\.\COM3 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: '?\r\r>'
[obd.elm327] Choosing baud 38400
[obd.elm327] write: 'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'\r\rELM327 v1.4\r\r>'
[obd.elm327] write: 'ATE0\r'
[obd.elm327] read: b'ATE0\rOK\r\r>'
[obd.elm327] write: 'ATH1\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: 'ATL0\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: 'AT RV\r'
[obd.elm327] read: b'14.1V\r\r>'
[obd.elm327] write: 'ATSP0\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: '0100\r'
[obd.elm327] read: b'SEARCHING...\r7EC 03 7F 01 7F \r\r>'
[obd.elm327] write: 'ATDPN\r'
[obd.elm327] read: b'A6\r\r>'
[obd.protocols.protocol] map ECU 4 --> ENGINE
[obd.elm327] Connected Successfully: PORT=\.\COM3 BAUD=38400 PROTOCOL=6
[obd.obd] querying for supported commands
[obd.obd] Sending command: 0100: Supported PIDs [01-20]
[obd.elm327] write: '0100\r'
[obd.elm327] read: b'7EC 03 7F 01 7F \r\r>'
[obd.OBDCommand] Message was shorter than expected (3<6). Padded message: bytearray(b'\x7f\x01\x7f\x00\x00\x00')
[obd.obd] Sending command: 0600: Supported MIDs [01-20]
[obd.elm327] write: '0600\r'
[obd.elm327] read: b'NO DATA\r\r>'
[obd.obd] finished querying with 21 commands supported
[obd.obd] ===================================================================
[obd.obd] '010C: Engine RPM' is not supported
None
```

Thanks in advance

alistair23, 2020-07-21:
I think you are going to have to dig into this and see what isn't working. Nothing stands out from the log

SalvoQuattropani, 2020-07-21:
Can you suggest me a way to investigate more deeply?

Thanks a lot

alistair23, 2020-07-21:
Start by adding print statements everywhere. See if you can figure out why your message is too short.

beardedone55, 2020-07-22:
The 3-byte response is a "Negative Response Code (NRC)".  It indicates that the ECU returned some sort of error in response to the request.  

- The first byte ($7F) just means that the the response is an NRC.  
- The second byte is the service requested, in this case $01, which makes sense; the request was for service $01, PID $00.
- The third byte is the "Response Code", which in this case is $7F.

Unfortunately, I don't know what a response code of $7F means.  I also don't have access to the latest version of SAE J1979, so it may be a newer code.  The version I have is from 2006.  Does anyone else have access to the latest version of SAE J1979?

If you can find out what an NRC with a response code of $7F means, it might get you closer to figuring out what is wrong.

beardedone55, 2020-07-22:
It is odd that the CAN ID in the response is $7EC.  This is the CAN ID for a response from ECU#5.  I would have expected the response to come from ECU#1, which would respond with a CAN ID $7E8.

By default, the ELM327 issues all requests with CAN ID $7DF, which should cause all emissions related ECUs to respond (there could be more than 1.)  You could try forcing the CAN ID to $7E0, so that only ECU#1 responds using the ATSH command like this:
```
AT SH 7E0
```
You could manually enter this command using a serial terminal, then enter '0100' and see if you get something different.

Otherwise, if you'd prefer to use this library, and you're up to modifying its code, I think you could force the obd library to do the ATSH command for you if you change the __init__ function for the OBD class in obd/obd.py to set the __last_command to something other than '7E0' like this:
```python
    def __init__(...
           ...
          self.__last_command = b'7DF'
```
This should cause the library to issue the ATSH 7E0 command before sending the command to get supported PIDs.

By the way, @alistair23, it seems like the code is assuming that the ELM327 uses a CAN ID of '7E0' by default, which is not true; so I think this is a bug.  I'm not sure how to fix it though, because the change I suggested above would only work with 11-bit CAN, and it would break any other protocol.

SalvoQuattropani, 2020-07-22:
Thanks guys, I try what you suggested and I will let you know as soon as possible.

kaelruland, 2020-09-30:
I've skimmed through the 2007 version of SAE J1979 and ISO 14229. It appears the response code $7F is for "serviceNotSupportedInActiveSession". Some parameters require SecurityAccess seed/key response from diagnostic equipment. I'm not sure through; I'm just reading through this for the first time.

§9.4
http://read.pudn.com/downloads191/doc/899044/ISO+14229+(2006).pdf

SalvoQuattropani, 2021-04-28:
Hi guys, thanks for your support! I have tried everything but to no avail. I'm stuck and don't know how to proceed. Other ideas? In case the key seed is needed, how should I proceed? :( Thanks in advance

namanbhardwaj42, 2022-11-25:
7f means -- > Service Not Supported In Active Session 
This NRC will return by the server if the requested service identifier is not supported in this session.

example document guide :  
session mode -->  
1. operation mode : 01 (0x01)
        service_1 command : 10
2. diagnostic mode: 03 (0x03)
        service_2 command:  20

working example --> sending command -- > 0x10 0x01
in above example you're sendig "10 01" and so following the above document, now I'm in "operation mode session" as I'm sending (0x01) and service I'm requesting (0x10) which lies in operation mode so I'll get the positive response.

 Error Not Working Example -- >  sending command --> 0x20 0x01
 now you are trying to request the service "20" which lies in diagnostic mode. and trying to access it from (0x01)  that lies in "operation mode".
 
 So check your document/guide properly from where you're following this code guide. That will solve your problems.