brendan-w/python-OBD #199: Can't connect: either fails to choose baud or read port

g4brie11e, 2020-08-29:
I'm having trouble making a stable connection to my OBD and am confused why - despite seeing the ports it either throws a _Failed to choose baud_ or with a given baud, _Failed to read port_ (though it does seem to read and write some data).

(MacOS (and Windows7 😬), Python v3.8.5, python-OBD v0.7.1, a few different adapters (inc. Cable Shack, Carly cable + bluetooth))

**My code, basically:**
```
import obd
connection = obd.OBD()
# connection = obd.OBD( protocol="8", baudrate=9600) 
# connection = obd.OBD( protocol="8", baudrate=9600, fast=False, timeout=3.0) 

x = self.connection.query(obd.commands.GET_DTC)
print(f"DTCs: {x.value}")
```

**Auto ports and auto baud:**

> [obd.obd] ======================= python-OBD (v0.7.1) =======================
> [obd.obd] Using scan_serial to select port
> [obd.obd] Available ports: ['/dev/tty.usbserial-1430']
> [obd.obd] Attempting to use port: /dev/tty.usbserial-1430
> [obd.elm327] Initializing ELM327: PORT=/dev/tty.usbserial-1430 BAUD=auto PROTOCOL=auto
> [obd.elm327] Response from baud 38400: b'\x7f\x7f\r'
> [obd.elm327] Response from baud 9600: b'\x7f\x7f\r'
> [obd.elm327] Response from baud 230400: b'\x7f\x7f\r'
> [obd.elm327] Response from baud 115200: b'\x7f\x7f\r'
> [obd.elm327] Response from baud 57600: b'\x7f\x7f\r'
> [obd.elm327] Response from baud 19200: b'\x7f\x7f\r'
> [obd.elm327] Failed to choose baud
> [obd.elm327] closing port
> [obd.elm327] write: b'ATZ\r'
> [obd.elm327] Failed to set baudrate
> [obd.obd] Closing connection
> [obd.obd] Cannot load commands: No connection to car
> [obd.obd] ===================================================================

**With specified baudrate and protocol:**

> [obd.obd] ======================= python-OBD (v0.7.1) =======================
> [obd.obd] Using scan_serial to select port
> [obd.obd] Available ports: ['/dev/tty.usbserial-A50285BI']
> [obd.obd] Attempting to use port: /dev/tty.usbserial-A50285BI
> [obd.elm327] Initializing ELM327: PORT=/dev/tty.usbserial-A50285BI BAUD=9600 PROTOCOL=8
> [obd.elm327] write: b'ATZ\r'
> [obd.elm327] wait: 1 seconds
> [obd.elm327] Failed to read port
> [obd.elm327] read: b'ATZ\r'
> [obd.elm327] write: b'ATE0\r'
> [obd.elm327] Failed to read port
> [obd.elm327] read: b'ATE0\r'
> [obd.elm327] closing port
> [obd.elm327] write: b'ATZ\r'
> [obd.elm327] ATE0 did not return 'OK'
> [obd.obd] Closing connection
> [obd.obd] Cannot load commands: No connection to car
> [obd.obd] ===================================================================

Any help greatly appreciated.

alistair23, 2020-08-29:
It looks like there is no connection to the ELM device. Can you connect to it with screen, minicom or something similar?

dexterac, 2020-11-25:
I am facing same issue. When I try to connect using screen to the ELM devices this is what happens:

pi@leaf01:~$ screen /dev/rfcomm0
[screen is terminating]
pi@leaf01:~$ 

`pi@leaf01:~$ python3
Python 3.7.3 (default, Jul 25 2020, 13:03:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import obd, time

>>> obd.logger.setLevel(obd.logging.DEBUG)
>>> connection = obd.OBD()
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/rfcomm0']
[obd.obd] Attempting to use port: /dev/rfcomm0
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: b''
[obd.elm327] Response from baud 9600: b''
[obd.elm327] Response from baud 230400: b''
[obd.elm327] Response from baud 115200: b'AT+BRSF=24\r'
[obd.elm327] Response from baud 57600: b''
[obd.elm327] Response from baud 19200: b''
[obd.elm327] Failed to choose baud
[obd.elm327] closing port
[obd.elm327] write: b'ATZ\r'
[obd.elm327] Failed to set baudrate
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
>>> connection = obd.OBD(baudrate=115200, fast=False)
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/rfcomm0']
[obd.obd] Attempting to use port: /dev/rfcomm0
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=115200 PROTOCOL=auto
[obd.elm327] write: b'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] Device disconnected while reading
[obd.elm327] cannot perform __write() when unconnected
[obd.elm327] cannot perform __read() when unconnected
[obd.elm327] ATE0 did not return 'OK'
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
>>> connection = obd.OBD(baudrate=115200)
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/rfcomm0']
[obd.obd] Attempting to use port: /dev/rfcomm0
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=115200 PROTOCOL=auto
[obd.elm327] write: b'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] Device disconnected while reading
[obd.elm327] cannot perform __write() when unconnected
[obd.elm327] cannot perform __read() when unconnected
[obd.elm327] ATE0 did not return 'OK'
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
>>> 
`