brendan-w/python-OBD #201: Can't connect to car; failed to set baudrate

troll3rsk8tr, 2020-09-09:
OS: Windows 10
Car: 1998 Toyota Camry LE 2.2L4
[OBD2 Scanner](https://www.amazon.com/OHP-ELMconfig-Adapter-Compatible-Diagnostics/dp/B01F0GVBWY/ref=sr_1_8?dchild=1&keywords=elm327+obd&qid=1599686906&sr=8-8)

Issue: `Failed to choose baud` & `Failed to set baudrate`
Code:
```python
import obd
obd.logger.setLevel(obd.logging.DEBUG)
connection = obd.OBD(timeout=5)
cmd = obd.commands.RPM
response = connection.query(cmd)

print(response.value)
````
Logs:
```
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['\\.\\COM3']
[obd.obd] Attempting to use port: \.\COM3
[obd.elm327] Initializing ELM327: PORT=\.\COM3 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 110: b'\xff'
[obd.elm327] Response from baud 300: b''
[obd.elm327] Response from baud 600: b''
[obd.elm327] Response from baud 1200: b''
[obd.elm327] Response from baud 2400: b''
[obd.elm327] Response from baud 4800: b''
[obd.elm327] Response from baud 9600: b''
[obd.elm327] Response from baud 14400: b''
[obd.elm327] Response from baud 19200: b''
[obd.elm327] Response from baud 38400: b''
[obd.elm327] Response from baud 57600: b'\x7f\x7f\xfd'
[obd.elm327] Response from baud 128000: b'\xff'
[obd.elm327] Response from baud 256000: b'\xff\xfd'
[obd.elm327] Failed to choose baud
[obd.elm327] closing port
[obd.elm327] write: b'ATZ\r'
[obd.elm327] Failed to set baudrate
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
[obd.obd] Query failed, no connection available
None
```
**Notes**
-The car has a crappy battery, I have to start the car before Windows will even recognize the USB device.
-I'm not totally sure the OBD scanner supports the car's protocol, if someone knows of one that would, please let me know.
-Tried EasyOBD software which fails at Interface Communication -check connections and power. (The reason I mentioned the battery previously.)
-I'm not well-versed in car protocols or things like baud rates, please keep that in mind.
-Thanks in advance to anyone that may be able to assist me. :)

alistair23, 2020-09-10:
There is no connection to the ELM327 chip. You need to check the connection

troll3rsk8tr, 2020-09-10:
> There is no connection to the ELM327 chip. You need to check the connection

Are you suggesting there's something wrong with the physical connection between the car and the chip? When I plug in the adapter to the OBD2 port on the car, there is a series of flashing yellow lights and then it switches to the static red light, which I think is the power. Are there any troubleshooting steps you recommend?

alistair23, 2020-09-10:
I don't know about the connection to the car. Their is no connection between the computer and the chip.

troll3rsk8tr, 2020-09-10:
> I don't know about the connection to the car. Their is no connection between the computer and the chip.

How can you tell there is no connection? It seems to be getting responses, no?

alistair23, 2020-09-10:
b'\x7f\x7f\xfd' is just gibberish. You are looking for a question mark. You possibly don't have the RX/TX connected correctly.

Try manually specify the baud and then go from there. Can you connect to the chip via a console?

troll3rsk8tr, 2020-09-10:
Any steps you recommend for determining the right baud rate?

alistair23, 2020-09-11:
See if the manual or datasheet has any hints. Otherwise just try guessing with a serial console application.

troll3rsk8tr, 2020-09-12:
It seems that I can open a COM3 connection via Putty, but I don't know where to go from here. The terminal doesn't allow me to input anything...

alistair23, 2020-09-12:
Do you see any messages?

troll3rsk8tr, 2020-09-12:
No, just a solid green cursor.

alistair23, 2020-09-12:
That seems like the problem then. You should be able to communicate with the chip.

troll3rsk8tr, 2020-09-13:
Interesting, what steps should I take from here?