brendan-w/python-OBD #202: Help in commands

zeesh1979, 2020-10-05:
First I am sorry of this question doesnt fit in the categor of reporting issues...
i wanted to know if i can use this package to control stuff in car like door lock/unlock, turn on/off headlights/indicators etc etc
i am currently having Honda Civic 2011 to test these on

studioj, 2020-11-23:
default obd (on board diagnostics) doesnt support that. 
Often cars implement those things on other busses which sometimes can be accessed through obd communication but that's really brand specific. I know for example on the BMW this is handled on what they call the K-Bus

maybe read up here https://python-obd.readthedocs.io/en/latest/Command%20Tables/ on the supported commands.

for your project maybe look here for more info
https://www.giac.org/paper/gcia/9927/hacking-bus-basic-manipulation-modern-automobile-bus-reverse-engineering/133228

zeesh1979, 2020-11-26:
can you tell me how i can get values without units from this library?
when i run RPM it gives me value with revolutions per minute ( i just need value). same with speed

https://www.giac.org/paper/gcia/9927/hacking-bus-basic-manipulation-modern-automobile-bus-reverse-engineering/133228

studioj, 2020-12-05:
to be honest I'd replace them 
pseudocode 
    value.replace("rpm","")

you can also have a look at pint which is a grrat lib for units
https://pypi.org/project/Pint/