brendan-w/python-OBD #204: Python ignoring obd module

lkv0315, 2020-10-21:
So I have been trying to install python-OBD as a proof of concept for a school project on my raspberry pi 4 4GB. I followed the instructions and did 
`pip install obd`
which seemed to go well, as it pulled packages and installed them.
I tried to use the default program you had written as an example: 
`
import obd
obd.logger.setLevel(obd.logging.DEBUG)
connection = obd.OBD("/dev/ttyUSB0")
cmd = obd.commands.SPEED
response = connection.query(cmd)
print(response.value)
`
with the /dev/ttyUSB0 being the path to my USB ELM327 adapter.
when I run this file: `python obd.py`
the following is returned:
`pi@raspberrypi:~ $ python obd.py
Traceback (most recent call last):
  File "obd.py", line 2, in <module>
    import obd
  File "/home/pi/obd.py", line 3, in <module>
    obd.logger.setLevel(obd.logging.DEBUG)
AttributeError: 'module' object has no attribute 'logger'`

I checked to see if it was an install location problem and moved the package from 
`~/.local/lib/python2.7/site-packages`
to
`usr/lib/python2.7/dist-packages`
since the second location contained lots of python modules. 
I'm not sure what the issue is since I followed the instructions letter for letter.

studioj, 2020-11-23:
try installing with
`python -m pip install obd` => which means "dear python please use pip with the parameters 'install obd' "
then you know for sure the package will be installed for the right python installation (copying packages around is always tricky)

geoffwright240, 2021-01-20:
Have you solved this issue yet? If so, please close it.

If not, I believe the problem may be that you've called your test program "obd.py",  the same name as the module. Please rename your program to something different and try again.