brendan-w/python-OBD #205: Everything connected but no data

zeesh1979, 2020-10-21:
Hi,
Please check as I made a very simple script to get RPM and Speed

` import obd
import time
from obd import OBDStatus
conn=obd.OBD(fast=False,check_voltage=False) //made check voltage false as sometimes i was getting Ignition off reply even though my car was running
time.sleep(5)
print("elm",OBDStatus.ELM_CONNECTED)
print("obd",OBDStatus.OBD_CONNECTED)
print("car",OBDStatus.CAR_CONNECTED)
time.sleep(5)
print(obd.commands.has_command(obd.commands.SPEED))
print(obd.commands.has_command(obd.commands.RPM))
cmd1=obd.commands.SPEED
cmd2=obd.commands.RPM
while(1):
	resp=conn.query(cmd1,force=True)
	print("speed val")
	print(resp.value)
	resp=conn.query(cmd2,force=True)
	print("rpm val")
	print(resp.value) `
I get Connected status for ELM, OBD and Car. I also get True for both commands available check. but the commands return nothing and i get None for each. i even increase car RPM by pressing the accelerator (car in parked state and engine running)
I connected my OBD with android phone and used the famous Torque app and i am getting all values

zeesh1979, 2020-10-21:
[image]
I am using this OBD module from [https://www.banggood.com/IMars-ELM327-bluetooth-Car-OBD2-Scanner-Diagnostic-Tool-Engine-Code-Reader-p-913030.html?rmmds=search&cur_warehouse=CN](url)

zeesh1979, 2020-10-21:
just now i ran the deubg with below output. unable to understand where is the issue

obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/rfcomm0']
[obd.obd] Attempting to use port: /dev/rfcomm0
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: b'?\r\r>'
[obd.elm327] Choosing baud 38400
[obd.elm327] write: b'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'\xfc\r\rELM327 v1.5\r\r>'
[obd.elm327] write: b'ATE0\r'
[obd.elm327] read: b'ATE0\rOK'
[obd.elm327] write: b'ATH1\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: b'ATL0\r'
[obd.elm327] read: b'OK\r'
[obd.elm327] write: b'ATSP0\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'\r\r>'
[obd.elm327] write: b'ATDPN\r'
[obd.elm327] read: b'A0\r\r>'
[obd.elm327] ELM responded with unknown protocol. Trying them one-by-one
[obd.elm327] write: b'ATTP6\r'
[obd.elm327] read: b'OK\r\r'
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'NO DATA\r\r>'
[obd.elm327] Connected Successfully: PORT=/dev/rfcomm0 BAUD=38400 PROTOCOL=6
[obd.obd] querying for supported commands
[obd.obd] Sending command: b'0100': Supported PIDs [01-20]
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'NO DATA\r\r>'
[obd.OBDCommand] b'0100': Supported PIDs [01-20] did not receive any acceptable messages
[obd.obd] No valid data for PID listing command: b'0100': Supported PIDs [01-20]
[obd.obd] Sending command: b'0600': Supported MIDs [01-20]
[obd.elm327] write: b'0600\r'
[obd.elm327] read: b'NO DATA\r\r>'
[obd.obd] finished querying with 7 commands supported
[obd.obd] ===================================================================

alistair23, 2020-10-21:
There is no `ELM327 v1.5` you are using a knock off ELM327 chip. We don't support knock off chips, as it's possible it doesn't match the ELM spec.

I would try specifying the protocol your car uses manually, maybe that will help.

closed 2020-10-21 by alistair23

zeesh1979, 2020-10-21:
thanks. how  can i manually specify the protocol?i did soem search and found out that Honda Civic uses CAN 29bit protocol. how to specify this connect command?

zeesh1979, 2020-10-21:
also can u recommend some good (a bit cheaper) adapter that is compatible with this package?

alistair23, 2020-10-21:
Just specify `protocol=x` when starting OBD. The documentation describes the different options: https://python-obd.readthedocs.io/en/latest/Connections/#protocol_id

alistair23, 2020-10-21:
I don't have a recommendation for a OBD connector. I designed my own with a official ELM327 chip.

[SparkFun](https://www.sparkfun.com/products/9555) have one that uses the STN1110 chip instead. This library works with that as well (as it's ELM compliant).

zeesh1979, 2020-10-22:
Thanks and how about syntax to do manual protocol setting in obd connect?

zeesh1979, 2020-10-22:
one last question hopefully :)
please check the logs that i also attached in raising issue. if you see yellow highlighted line, it forces manual protocol setting and gets success at 6 (ISO_15765_4_11bit_500k). shouldnt this mean that my adapter has responded well and next all commands will be sent via this protocol?orange highlighted checking supported commands and it returns no data...if package is communicating with right protocol, i.e. 6, then it should be ok. i was thinking if in my code i do manual protocol set as you suggested even then this will not work. regarding elm versions, i checked on many websites like aliexpress,banggood etc all of them saying v1.5

https://www.banggood.com/search/elm32.html
https://www.aliexpress.com/af/elm327.html?d=y&origin=n&SearchText=elm327&catId=0&initiative_id=SB_20201021232046

[obd.elm327] ELM responded with unknown protocol. Trying them one-by-one
[obd.elm327] write: b'ATTP6\r'
[obd.elm327] read: b'OK\r\r'
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'NO DATA\r\r>'
[obd.elm327] Connected Successfully: PORT=/dev/rfcomm0 BAUD=38400 PROTOCOL=6
[obd.obd] querying for supported commands
[obd.obd] Sending command: b'0100': Supported PIDs [01-20]
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'NO DATA\r\r>'
[obd.OBDCommand] b'0100': Supported PIDs [01-20] did not receive any acceptable messages
[obd.obd] No valid data for PID listing command: b'0100': Supported PIDs [01-20]
[obd.obd] Sending command: b'0600': Supported MIDs [01-20]
[obd.elm327] write: b'0600\r'
[obd.elm327] read: b'NO DATA\r\r>'

alistair23, 2020-10-24:
I have seen cases where Python-OBD sometimes guesses the wrong protocol. From memory I have only seen this when there is a hardware error, but with a knock-off adapter it's hard to know.

zeesh1979, 2020-10-24:
i installed a terminal device and sent some text commands. it well responded. attached are the logs. can this help me customize ur package for my knocked off obd? i see it responded to speed, rpm commands as well  as 0100 which was giving no data in your package

  1 Connection: Saturday, October 24, 2020 5:08:57 PM	
  2 OBD Now Terminal - Version 1.0.16       	08:57.221
  3 Android Version: 10                     	08:57. 225
  4 Build Number: HUAWEIHMA-L29             	08:57.225
  5 Manufacturer: HUAWEI                    	08:57.225
  6 Model: HMA-L29                          	08:57.225
  7 Brand: HUAWEI                           	08:57.225
  8 Device: HWHMA                           	08:57.225
  9 Product: HMA-L29                        	08:57.225
 10 Country: United States                  	08:57.225
 11 Bluetooth Device: OBDII                 	08:58.407
 12 Bluetooth Address: 00:1D:A5:68:98:8A    	08:58.408
 13 at                                      	09:04.646
 14 ?                                       	09:04.734
 15 >                                       	09:04.734
 16 AT                                      	09:08.909
 17 ?                                       	09:08.970
 18 >                                       	09:08.970
 19 ATL1                                    	10:16.062
 20 OK                                      	10:16.096
 21 
                                       	10:16.096
 22 
 23 ath1                                    	10:19.812
 24 OK                                      	10:19.833
 25 
                                       	10:19.833
 26 
 27 ats1                                    	10:29.989
 28 OK                                      	10:30.011
 29 
                                       	10:30.011
 30 
 31 atal1                                   	10:33.093
 32 OK                                      	10:33.135
 33 
                                       	10:33.135
 34 
 35 atal                                    	10:40.239
 36 OK                                      	10:40.255
 37 
                                       	10:40.255
 38 
 39 atma                                    	10:52.515
 40 ATMA                                    	11:09.807
 41 STOPPED                                 	11:09.831
 42 
                                       	11:09.831
 43 
 44 
                                       	11:09.831
 45 
 46 ATMA                                    	11:17.718
 47 atma                                    	11:52.957
 48 STOPPED                                 	11:53.046
 49 
                                       	11:53.046
 50 
 51 
                                       	11:53.046
 52 
 53 ATMA                                    	11:57.018
 54 ATMA                                    	12:03.264
 55 STOPPED                                 	12:03.278
 56 
                                       	12:03.278
 57 
 58 
[... 95 lines cut ...]
130 
131 0103                                    	22:33.785
132 18 DA F1 11 04 41 03 02 00              	22:33.919
133 
                                       	22:33.919
134 
135 0303                                    	22:37.303
136 NO DATA                                 	22:37.555
137 
                                       	22:37.555
138 
139 0104                                    	24:47.799
140 18 DA F1 11 03 41 04 6B                 	24:47.925
141 
18 DA F1 1D 03 41 04 6C                	24:47.925
142 
                                       	24:47.925
143 
144 0105                                    	24:59.921
145 18 DA F1 11 03 41 05 6F                 	25:00.055
146 
18 DA F1 1D 03 41 05 6F                	25:00.055
147 
                                       	25:00.055
148

zeesh1979, 2020-10-26:
Hi,
I have been successful communicating with my same adapter using below script and it responds very well....now can you guide me if its possible to customize your library and get it worked with this adapter?

import bluetooth
import threading
import serial
import time
sockfd = bluetooth.BluetoothSocket(bluetooth.RFCOMM)
sockfd.connect(('98:D3:31:xx:xx:xx', 1)) # BT Address
def RViaBluetooth():
    while(1):
        data = sockfd.recv(4096)
        print(data)

x=threading.Thread(target=RViaBluetooth)
x.start()
#sockfd.close()
while(1):
    tosend = input("quit?" )
    if tosend != 'q':
        #sockfd.send(tosend)
        print("sending")
        sockfd.send('ATSP0')
        time.sleep(2)
        sockfd.send('ATI')
        time.sleep(2)
        sockfd.send('0103')
        time.sleep(2)
        sockfd.send('0104')
    else:
        break
    sockfd.close()

zeesh1979, 2020-11-15:
Hi,
I have finally mamaged to get v2.1 obd and it successfully give me working debug as below log. but now issue is when i use your simple script to get speed and RPM, script returns None. below was done after adding delay=1 in every command of elm327.py

[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/rfcomm0']
[obd.obd] Attempting to use port: /dev/rfcomm0
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: b'?\r\r>'
[obd.elm327] Choosing baud 38400
[obd.elm327] write: b'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'\r\rELM327 v2.1\r\r>'
[obd.elm327] write: b'ATE0\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'ATE0\rOK\r\r>'
[obd.elm327] write: b'ATH1\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'ATL0\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'ATSP0\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'SEARCHING...\r18 DA F1 11 06 41 00 BC 3F A8 03 \r18 DA F1 1D 06 41 00 98 3A 80 01 \r\r>'
[obd.elm327] write: b'ATDPN\r'
[obd.elm327] read: b'A7\r\r>'
[obd.protocols.protocol] map ECU 17 --> ENGINE
[obd.protocols.protocol] map ECU 29 --> UNKNOWN
[obd.elm327] Connected Successfully: PORT=/dev/rfcomm0 BAUD=38400 PROTOCOL=7
[obd.obd] querying for supported commands
[obd.obd] Sending command: b'0100': Supported PIDs [01-20]
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'18 DA F1 11 06 41 00 BC 3F A8 03 \r18 DA F1 1D 06 41 00 98 3A 80 01 \r\r>'
[obd.obd] Sending command: b'0120': Supported PIDs [21-40]
[obd.elm327] write: b'0120\r'
[obd.elm327] read: b'18 DA F1 1D 06 41 20 80 00 A0 01 \r18 DA F1 11 06 41 20 80 1D B0 01 \r\r>'
[obd.obd] Sending command: b'0140': Supported PIDs [41-60]
[obd.elm327] write: b'0140\r'
[obd.elm327] read: b'18 DA F1 11 06 41 40 7A D0 00 00 \r18 DA F1 1D 06 41 40 CA C0 00 00 \r\r>'
[obd.obd] Sending command: b'0600': Supported MIDs [01-20]
[obd.elm327] write: b'0600\r'
[obd.elm327] read: b'NO DATA\r\r>'
[obd.obd] finished querying with 73 commands supported

________________________________
From: Muhammad Zeeshan Akram <zeesh1979@hotmail.com>
Sent: Thursday, October 22, 2020 10:22 AM
To: brendan-w/python-OBD <reply@reply.github.com>
Subject: Re: [brendan-w/python-OBD] Everything connected but no data (#205)

one last question hopefully :)
please check the logs that i also attached in raising issue. if you see yellow highlighted line, it forces manual protocol setting and gets success at 6 (ISO_15765_4_11bit_500k). shouldnt this mean that my adapter has responded well and next all commands will be sent via this protocol?orange highlighted checking supported commands and it returns no data...if package is communicating with right protocol, i.e. 6, then it should be ok. i was thinking if in my code i do manual protocol set as you suggested even then this will not work. regarding elm versions, i checked on many websites like aliexpress,banggood etc all of them saying v1.5

https://www.banggood.com/search/elm32.html
https://www.aliexpress.com/af/elm327.html?d=y&origin=n&SearchText=elm327&catId=0&initiative_id=SB_20201021232046

[obd.elm327] ELM responded with unknown protocol. Trying them one-by-one
[obd.elm327] write: b'ATTP6\r'
[obd.elm327] read: b'OK\r\r'
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'NO DATA\r\r>'
[obd.elm327] Connected Successfully: PORT=/dev/rfcomm0 BAUD=38400 PROTOCOL=6
[obd.obd] querying for supported commands
[obd.obd] Sending command: b'0100': Supported PIDs [01-20]
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'NO DATA\r\r>'
[obd.OBDCommand] b'0100': Supported PIDs [01-20] did not receive any acceptable messages
[obd.obd] No valid data for PID listing command: b'0100': Supported PIDs [01-20]
[obd.obd] Sending command: b'0600': Supported MIDs [01-20]
[obd.elm327] write: b'0600\r'
[obd.elm327] read: b'NO DATA\r\r>'