brendan-w/python-OBD #208: Stuck on "Initializing ELM327: PORT=\.\COM3 BAUD=auto PROTOCOL=auto"

Jdrinkwater1, 2020-11-12:
I am trying to connect to a Bluetooth obd2 reader and I am getting stuck on Initializing ELM327: PORT=\.\COM3 BAUD=auto PROTOCOL=auto. Not sure why this is happening. I can connect to it with scan tool, so it must be something that I am doing.

Here is a part of my code 

`print("Test")
ports = obd.scan_serial()
print(ports)
obd.logger.setLevel(obd.logging.DEBUG)
connection = obd.OBD(ports[2])
print("test")
connect = obd.Async (fast=False)
print("test")
rpm = obd.commands.RPM
throttlePos = obd.commands.THROTTLE_POS`

ormopl, 2020-12-30:
The same here. Tried with various protocols, bauds and adapters. No response from a car, while other application using the same adapters works fine.

VicDominguez, 2021-01-06:
> I am trying to connect to a Bluetooth obd2 reader and I am getting stuck on Initializing ELM327: PORT=.\COM3 BAUD=auto PROTOCOL=auto. Not sure why this is happening. I can connect to it with scan [...]

Could you upload your console output? And other data, like car model. I had a similar bug days before and maybe is similar to my failure

Jdrinkwater1, 2021-01-06:
[image]

Also the car is a 2008 Mercury Milan

VicDominguez, 2021-01-06:
> [image]
> 
> Also the car is a 2008 Mercury Milan

Are you sure that these port works? It's look like python-OBD and ELM327 doesn't communicate.. If you don't specify any port python-OBD checks all ports. If AT commands doesn't work, the problem isn't on the car. You can check your ELM327 using tools like ELM327 FG Tester

austinlg96, 2021-05-09:
Some additional data:

I've got an OBDLink MX+ connected to my Windows 10 computer over bluetooth

It seems to add two different COM ports to my computer, one of which hangs and one of which works successful. I've had trouble using a keyboard interrupt to break out of it and have resorted to just killing the shell.

(Note: I also wasn't able to get it working with `\\.\\COMn`, I had to adjust the code to use just `COMn`. Unfortunately I'm not certain if this is a bug or if something is different/unique to my situation.)