brendan-w/python-OBD #209: ELM327 not detected

PietroMacori, 2020-11-30:
Hi, I have an issue trying to connect to my ELM327 device to my Raspberry Pi 2 running ubuntu mate. The adapter is perfectly running on my Android phone (using torque) and even with the raspberry running a java program I wrote last year. But using this library I cannot connect since it looks like the library doesn't detect my ELM327 at all.

Using obd.OBD() the program tells me he didn't find any adapter. I also tried to explicitely tell the program where to find the ELM327 (in /dev/rfcomm0) but I received the following error "could not open port /dev/rfcomm0". 
Then I tried to execute python using sudo and I get a different error "ATH1 did not return 'OK' or echoing still on".

Finally I tried to set fast=False and timeout=30 as described in the documentation but again the adapter is not detected at all.

What should I do? Does anyone had the same problem. Just to be clear, I use a bluetooth dongle but I don't think it's the problem since as I said I could run a java OBD program on the raspberry.

Thank you in advance.

alexastro80, 2020-12-11:
So I will start by saying I am having a similar problem except with a Raspberry Pi 4. The using sudo problem, is probably due to your current user's write/read access to the port. 

I have seen other similar problems with the error of: "ATH1 did not return 'OK' or echoing still on". In other issues, one was resolved, but I don't know why the problem is still happening. I myself am running into this error. Could it be we are running a version behind the patch? pip says I am running 0.7.1, which this repository claims to be at. Maybe deleting the one from pip and cloning the repo will include this patch.

alexastro80, 2020-12-11:
@PietroMacori I checked within the module that pip installed and it is behind. Try and clone the one in the repository and see if it works. I will check on my pi later to see if it solves my problem.