brendan-w/python-OBD #212: Can't connect to Obd after first time

mddrtg, 2021-01-25:
Hi, I'm having a strange issue in which the simple example works only the first time I execute the code ,but once  I try to execute again , I get the error which seems to be common here , the only way to make it work again is to reboot the whole pi.

"serial.serialutil.SerialException: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?"

I've already configured dbus-org.bluez.service  with 
ExecStart=/usr/lib/bluetooth/bluetoothd -C
ExecStartPost=/usr/bin/sdptool add SP 

and I also tried to clone the library from repository but still ..

the second issue which might relates to first one is that whenever I use the Async Connection , it always returns none.

I'm using Raspberry Pi 4 with Bluetooth OBD

gmaniak, 2021-06-24:
From what you're describing it seems more of a problem with the connection to the adapter than a problem with the OBD library itself. Maybe try to manually open a serial connection to the adapter to see if that works?

What OBD adapter are you using?
I saw a problem with fake ELM327 adapters. A second connect would fail. For me unplugging the OBD adapter and plugging it back in solved the problem.

mptak12, 2022-10-31:
> Hi, I'm having a strange issue in which the simple example works only the first time I execute the code ,but once I try to execute again , I get the error which seems to be common here , the only [...]

Hello, 
Now I am facing the same problem. 

Is there any solution for Raspberry Pi 3?