brendan-w/python-OBD #216: Serialize OBDResponse to JSON

NaturesProphet, 2021-03-25:
Hello folks.

Please, i need some help to serialize OBDResponse objects to JSON, any idea about how can i do it? i'm new to python.

ConnorNovak, 2021-03-29:
Being more specific with your request will help you get help:
- Do you need to serialize an arbitrary OBDResponse object to JSON, or just some specific ones you happen to be using?
- What is the purpose of the serialization? Are you storing the objects - writing many at once, or streaming them to another service - handling one-by-one?
- How much data do you need in the serialization? Python's `pickle` module will serialize entire objects, for example. Do you need that level of serialization in JSON? Or do you need only the measurement type and value for each reading?

NaturesProphet, 2021-03-29:
Hello Connor, thanks for reading this.

I want to serialize the entire OBDResposne object to json and use this json as a response on an flask rest endpoint that i am integrating with the obd. my objective is to serve a http rest api that will respond with OBDResponse as a json, and i want all the nested documents inside that object.

well, as i said, i'm new to python, my home language is javascript and it would be really easy to do such task on that language without write a logic for each field. i want something in javascript way if possible.

ConnorNovak, 2021-03-29:
My personal opinion: Your best bet would be creating a function that serializes a given `OBDResponse` object. Looking at the documentation for the object, it appears to have four salient attributes (you should probably include the units property as well, so make it 5): `command`, `messages`, `value`, `time`, and `units`. It looks, at first glance, like each of these can be cast to a base type. A single function should then suffice for any `OBDResponse`.

If you wanted to contribute to the library, you could make an `as_json()` method for OBDResponse, or something similar. You could also drop your function in `utils.py`. I'm not associated with this repository in any way, so I can't say what reception if would have.

Salient code points:
https://github.com/brendan-w/python-OBD/blob/master/obd/obd.py  line 298: Creation of `OBDResponse` object
https://github.com/brendan-w/python-OBD/blob/master/obd/OBDCommand.py line 85: unpacking of response data into `OBDResponse` object

bgunson, 2022-07-03:
Hi, I wrote a JSON encoder which works for most of the python-OBD types. 

Check it out at https://github.com/bgunson/obd-socketio

To use it `pip install obd-socketio`
```
import obdio

obdio.loads(...)
obdio.dumps(...)    # etc
```

Contribution is also welcome becuase its not fully implemented but does the job.

NaturesProphet, 2022-07-03:
awesome dude