brendan-w/python-OBD #217: Read steering angle

sieuwe1, 2021-06-05:
Hi

I am trying to read the raw steering angle data from my bmw e46. The sensor is located on the can bus and bus able to be read using the diagnostics bus because I can view it using the official bmw diagnostic tool. 

I however saw that this library does not have predefined messages for steering angle. Is there anything known about reading steering angle with this library or hidden features which can enable this? Or can I set a costom message filter with the id of the sensor and read the values?

thanks

sieuwe

cewbdex, 2021-06-08:
I think the sensor is not OBD compliant, you would need to reverse engineer the CAN diagnostics, in theory you could hook up the official tool and capture the traffic between those devices, decode it and get the value.

lkoczorowski, 2021-06-30:
OBD is one type of message on your car's CAN bus. If you want to read something outside of the OBD standard you need to go level deeper and read the CAN bus directly. Check out "python-can".  Your ELM OBD reader wont' help you either. You'll need hardware like a PiCAN2 if you're using an RPi. 

On most cars you can access the CAN bus directly through the OBD port but I think that's changing now so you'll need to check how people are connecting to CAN on your car. Sometimes you literally have to splice into the bus directly with a pigtail. 

Once you are able to connect direct to your CAN bus, you'll have to do the work of deciphering the 100s of messages on there and find the one that's steering angle. Hopefully someone's already done it for BMW and you can find it online. 

Good luck!