brendan-w/python-OBD #218: Problem with Baudrate

ThomasMeister123, 2021-06-13:
Hello,

in my case there is a problem with the baudrate. The program connect to the OBD-adapter but then there comes an error:

`

>>> %Run OBD.py
###########
['/dev/rfcomm0']
###########
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/rfcomm0']
[obd.obd] Attempting to use port: /dev/rfcomm0
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: b'\x7f\x7f\r?\r\r>'
[obd.elm327] Choosing baud 38400
[obd.elm327] write: b'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'ATZ\r\r\rELM327 v2.1\r\r>'
[obd.elm327] write: b'ATE0\r'
[obd.elm327] read: b'ATE0\rOK\r\r>'
[obd.elm327] write: b'ATH1\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'ATL0\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'AT RV\r'
[obd.elm327] read: b'11.8V\r\r>'
[obd.elm327] write: b'ATSP0\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'\r\r>'
[obd.elm327] write: b'ATDPN\r'
[obd.elm327] Failed to read port
[obd.elm327] read: b''
[obd.elm327] Failed to retrieve current protocol
[obd.elm327] Adapter connected, but the ignition is off
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: b''
[obd.elm327] Response from baud 9600: b''
[obd.elm327] Response from baud 230400: b''
[obd.elm327] Response from baud 115200: b''
[obd.elm327] Response from baud 57600: b''
[obd.elm327] Response from baud 19200: b''
**[obd.elm327] Failed to choose baud
[obd.elm327] closing port
[obd.elm327] write: b'ATZ\r'
[obd.elm327] Failed to set baudrate**
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
[obd.obd] Query failed, no connection available
None
Traceback (most recent call last):
  File "/home/pi/Skripte/OBD.py", line 26, in <module>
    print(response.value.to("mph")) # user-friendly unit conversions
AttributeError: 'NoneType' object has no attribute 'to'
>>> 

`

Whats wrong?

Python-Code:
`
#!/usr/bin/python3
import obd
from obd import OBDStatus

ports = obd.scan_serial()
print("###########")
print(ports)
print("###########")
obd.logger.setLevel(obd.logging.DEBUG) # enables all debug 
connection = obd.OBD(ports[0]) # connect to the first port in the list
cmd = obd.commands.SPEED # select an OBD command (sensor)
response = connection.query(cmd) # send the command, and parse the response
print(response.value) # returns unit-bearing values thanks to Pint
print(response.value.to("mph")) # user-friendly unit conversions
`

ThomasMeister123, 2021-06-13:
So, I hat a working connection ONE time. With thise line:
connection = obd.OBD(baudrate=None,fast=False, timeout=30) 

Log:

`[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/rfcomm0']
[obd.obd] Attempting to use port: /dev/rfcomm0
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: b'?\r\r>'
[obd.elm327] Choosing baud 38400
[obd.elm327] write: b'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'\r\rELM327 v2.1\r\r>'
[obd.elm327] write: b'ATE0\r'
[obd.elm327] read: b'ATE0\rOK\r\r>'
[obd.elm327] write: b'ATH1\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'ATL0\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'AT RV\r'
[obd.elm327] read: b'14.2V\r\r>'
[obd.elm327] write: b'ATSP0\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'\r\r>'
[obd.elm327] write: b'ATDPN\r'
[obd.elm327] read: b'A0\r\r>'
[obd.elm327] ELM responded with unknown protocol. Trying them one-by-one
[obd.elm327] write: b'ATTP6\r'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'7E8 06 41 00 BE 1F B8 13 \r\r>'
[obd.protocols.protocol] map ECU 0 --> ENGINE
[obd.elm327] Connected Successfully: PORT=/dev/rfcomm0 BAUD=38400 PROTOCOL=6
[obd.obd] querying for supported commands
[obd.obd] Sending command: b'0100': Supported PIDs [01-20]
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'7E8 06 41 00 BE 1F B8 13 \r\r>'
[obd.obd] Sending command: b'0120': Supported PIDs [21-40]
[obd.elm327] write: b'0120\r'
[obd.elm327] read: b'7E8 06 41 20 80 05 A0 01 \r\r>'
[obd.obd] Sending command: b'0140': Supported PIDs [41-60]
[obd.elm327] write: b'0140\r'
[obd.elm327] read: b'7E8 06 41 40 6A D0 00 00 \r\r>'
[obd.obd] Sending command: b'0600': Supported MIDs [01-20]
[obd.elm327] write: b'0600\r'
[obd.elm327] read: b'7E8 06 46 00 C0 00 00 01 \r\r>'
[obd.obd] Sending command: b'0620': Supported MIDs [21-40]
[obd.elm327] write: b'0620\r'
[obd.elm327] read: b'7E8 06 46 20 80 00 00 00 \r\r>'
[obd.obd] finished querying with 73 commands supported
[obd.obd] ===================================================================
[obd.obd] Sending command: b'010D': Vehicle Speed
[obd.elm327] write: b'010D\r'
[obd.elm327] read: b'7E8 03 41 0D 00 \r\r>'
0.0 kph
0.0 mph
`

The I tried a second time (same code, restarted PI (car: ignition off, ignition on, started engine).

Log:

`###########
['/dev/rfcomm0']
###########
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/rfcomm0']
[obd.obd] Attempting to use port: /dev/rfcomm0
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=38400 PROTOCOL=auto
[obd.elm327] write: b'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] Failed to read port
[obd.elm327] read: b''
[obd.elm327] write: b'ATE0\r'
[obd.elm327] Failed to read port
[obd.elm327] read: b''
[obd.elm327] closing port
[obd.elm327] write: b'ATZ\r'
[obd.elm327] ATE0 did not return 'OK'
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
[obd.obd] Query failed, no connection available
None
Traceback (most recent call last):
  File "/home/pi/Skripte/OBD.py", line 26, in <module>
    print(response.value.to("mph")) # user-friendly unit conversions
AttributeError: 'NoneType' object has no attribute 'to'
>>> `

I dont get it ;(

ThomasMeister123, 2021-06-13:
So, it works with THIS line:

connection = obd.OBD(baudrate=38400,protocol="6",fast=False, timeout=40)

But not with the default (auto-)Settings?

gmaniak, 2021-06-24:
With this line `connection = obd.OBD(baudrate=38400,protocol="6",fast=False, timeout=40)` it works every time? Even when you do the restart procedure: rpi restart, ignition off, ignition on, engine on.

csurf, 2023-07-03:
You have a weak or unstable bluetooth connection, or a bad adapter. I would first try manually connecting to the adapter serial port, `rfcomm0`, using a serial terminal application, and trying to manually send commands to the elm327 in order to see if you can easily establish a working connection & get valid responses to your commands. If not, then there's a problem with your link.