brendan-w/python-OBD #225: [obd.obd] 'b'010D': Vehicle Speed' is not supported

guruvishnuvardan, 2021-08-14:
Hello Folks,

@brendan-w  #@alistair23

I am using USB OB2 Connector, my objective is to get the speed of the Car(Honda City 2015 Model India)

Please find the following Code:
**import obdq
connection = obd.OBD("/dev/ttyUSB0")
ports = obd.scan_serial()
print(ports)

cmd = obd.commands.SPEED
r = connection.query(cmd)
print(r.value)
OutPut:
['/dev/ttyUSB0']
[obd.obd] 'b'010D': Vehicle Speed' is not supported
None**
I have made an attempt to print using connection.print_discovered(), Please find the output
**connection.print_discovered()
AttributeError: 'OBD' object has no attribute 'print_discovered'**

import obd
connection = obd.OBD("/dev/ttyUSB0")
a = connection.get_supported_commands()
Traceback (most recent call last):
File "", line 1, in
**AttributeError: 'OBD' object has no attribute 'get_supported_commands'**

System Information:
Raspberry Pi 4
Raspbian OS
pyOBD

Can you please help me resolve this issue
Note: I have attached the USB OBD Connector Picture:

[image]

guruvishnuvardan, 2021-08-19:
@brendan-w - Can you please help me fix this issue?

connervieira, 2021-08-25:
I'm having the same issue. Regardless of what command I choose (RPM, speed, etc.) I get the same error as you. The only command I can successfully receive are the ELM commands, like ELM_VOLTAGE

guruvishnuvardan, 2021-08-25:
@connervieira -- Yes, I guess it's about the OBD 2 adapter ... @brendan-w -- Can you please help us

csurf, 2023-07-03:
When python-OBD initializes it's connection to the elm327 chip, it sends PID '01 00', which is a request to get all PID's that are supported by the vehicle. I've found that python-OBD will sometimes fail to process the response to this command. When this happens, python-OBD won't build it's internal list of 'supported commands' and will simply reject any attempt at requesting those commands. The ony way around it is to either use the 'force' option when running a command query, or to fix the root problem with the initial '01 00' command failing.

In my case, I needed to manually set the protocol and also add a timeout to the initial 'get supported commands' query within the python-OBD code.