brendan-w/python-OBD #232: Slower response vs Torque - any ideas?

rack201, 2021-12-24:
I seem to get significantly faster responses using Torque Pro vs. what I can do with Python-OBD using the same BT adaptor.  I thought maybe Python-OBD was not adding the 'reply line length' (e.g. 010C**1**) but I see in the code that it does.  Any thoughts on what the difference might be?  Torque has an 'enable faster responses' option which makes a difference, but I don't know what it really does.

I'm not using async but not sure if that would help in my case.  My script takes the PID response and send via MQTT to node-red.

Any ideas?

rack201, 2021-12-25:
I analyzed a BT capture when using Torque in 'normal' and 'faster responses' modes.  The difference seems to be the use of ATAT2 during initialization for the faster mode which is more aggressive adaptive timing than the default.  I've added this to elm327.py and will test.  I also note that Torque turns headers off (ATH0) which breaks pypthon-OBD and spaces off (ATS0) - not sure if these would make a real word difference.

csurf, 2023-07-03:
turning spaces off *seemed to make my cheap elm327 clone BT adapter respond a tiny bit faster, but I didn't actually measure it, so it might be completely subjective & imagined. I'd venture to guess that adding the spaces adds a slight extra load to the processor. FYI, this is for the raw commands sent to the elm327, without even including python-OBD. 

I've also considered disabling the headers, but haven't looked too deeply into what the implications would be. From what I understand, they're only really necessary to distinguish between different onboard ECU's in vehicles which may have multiple ECU's responding to OBD request messages, although I might be incorrect on this.